DacoTaco/priiloader #204: Priiloader download link on WiiBrew broken?

FuzzyQuills, 2016-06-27:
So... i went to download priiloader today...
...But the download link on WiiBrew redirects to a blank page. inputting the site itself without the link redirects to a cosplay website...

Has someone hijacked your domain by any chance? And above all, where's the actual download now? I'd rather not have to build from source...

FuzzyQuills, 2016-06-27:
Ok, adding that indeed, it does appear to be our site, but atm, download link returns an error 500

avarun42, 2016-08-13:
Did you ever find the download?

FuzzyQuills, 2016-08-14:
WOW. I completely forgot about this, lol.

I do have Priiloader now, not sure if it was a slightly older version someone else uploaded somewhere, or if I found the download itself. I'll test it again today if you like.

NuclearFej, 2016-08-31:
[priiloader.zip](https://github.com/DacoTaco/priiloader/files/446237/priiloader.zip)

This is either 0.8 or 0.8.1. META.XML says 0.8 but two downloads I tried both said that it is 0.8.1. Not too many differences between the two anyway.

DacoTaco, 2016-10-03:
so euh.... i just saw this issue report lol XD

this probably happened when i updated php some time ago. 

thanks @NuclearFej for the mirror. ill see when i can fix this :)

DacoTaco, 2016-10-03:
ye, it was a PHP function that got deleted in PHP5 (this was a major upgrade of several versions at once XD )

it should be fully functioning again, can you guys confirm?

closed 2016-11-06 by DacoTaco