DacoTaco/priiloader #206: priloader setup password but wiimote cant sync

kimnue, 2017-07-05:
hi. i really no idea how to do . i test function lock system menu with black screen. (curious to know what is this function)

the problem is after i test. i close the machine and use wiimote to another wii.

after i come back . i open wii. it is black screen. so i try go to wii priloader . The problem is i have password setup and i try to type the password but my wiimote cant sync. Any way to fix this issue

i have no gamecube controller

KcrPL, 2017-10-29:
Pair your Wii Remote with your Wii i guess :P

DacoTaco, 2017-10-30:
that would be the best, however afaik syncing wiimotes doesn't work in homebrew for some reason (i dont know why). would be great if it could though...

anyway, my best bet would be to use a gamecube controller and use the bypass password to get into the system and fix everything..

closed 2018-01-26 by DacoTaco