DacoTaco/priiloader #207: SYNC buttons - it's works or not ?

DjDiabolik, 2017-12-20:
Hi... i'm italian and i try to ask a support request to me (wrote in english) by a my friends.

He have a Wii......That's wii it's locked (black screen whit green LED only) at startup but he can reach the priiloader (apparently the version 0.7).
My friends as tryed but apparently the wiimote it's not syncing to the wii and he can't move (currently we can't have a gamecube controller) inside the various voices.

My simply question it's........... the SYNC button on wii it's works on priiloader ? I know for example on HomeBrew Channel yes, on HBC the sync buttons it's works but we don't know if also works on priiloader.

If right now the SYNC buttons works it's problably an hardware bluethoot on wii console... but if the sync buttons it's not works we need by force of find a gamecube controller and try to reach HBG and see what's append exactly on this console...

Thanks in advance for info........

DacoTaco, 2017-12-20:
first off all,
sorry to hear about the wii. 
ive been away for a very long time and my information might be outdated.
however, as far as i know, homebrew does not support the sync button and therefor priiloader does not support it.
but if you say HBC can do it now , that means somewhere along the line of me not being active support got added?

i suggest trying to find a gamecube controller to be sure. it could be the module, but it could also be something else.

was the wiimote synced somewhere else recently?

DjDiabolik, 2017-12-20:
mmmmmm.. i don't know if the wiimote it's used on another wii but i thinks no. It's probably append my friedns has kept active priiloader whit 15 seconds whitout wiimote and this cause the reset of know syncing right ?

I could say a bullshit (lol) but if i remember correctly from homebrew channel the sync buttons it's works but also i i don't use or turned on my wii from about two months :)

At this point i thinks the best solution it's to remedy a gamecube controller and from priiloader try to reach the homebrew channel and from here, for example, whit syscheck or similar see exactly what's append on the console.

After that i thinks whit some WAD Manager i can try to reinstall the system menu wad or the correct ios if exist a method to found it's on internet.............

DacoTaco, 2017-12-20:
i'd have to check when i have time. if libogc has syncing of wiimote working i will need to update and retest/rerelease the latest build lol. that would be huge.

but untill that time, all you can try is indeed wait in priiloader menu, then press reset (which should cause System menu to boot) and try and sync.
if that doesn't work you'll need a GC controller to navigate ye. first you will need to see if you can boot a game from the homebrew Channel and sync a wiimote(or from HBC itself if thats possible).
if system menu itself is broken i suggest making a bootmii backup, check if system menu is ok and reinstall system menu ios and system menu itself (in that order), then without leaving homebrew channel, installing priiloader (incase system menu is still broken) and then keep looking what is wrong.

DjDiabolik, 2017-12-21:
mmmmmmmm... ok..... the bootmii i don't know if it's installed (i thinks if yes as boot2 on sd) but if yes you suggest to do a backup in this status whit the system menu can't boot ? Because ?

If i want to reinstall the system menu ios and the system menu what's wad i need to search over internet ?

DacoTaco, 2017-12-21:
if it is installed as boot2 than even better :) 
i suggested the backup incase things get worse. this way you still have your data or a point to return too. check if it has bootmii first!

as for the WAD's, i would suggest using something like NUSD ( http://wiibrew.org/wiki/NUSD ) to download IOS 80 v6943 and system menu 4.3. followed by installing priiloader 0.8.2

DjDiabolik, 2017-12-21:
mmmm... but the nus downloader it's not works if i remember. The nintendo servers it's not offline at today?

FGOD1983, 2017-12-21:
if I remember correctly Nintendo took their servers offline for the Wii last year, so most probably you will not be able to download them with NUSD, but as far as I can see you can download them here: 
http://wehackwii.pbworks.com/w/page/8844411/IOS%20and%20RVL%20Downloads

DjDiabolik, 2017-12-21:
mmmmmm.. very strange because appartly nus download it's works. It's problably nintendo put in offline server for online gaming or similar services but it's works for system update or similar ?

I have check the files.... the IOS80-64-v6943.wad it's about 2MB.... hash it's:
A7540A54D37B0275F625238D54C438D8

And the system menu... it's file RVL-WiiSystemmenu-v514 about 24MB hash:
CAD93CA3CA41704B15BBEE95B2E6AAE2

I have download also the booth files from pbworks.com and very strange the md5 it's different.....

DacoTaco, 2017-12-21:
nintendo took al wii gaming and channel servers down. their update/shop server is pretty much alive since its also used by DSi, 3DS, WIIu and i can only guess switch too. so yes, nusd works just fine.

yes, it can be that their WAD has been modified in the process of making the WAD or they modified it with various pieces of unofficial code. my preferred method for getting official code is trough official means. aka, NUSD.

i hope for you it works

DjDiabolik, 2017-12-21:
Ok for moment i try to use the files maked by NUSD..... recapping all:
1) I need a gamecube controller
2) And see if HBC it's boot
3) From HBC i need a WAD Manager
4) Ever from gamecube controller i need to install in order.. first the IOS80 and after the system menu 514.

At this point... if i understand correctly i can reboot the wii and try to boot the system menu and cross the fingers.

Also if i understand correctly after the reinstallation of ios80 and the system menu i lost the priiloader on that wii but hbc it's remain... it's correct ?

DacoTaco, 2017-12-21:
dont cross your fingers, install priiloader after installing system menu 4.3 (using the 0.8.2 installer) :P 
so even if it doesn't work, you can still access priiloader/HBC

priiloader does get removed, as it installs itself as part of system menu. installing/updating system menu therefor removes it.
HBC is a channel on its own, it only gets deleted if the system menu does it (aka, if HBC is not any of the newer versions)
verify if your homebrew channel doesn't get deleted by the system menu. afaik, anything above 1.0.7 (1.0.7 including) is fine and will not get deleted.

other then that, you got that right

DjDiabolik, 2017-12-22:
Oh ok... i test if i can reach HBC.. if yes i install the priiloader 0.8.2 and only after try to see what's i can obtain after install of ios 80 and system menu :)

Wait some feedback or some post on next future day... problably after 25 December.

Ah.. and Marry Christmas :)

DacoTaco, 2017-12-22:
good luck and merry christmas to you too!

DacoTaco, 2018-01-26:
any news on this?

DjDiabolik, 2018-01-26:
No.... i could not to found a Gamecube controller and my friends don't want to buy it so we are stalled.....

DacoTaco, 2018-01-26:
thats sad to hear. if i have time i will look into the syncing of wiimotes in homebrew. it wont help you but it will help all others

DjDiabolik, 2018-01-27:
Yeah.. exactly. Most likely the life of that wii it's not the first priority of my friends.........

If i have some news i will back here to write new reply.....

DjDiabolik, 2018-02-14:
@DacoTaco i have some news.

My friends has remedied an GAMECUBE CONTROLLER!! It's have attached to the blocked WII and.... yeah.
It's possibly to REACH the HomeBrew Channel................... i hope in this next days i can check manually have this same wii on my hands and i/we can make all attemps to RE-BACK in life 👍

DacoTaco, 2018-02-19:
thats nice to hear!
good luck!

i also did a quick check and in the older HBC you can't sync wiimotes. maybe in the latest HBC, idk, but with the latest libogc it didn't work either...

DjDiabolik, 2018-02-19:
Ok.... meantime i read on some thread this problems it could be a symptom on a bluethoot hardware in the wii:
It seems that if the wii have a psychical hardware problems the start menu fail to boot and the console goes in error status called error -05- if i remember.

It's only need to do some test.....

DacoTaco, 2018-02-20:
its true a black screen could be the result of bluetooth module failure, but the error msg i never heard of. could be true though, idk. 
homebrew doesn't have this, and just keeps running. 

this is why i said to try and reinstall the same system menu version and instantly reinstall priiloader. 
incase reinstalling doesn't fix the issue, having priiloader will prevent a full brick.
replacing a wii's bluetooth module isn't hard btw, so incase if that is the issue.

DjDiabolik, 2018-02-20:
I'm not sure about error number... probably i confuse this on another console or other totally different device. Every days i check the life on so many pc/device/tablet and it's simply to make confusion :)

About other i have already all download on my pc.... the newest priiloader.. a wad manager and some files try to reinstall.
I thinks i need only to copy all in sd and see what's append...........

DacoTaco, 2018-02-20:
good luck and be careful with every step in the way ! ( get bootmii backup, install IOS of the system menu you will install, install system menu , go to homebrew channel, install priiloader , and reboot into priiloader. then see if SM boots! )

do you happen to know if the latest open homebrew channel can sync wiimotes? i didn't try that one yet

DjDiabolik, 2018-02-21:
I have right now this WII on my hands....

I have attached to my TV and try to turn it...... black screen.
I forced turned off................ whit the reset button i turned on.... priiloader 0.7.0 it's be showed up.
On the SD i have before copyed this:
- Priiloader 0.8.2 folder
- WAM manager
- and i have copyed the ios 80 and the system menu wad.

From priiloader 0.7.0 i have reach the HBC. I have load the WAM Manager and i have installed first the ios 80 and after the SystemMenu.
At this point from HBC i have try to back to "System Menu" and obtain black screen.

I have re-boot again the Priiloader 0.7.0 and again relaunch the HBC. From here i have updgraded the priiloader to latest 0.8.2.
Restart the wii and yes... Priiloader 0.8.2 it's works. I have again reach the HBC from priiloader and i have tryed again to reinstall ios 80 and systemmenu from WAM.

Apparently boot installation it's completed........ but now i have another big trouble.

I have tryed so many time to re-boot the priiloader..... i have tryed to:
- Press power and immediatelly after reset... nothings append
- Press reset and power togheter... nothigs append

I ever obtain black screen whit the led it's green.

Apparently i can't reach anymore the priiloader................. i have created a big paper clip ????

DacoTaco, 2018-02-23:
after installing system menu priiloader is indeed gone, no longer installed. this is why trying to boot it, it will not work and why i said to always install priiloader before booting system menu.
why priiloader was bootable after installing first time is beyond me. because it shouldve been removed. proves system menu didn't install properly though i guess.

anyway, from this we can conclude that a broken system menu was probably not the cause before.
this will not change now (unless IOS is causing the black screen now. but lets assume it isn't).
this means that a broken bluetooth module can be the cause of it, and replacing it can cause the wii to become bootable again like discussed before. there was a way to test this long long time ago, but i think it was removed in latest SM versions. had to do with a module team twiizers sold, but like i said, i think they patched it out.

a quick check on ebay shows they are still sold for around 10€ (with shipping to belgium), so to italy shouldn't be much more of an issue i think? 
no idea how confident you feel in replacing one though, but they aren't that hard to replace.

good luck and keep me posted.

DjDiabolik, 2018-02-23:
mmmmmmm.. it's interesting.
At this point i also things this not a system menu...... i have tryed to found some site sell the bluethoot module and the wifi module because i read both module when fail can create this issue whit system menu fail to boot.

I have another idea....... i have a my working wii. I can test to pick this module from my wii and install in this wii and if it's works my friends can rebuy the module and reinstall it on my wii.

It's a good alternative but i need to found someone whit experience about this procedure..... I want at least 99% sure can be re-obtain my full working wii :)

DacoTaco, 2018-02-27:
i too find it strange that priiloader was still there at first. hence why i think the system menu install must have failed and you didn't notice? i dont know, its weird. like i said, Priiloader is part of system menu. if system menu gets upgraded or rewritten priiloader gets removed as the system menu executable gets put back into place & the TMD gets reset, there is no way priiloader could still be there AND system menu being written from WAD (unless the WAD manager did something fancy to keep it there. but then why didn't it do that the second time?)

anyway, the re-installation of the bluetooth and wifi modules is imo not that hard. just a lot of screws to undo and being careful to have all screws removed before taking a part off.

you can follow the ifixit guide and skip part 9 as far as i know (no need to take the dvd drive apart. but it does need to be removed to access everything underneath it)
https://www.ifixit.com/Teardown/Nintendo+Wii+Teardown/812

however, despite it being not that hard, dont do this unless you feel confident. i would recommend not touching your wii unless you feel ok doing so, but it would be a good idea to test if it is in fact those modules screwing up without paying for anything yet.

DjDiabolik, 2018-02-27:
mmmmmmmmm.. problably for bluethoot module on ifixit there's this specific guide:
https://it.ifixit.com/Guide/Nintendo+Wii+Bluetooth+Board+Replacement/3456?lang=en

And problaly the most lowest price for a newest bluethoot chip:
https://www.chipspain.com/es/nintendo-wii/bluetooth-wireless-board-for-wii-new-2358.html

You thinks it's ok ? I can suggest this to my friends ?

DacoTaco, 2018-02-28:
ha, and the wifi module isn't that far away from the bluetooth either. same guide you linked can be used to get that one too.
the guide you linked seems to be better then mine too, a lot more obvious pictures

you can try, what do you have to loose besides a few euros and an experience to open a wii up?

DjDiabolik, 2018-02-28:
Yeah exactly......... problably my friends has already sold a newest bluethoot for the site linked above.

Now we need to wait....... at this point yes. Whit some patience i thinks it's not difficult to reach the bluethoot module and fortunately it is not soldered to the board directly.

Also a broken wi-fi module can create this not booting start menu ?

DacoTaco, 2018-03-07:
i doubt it. as far as i know only the bluetooth module creates a black screen.

good luck, and take it slow and easy, dont force anything.

also, if youre closing it up again, i have a tip.
inside the wii (not the casing) you can see arrows near the holes where a screw should be. no arrow is no screw. once you are putting the casing back this tip is no longer applied, as it talks about the metal casing inside.

DjDiabolik, 2018-03-08:
mmmmmm.. interesting :)

Otherwise i still to receive the bluethoot module... my friends has say to me have buy it about one week ago. Problably from spain....... it's to much long.

closed 2019-01-23 by DacoTaco