DacoTaco/priiloader #209: Cannot load HBC

Nemu64, 2018-01-23:
Priiloader can't seem to load HBC 1.1.2.
It suggests that it may not be installed.
(I know I'm a dip sometimes but come on)

About Wii:
Sys Menu: 4.0U
HBC 1.1.2 Channel ID LULZ (perhaps Priiloader doesn't recognize the new ID?)
Priiloader 0.8.2

DacoTaco, 2018-01-23:
the LULZ ID should be recognised :/ 

do you have more then 1 wiimote synced to the wii or a gamecube controller? 
this might sound unrelated, but im asking this wondering what you, or i, can do to fix the issue :)

DacoTaco, 2018-01-23:
can you boot it from the title menu btw? (instead from the main menu)

Nemu64, 2018-01-26:
I only have one Wiimote synced and my Wii is a family edition an therefore
doesn't have GameCube ports. It will load when selected in the title menu.
Having it autoboot to the hbc also fails. Hope this helps.

https://ci4.googleusercontent.com/proxy/b0OjHAZUD4Ac9JJ1HDwO6PCxdHTK4Bf9mK_zzemC-RG6JOpR8-TIrHqtTLUcMzbQyEqG672yQiy2C2tJpVS7-Ao2w9FWZNiHczAK-YGF6Q46wFGWyOgKFCe6ElP7ZeF9OihRfLhbSNHtu0kp7iUedcNE5eY7Hw=s0-d-e1-ft#https://github.com/notifications/beacon/AiEVLK7NwevYdYiBPgymZBTRiBbABBVcks5tNfkbgaJpZM4RpPTG.gif>

Nemu64, 2018-01-26:
I had Priiloader make a log file. Maybe it'll come in handy. Also made a mistake in first post. I'm running HBC 1.1.0.
[prii.log](https://github.com/DacoTaco/priiloader/files/1668708/prii.log)

Nemu64, 2018-01-27:
I had Priiloader create a new log. Not sure if it's relevant but when I selected HBC from the main menu the wii crashes, however when set to autoboot to HBC it simply shows an error message but doesn't freeze.

[prii.log](https://github.com/DacoTaco/priiloader/files/1669942/prii.log)

DacoTaco, 2018-01-27:
oh, that log doesnt look right. its "detecting" LULZ, but that one isn't even installed if i look at your previous log.im guessing thats why its failing...

this log comes from the autobooting im guessing? (since bootstate is 255, which is mostly seen when after a crash)

im currently trying to build a priiloader dol that you should run from HBC after disabling all autorun options in priiloader, but luck have it i can't compile after having to forcefully update all libraries xD

ill keep you posted

Nemu64, 2018-01-27:
Well, I decided to reinstall the HBC and now everything works as it should. Sorry to trouble you.

DacoTaco, 2018-01-27:
np, but i still want to have the bug fixed. ill test it on my own if i can. this is obviously not how it should be :')

Nemu64, 2018-01-27:
Good luck. And thanks for you patience. (and for not killing me) ;)

DacoTaco, 2018-02-11:
can't reproduce the issue so im not sure why it was happening. leaving open for a while in case somebody has the issue again

closed 2018-10-20 by DacoTaco