DacoTaco/priiloader #211: Entrypoint is unusable

Kirbeast, 2018-04-21:
When I install some dols, I can install them just fine, then when I try to boot installed file I get the error Entrypoint is unusable!. I tried the same dol on Priiloader v0.81 and it works without any errors. The dol was created with WIICRAZY's WIIGSC tool, here is the dol I'm trying to install

> https://drive.google.com/open?id=1ZP5eEN8lqQRCk4WrfKTfJXInTL1m1C8o

I would greatly appreciate your help, if I get this to work it may benefit many Project M players around the world

Kirbeast, 2018-04-21:
I tried doing a clean install already, I got Priiloader from the tool ModMii, maybe that source priiloader got messed up somehow. I would also like to know how to add titleIDs, I know how to compile programs with programmers notepad, but I can't get it to compile its a possibility that it is my version of devkitpro

Kirbeast, 2018-04-21:
Also after I try to load the installed file and get the error, it breaks going to the Homebrew channel and basically everything else

Kirbeast, 2018-04-22:
I tried it on version 0.7 r142 ios 236 mod version and booting the installed file works

Kirbeast, 2018-04-22:
Also when I try to load the binary it says failed to load binary

DacoTaco, 2018-04-22:
its entrypoint is not compatible with priiloader, even trying to load that file would crash priiloader as its overwriting priiloader. thats why you get the entrypoint error.
get the dol to be kickstarted or loaded into a different memory region.

also, i dont know what this "236 mod" is, but i dislike its name already. please use the official release found on wiibrew

Kirbeast, 2018-04-22:
Thank you for responding, that makes sense, I will use the latest version of Priiloader, sorry that I bothered you. Can I use doltool.exe to change the entry point address?

DacoTaco, 2018-04-23:
i dont think hex editing the entrypoint will make the program work. however, using a forwarder or kickstarter might work. 
included to this comment should be a link to some source code that shows how thats done (and can also be used to load your dol if you compile it with your dol included).
the source is found in kickstart\source

good luck!
(or, if youre thinkering with that code, add a kickstarter to priiloader that runs in mem2 and starts an executable in mem1. pretty sure thats how HBC does it)

http://upload.dacotaco.com/kickstart.zip

Kirbeast, 2018-04-23:
Okay, I don't know what to do with these files, what program do I use? How or what files I'm I supposed to open to use?

Kirbeast, 2018-04-23:
Okay I changed the directories to match my computers and now it will build with Programmers notepad, now I just have to figure out what to build and where to put my dol

Kirbeast, 2018-04-23:
So I put my dols binary in the data folder and named it swiss, then ran make in Programmers notepad, then this is the output I get, did I do it correctly? Do I need to listen to the warnings warning: implicit declaration of function 'memalign' [-Wimplicit-function-declaration]
`> "make" 
make -f Makefile.gc
make[1]: Entering directory '/opt/devkitpro/Projects/kickstart/kickstart'
swiss.bin
bin2s -a 32 /opt/devkitpro/Projects/kickstart/kickstart/data/swiss.bin | powerpc-eabi-as -o swiss.bin.o
echo "extern const u8" `(echo swiss.bin | sed -e 's/^\([0-9]\)/_\1/' | tr . _)`"_end[];" > `(echo swiss.bin | tr . _)`.h
echo "extern const u8" `(echo swiss.bin | sed -e 's/^\([0-9]\)/_\1/' | tr . _)`"[];" >> `(echo swiss.bin | tr . _)`.h
echo "extern const u32" `(echo swiss.bin | sed -e 's/^\([0-9]\)/_\1/' | tr . _)`_size";" >> `(echo swiss.bin | tr . _)`.h
main.cpp
C:/devkitPro/Projects/kickstart/kickstart/source/main.cpp: In function 's8 BootDolFromFat(FILE*, __argv*)':
C:/devkitPro/Projects/kickstart/kickstart/source/main.cpp:93:14: warning: 'entrypoint' may be used uninitialized in this function [-Wmaybe-uninitialized]
  entrypoint();
              ^
dvd.c
C:/devkitPro/Projects/kickstart/kickstart/source/dvd.c: In function 'DVDCleanUp':
C:/devkitPro/Projects/kickstart/kickstart/source/dvd.c:97:3: warning: implicit declaration of function 'IOS_Close' [-Wimplicit-function-declaration]
   IOS_Close(di_fd);
   ^~~~~~~~~
C:/devkitPro/Projects/kickstart/kickstart/source/dvd.c: In function 'DVD_DoCommand':
C:/devkitPro/Projects/kickstart/kickstart/source/dvd.c:133:11: warning: implicit declaration of function 'IOS_Open' [-Wimplicit-function-declaration]
   di_fd = IOS_Open("/dev/di",0);
           ^~~~~~~~
C:/devkitPro/Projects/kickstart/kickstart/source/dvd.c:137:17: warning: implicit declaration of function 'memalign' [-Wimplicit-function-declaration]
    inbuf = (u8*)memalign( 32, 0x20 );
                 ^~~~~~~~
C:/devkitPro/Projects/kickstart/kickstart/source/dvd.c:161:4: warning: implicit declaration of function 'IOS_Ioctl' [-Wimplicit-function-declaration]
    IOS_Ioctl( di_fd, command, inbuf, 0x20, outbuf, 0x20);
    ^~~~~~~~~
C:/devkitPro/Projects/kickstart/kickstart/source/dvd.c:168:4: warning: implicit declaration of function 'IOS_IoctlAsync' [-Wimplicit-function-declaration]
    IOS_IoctlAsync( di_fd, command, inbuf, 0x20, outbuf, 0x20, SetDriveState, NULL);
    ^~~~~~~~~~~~~~
linking ... kickstart_GC.elf
output ... kickstart_GC.dol
make[1]: Leaving directory '/opt/devkitpro/Projects/kickstart/kickstart'
make -f Makefile.wii
make[1]: Entering directory '/opt/devkitpro/Projects/kickstart/kickstart'
swiss.bin
bin2s -a 32 /opt/devkitpro/Projects/kickstart/kickstart/data/swiss.bin | powerpc-eabi-as -o swiss.bin.o
echo "extern const u8" `(echo swiss.bin | sed -e 's/^\([0-9]\)/_\1/' | tr . _)`"_end[];" > `(echo swiss.bin | tr . _)`.h
echo "extern const u8" `(echo swiss.bin | sed -e 's/^\([0-9]\)/_\1/' | tr . _)`"[];" >> `(echo swiss.bin | tr . _)`.h
echo "extern const u32" `(echo swiss.bin | sed -e 's/^\([0-9]\)/_\1/' | tr . _)`_size";" >> `(echo swiss.bin | tr . _)`.h
main.cpp
C:/devkitPro/Projects/kickstart/kickstart/source/main.cpp: In function 's8 BootDolFromFat(FILE*, __argv*)':
C:/devkitPro/Projects/kickstart/kickstart/source/main.cpp:93:14: warning: 'entrypoint' may be used uninitialized in this function [-Wmaybe-uninitialized]
  entrypoint();
              ^
dvd.c
C:/devkitPro/Projects/kickstart/kickstart/source/dvd.c: In function 'DVD_DoCommand':
C:/devkitPro/Projects/kickstart/kickstart/source/dvd.c:137:17: warning: implicit declaration of function 'memalign' [-Wimplicit-function-declaration]
    inbuf = (u8*)memalign( 32, 0x20 );
                 ^~~~~~~~
linking ... kickstart_WII.elf
output ... kickstart_WII.dol
#@wiiload ctr-gcs-DacoTaco-Edition_WII.dol
make[1]: Leaving directory '/opt/devkitpro/Projects/kickstart/kickstart'

> Process Exit Code: 0
> Time Taken: 00:03
`

DacoTaco, 2018-04-24:
looks likt eht build succeeded then. try starting the kickstart_wii.dol from priiloader and see if it works

Kirbeast, 2018-04-24:
It still throws an error entrypoint is unusable if I try to load its binary it says failed to load binary

DacoTaco, 2018-04-25:
mind sharing the kickstart_wii.dol ?
i just realised i might have made a mistake xD

how fast do you want a fix for this? cause i have some personal programming i need doing first but ive always wanted to know if it was possible to load a dol from MEM2... :')

Kirbeast, 2018-04-25:
Oof, yeah I'll share it, I would like a fix as soon as possible, so the wii has two memory places interesting. Thanks for your help, this will help out a lot of people.

[kickstart_WII.zip](https://github.com/DacoTaco/priiloader/files/1948308/kickstart_WII.zip)

Kirbeast, 2018-06-02:
I figured out a patch for my loader, I made a forwarder for my app with ModMii, The forwarder works with Priiloader autoboot, then I put the forwarder in my apps folder as a boot.dol, and converted my app to a boot.elf, homebrew and Priiloader look for the boot.dol before the boot.elf, so it is like the forwarder is not even there.

DacoTaco, 2019-12-18:
sorry for the late response,

but do you mind resharing the dol of the actual application?

with the recent changes in homebrew loading for version 0.9 ive been looking at issues with homebrew loading & entrypoints and seeing if much will change.

closed 2019-12-29 by DacoTaco