DacoTaco/priiloader #213: Press A to exit back to loader

KevinGA270, 2018-05-29:
[image]
[image]

hi, my friend wanted to update his priiloader
his wii is running on 4.1U
with priiloader installed V0.6 (r123) iosV60
he's been following all the step, put the Priiloader to app folder and using HBC,
but every time he press the priiloader V0.8 installer
its says "Press A to exit back to loader" then just nothing happened
any suggestion?
thanks!

DacoTaco, 2018-05-29:
normally there should be a message above the press A to exit line, but it was cut off by the TV.

thankfully i can see what is wrong from your first picture of priiloader.
his wii has been modded with a custom System Menu version and is not supported by priiloader. 

please find a guide to install a normal system menu version before updating priiloader. maybe look around on gbatemp.net. they made that custom system menu version.

KevinGA270, 2018-05-29:
okay thx for the info ;D
we'll find our way to install the right version of system channel

KevinGA270, 2018-05-29:
btw what version of wii sys menu that compatible with priiloader?

DacoTaco, 2018-05-30:
all system menu versions are compatible, but i suggest to install 4.1. its the same version as installed now (please verify in system menu!), safest because no new IOS install is needed

closed 2018-09-08 by DacoTaco

MarshTheBacca, 2022-12-31:
I'm having a similar issue. I'm running system menu version 4.1E, and using PriiLoader 0.9.1 on a USB, the following happens: Image 1 appears for a split second, followed by image two, which remains until I press A.

AndrewPiroli, 2022-12-31:
As per syscheck posted on Discord the installed system menu is 'System Menu 4.1E (v54450)'.
This is a modified System Menu and needs to be reinstalled.