DacoTaco/priiloader #214: Control selections via Power and Reset buttons

DongVGH, 2018-09-05:
I have recently gotten a Logitech Dual Action gamepad which i have been using with Nintendont in conjunction with Priiloader to load directly into it, not requiring any Wiimote at all.

I have also installed SRB2Wii and i'm working on getting gamepad support on it, and there's the block that i still need to use a Wiimote to boot into it without autoboot (which i already use for Nintendont), so i thought that it would be a great addition to make Power move through selections and have Reset select it, while holding Reset will be like pressing B, in that it will move you to the previous menu.

I know that i still need to use a Wiimote to update Priiloader but i'm ok with buying batteries one last time.

DacoTaco, 2018-09-06:
though i like the idea, i have a few questions about it.

i dont know how youre using the logitech controller on the wii but im guessing its through some sort of adaptor which connects to the wii ?
so in theory priiloader could (and might?) already pick up this input depending on how it sends its input to the wii.

does this adaptor function like a wii gamepad? or like a gamecube controller? is it connected straight to the usb port of the wii?

DacoTaco, 2018-09-16:
thats an interesting thing. i mean, it implemented its own HID driver to get gamepad input. awesome.

i should look into it but im kinda busy -_-
is the code big? does it patch ios or is it an actual HID stack/driver implementation?

DongVGH, 2018-09-16:
https://github.com/FIX94/Nintendont/tree/master/kernel Check it out here, it's named HID.h, HID.c and HID_Controllers.h I'm no expert but i'm fairly sure this code patches IOS 59, or it runs directly on the Starlet, because of course you can't run it on the same CPU that the GC games take up.

DacoTaco, 2018-10-20:
as much as i'd like to implement his HID driver, i just have to much frustrations with it right now.
he seems to redefine EVERY piece of code even the general C/C++ code that is in libogc or EVERY C COMPILER LIBRARY EVER

*sigh* ill keep you posted on my progress

DacoTaco, 2018-10-20:
say, would you happen to know anything about this project?
https://github.com/xerpi/libsicksaxis 
apparently a library that adds dualshock3 support as input, but from what i can see it could be tweaked to allow other stuff i think

DongVGH, 2018-11-29:
It's enough for me if you just make it so the Power and Reset buttons move the cursor and select, respectively.

Althrough if you can add USB-HID support to libsicksaxis that would be AWESOME!

niuus, 2019-01-22:
Hi DacoTaco, did you have a chance to check libsickaxis implementation?

DacoTaco, 2019-01-23:
seems my comment vanished.

i did check it out in a quick go. it works from what i saw but to change its implementation i would need to understand how HID devices work and i don't. im leaving this open as i want to eventually learn HID devices and how they work though. 
ive been thinking of implementing the power/reset button thing after all to help you guys out cause it could take a while ebfore i get off my ass to learn HID XD

DongVGH, 2019-01-24:
Yes, please implement the power and reset buttons that would be really nice until you do the HID support

DacoTaco, 2019-01-26:
ok, so ive started working on reworking the input and adding the power/reset button to this.

this will kill the 'press reset to boot SM' trick , but since you can +/- navigate i figured thats ok.
the power & reset button will only emulate a down or A button press. 

however, i was wondering if any of you wanted to test a build (all inputs in all menus) , and preferably have Bootmii as boot2 (just in case, because ill be updating devkitppc/libogc while im at it ).
this wont be an install build, just priiloader itself and will not replace any currently installed priiloader.

DacoTaco, 2019-01-26:
[priiloader.zip](https://github.com/DacoTaco/priiloader/files/2799046/priiloader.zip)

here is the dol if you want to try.

DongVGH, 2019-01-29:
Curious, will Priiloader auto-boot into it's own menu normally or i will need some other way to access it now?

DacoTaco, 2019-01-30:
ah right, i forgot to mention that part. you will need to have autoboot set to off/priiloader. otherwise booting the dol will execute the autoboot settings xD

abbcto, 2019-02-03:
@DacoTaco It is strange that this was not done with the release of RVL-101, which  does not contain GameCube ports, and with the loss of the synced remote it becomes impossible to unbrick console via priiloader.

DacoTaco, 2019-02-03:
ye, i know. and in my defence back then i didn't care as much anymore about priiloader. nor did i understand why you'd want that. and i still have issues with it cause right now there is no back button and i can't think of a way to implement a back button (cause eject is not an option atm nor is holding the button).

right now i just implemented it to solve the issue/RFC since i can't seem to find time to look into HID interfaces. and it wont solve the issue completely too since every HID device is different in some way. 

anyway, has the dol been confirmed working?

abbcto, 2019-02-03:
@DacoTaco tested a little. works as intended. it may make sense to make support for usb keyboards or some one highly common hid device. the back button can be done via timer, i.e after 10 seconds of inactivity, the back button is automatically activated.

DacoTaco, 2019-02-03:
that is actually not a bad idea to go around holding button not working imo. ill looking into it when i have time! 

did you try gamecube and wiimotes? classic controllers? power/reset? these are all handled by a separate 'class' in code now, so i'd love to see it all tested atm since it is a complete input overhaul xD 

for me it looked fine, but maybe i overlooked :) 

as for the keyboard. ye, if i get a HID driver for the wii working keyboard might be a good choice, but i would still need to learn and implement HID drivers >_<

abbcto, 2019-02-03:
@DacoTaco classic controller also works fine. don't push yourself hard, it's just emergency mode for those who can't use GameCube gamepad or remote. current power reset control realisation is fine enough for access to hbc or bootmii app.

DacoTaco, 2019-02-03:
in that case ill looking into the back button and create an installer for testing, and release beta afterwards

abbcto, 2019-02-03:
Possible to add menu entry for back on every screen. But this is an option for idealists, quite a lot of changes. Either pressing both buttons if possible at the same time. Either the sync button if possible. The best temporary option for me is the menu item to turn on the auto backward feature.

DacoTaco, 2019-02-28:
ive just made some adjustments so the reset doesn't instantly trigger priiloader to load the SM.

there is also a 8.3 beta 1 online that has these changes

Leseratte10, 2019-12-30:
Just saw this old issue, this seems to cause a small issue on my Wii sometimes: My TV takes a while to "notice" that there is an incoming video signal from the Wii (from Priiloader), and thus, when I hold RESET to get into Priiloader and wait until Priiloader is visible on my screen, and then release RESET (as I always did in the past), the 2 seconds are already over and as soon as I let go of the button, Priiloader boots into the System Menu. 

Can you make it so that it only starts reacting to RESET button presses once you've let go of RESET once? Or maybe require pressing POWER again once to "activate" POWER and RESET navigation? It's kinda annoying having to "guess" when Priiloader is loaded, if I let go too early I get into the System Menu, if I let go too late I get into the system menu, too, because Priiloader loads it.

DacoTaco, 2019-12-30:
personally, i expected it to happen. im really not a fan of the reset & power input devices.
i added it though to help these guys out.
at best those would need to be replaced with HID device support and reverted to the ways it was before this.

anyway, i can increase the delay that it takes for the reset to be accepted as input (currently set to 2s after input is init)

Leseratte10, 2019-12-30:
Yeah, increasing it to like 5 seconds, or always ignoring the first RESET press would probably work fine.

Or maybe make RESET navigate to other entries and POWER to select / activate them, then the worst thing that would happen would be that you move down one entry?

DacoTaco, 2020-01-01:
it has been changed to 5s.

i believe its the best choice atm.. :/

DacoTaco, 2020-01-05:
@Leseratte10 : please test https://github.com/DacoTaco/priiloader/tree/ResetFix with your monitor. it worked for perfectly fine, without losing an input button here (and is a lot cleaner then your PR code)

Leseratte10, 2020-01-05:
oh, I didn't know you could use that to check the RESET event. I'll give that a try tomorrow. If that works, it is indeed way cleaner than my solution.

DacoTaco, 2020-01-12:
closing this as the power/reset has been implemented.

the HID part can be discussed in #231

closed 2020-01-12 by DacoTaco

DacoTaco, 2020-06-26:
pinging @DongVGH to see if the support for usb controllers is still open for investigation.
if you , or anyone, could give some feedback on my testing app in #231