DacoTaco/priiloader #216: Error while building

ThatWeirdAndrew, 2019-03-05:
I am following the guide on building for windows. When I get to the actual building step, it doesn't work. Here is the log:

```
1>------ Build started: Project: Priiloader, Configuration: Release Win32 ------
1>Performing Makefile project actions
1>#../Shared/SubWCRev.exe "../." "../Shared/svnrev_template.h" "../Shared/svnrev.h"
1>../Shared/GitVersion.exe
1>fatal: Not a git repository (or any of the parent directories): .git
1>fatal: Not a git repository (or any of the parent directories): .git
1>generated versions :  and 
1>stub.bin
1>bin2s -a 32 /c/Priiloader/trunk/priiloader/data/stub.bin | powerpc-eabi-as -o stub.bin.o
1>echo "extern const u8" `(echo stub.bin | sed -e 's/^\([0-9]\)/_\1/' -e 's/[^A-Za-z0-9_]/_/g')`"_end[];" > `(echo stub.bin | tr . _)`.h
1>echo "extern const u8" `(echo stub.bin | sed -e 's/^\([0-9]\)/_\1/' -e 's/[^A-Za-z0-9_]/_/g')`"[];" >> `(echo stub.bin | tr . _)`.h
1>echo "extern const u32" `(echo stub.bin | sed -e 's/^\([0-9]\)/_\1/' -e 's/[^A-Za-z0-9_]/_/g')`_size";" >> `(echo stub.bin | tr . _)`.h
1>make[1]: Warning: File 'certs.bin' has modification time 1180 s in the future
1>certs.bin
1>bin2s -a 32 /c/Priiloader/trunk/priiloader/data/certs.bin | powerpc-eabi-as -o certs.bin.o
1>echo "extern const u8" `(echo certs.bin | sed -e 's/^\([0-9]\)/_\1/' -e 's/[^A-Za-z0-9_]/_/g')`"_end[];" > `(echo certs.bin | tr . _)`.h
1>echo "extern const u8" `(echo certs.bin | sed -e 's/^\([0-9]\)/_\1/' -e 's/[^A-Za-z0-9_]/_/g')`"[];" >> `(echo certs.bin | tr . _)`.h
1>echo "extern const u32" `(echo certs.bin | sed -e 's/^\([0-9]\)/_\1/' -e 's/[^A-Za-z0-9_]/_/g')`_size";" >> `(echo certs.bin | tr . _)`.h
1>main.cpp
1>C:/Priiloader/trunk/priiloader/source/main.cpp: In function 'void BootMainSysMenu(u8)':
1>C:/Priiloader/trunk/priiloader/source/main.cpp:1537:9: warning: variable 'entrypoint' set but not used [-Wunused-but-set-variable]
1>  void (*entrypoint)();
1>         ^~~~~~~~~~
1>In file included from C:/Priiloader/trunk/priiloader/source/main.cpp:50:
1>C:/Priiloader/trunk/priiloader/source/main.cpp: In function 'int main(int, char**)':
1>c:\priiloader\trunk\shared\gitrev.h:2:21: error: 'ECHO' was not declared in this scope
1> #define GIT_REV_STR ECHO is off.
1>                     ^~~~
1>C:/Priiloader/trunk/priiloader/source/main.cpp:3479:95: note: in expansion of macro 'GIT_REV_STR'
1>  gprintf("Version : %d.%d.%d (rev %s)\n", VERSION>>8, (VERSION&0xFF) / 10,(VERSION&0xFF) % 10,GIT_REV_STR);//VERSION>>16, VERSION&0xFFFF, SVN_REV_STR);
1>                                                                                               ^~~~~~~~~~~
1>c:\priiloader\trunk\shared\gitrev.h:2:21: note: suggested alternative: 'EIO'
1> #define GIT_REV_STR ECHO is off.
1>                     ^~~~
1>C:/Priiloader/trunk/priiloader/source/main.cpp:3479:95: note: in expansion of macro 'GIT_REV_STR'
1>  gprintf("Version : %d.%d.%d (rev %s)\n", VERSION>>8, (VERSION&0xFF) / 10,(VERSION&0xFF) % 10,GIT_REV_STR);//VERSION>>16, VERSION&0xFFFF, SVN_REV_STR);
1>                                                                                               ^~~~~~~~~~~
1>make[1]: *** [/opt/devkitpro/devkitPPC/base_rules:13: main.o] Error 1
1>make: *** [Makefile:109: build] Error 2
1>Project : error PRJ0019: A tool returned an error code from "Performing Makefile project actions"
1>Build log was saved at "file://c:\Priiloader\trunk\priiloader\Release\BuildLog.htm"
1>Priiloader - 2 error(s), 1 warning(s)
2>------ Build started: Project: Installer, Configuration: Release Win32 ------
2>Performing Makefile project actions
2>make[1]: Warning: File 'certs.bin' has modification time 1177 s in the future
2>certs.bin
2>bin2s -a 32 /c/Priiloader/trunk/Installer/data/certs.bin | powerpc-eabi-as -o certs.bin.o
2>echo "extern const u8" `(echo certs.bin | sed -e 's/^\([0-9]\)/_\1/' -e 's/[^A-Za-z0-9_]/_/g')`"_end[];" > `(echo certs.bin | tr . _)`.h
2>echo "extern const u8" `(echo certs.bin | sed -e 's/^\([0-9]\)/_\1/' -e 's/[^A-Za-z0-9_]/_/g')`"[];" >> `(echo certs.bin | tr . _)`.h
2>echo "extern const u32" `(echo certs.bin | sed -e 's/^\([0-9]\)/_\1/' -e 's/[^A-Za-z0-9_]/_/g')`_size";" >> `(echo certs.bin | tr . _)`.h
2>main.cpp
2>C:/Priiloader/trunk/Installer/source/main.cpp:59:10: fatal error: priiloader_app.h: No such file or directory
2> #include "priiloader_app.h"
2>          ^~~~~~~~~~~~~~~~~~
2>compilation terminated.
2>make[1]: *** [/opt/devkitpro/devkitPPC/base_rules:13: main.o] Error 1
2>make: *** [Makefile:106: build] Error 2
2>Project : error PRJ0019: A tool returned an error code from "Performing Makefile project actions"
2>Build log was saved at "file://c:\Priiloader\trunk\Installer\Release\BuildLog.htm"
2>Installer - 2 error(s), 0 warning(s)
========== Build: 0 succeeded, 2 failed, 0 up-to-date, 0 skipped ==========

```

DacoTaco, 2019-03-06:
how did you acquire the source? through git or the zip function of github?

ThatWeirdAndrew, 2019-03-06:
I used SVN. I tried using git and I got the error:

```
1>------ Build started: Project: Priiloader, Configuration: Release Win32 ------
1>Performing Makefile project actions
1>#../Shared/SubWCRev.exe "../." "../Shared/svnrev_template.h" "../Shared/svnrev.h"
1>../Shared/GitVersion.exe
1>generated versions : 0xdf0a869 and "df0a869-dirty"
1>main.cpp
1>C:/Users/andre/Downloads/priiloader/priiloader/source/main.cpp: In function 'void BootMainSysMenu(u8)':
1>C:/Users/andre/Downloads/priiloader/priiloader/source/main.cpp:1537:9: warning: variable 'entrypoint' set but not used [-Wunused-but-set-variable]
1>  void (*entrypoint)();
1>         ^~~~~~~~~~
1>C:/Users/andre/Downloads/priiloader/priiloader/source/main.cpp: In function 'void InstallLoadDOL()':
1>C:/Users/andre/Downloads/priiloader/priiloader/source/main.cpp:1982:23: warning: '%s' directive writing up to 4095 bytes into a region of size 4086 [-Wformat-overflow=]
1>      sprintf(filepath,"fat:/apps/%s/boot.dol",filename);
1>                       ^~~~~~~~~~~~~~~~~~~~~~~ ~~~~~~~~
1>C:/Users/andre/Downloads/priiloader/priiloader/source/main.cpp:1982:13: note: 'sprintf' output between 20 and 4115 bytes into a destination of size 4096
1>      sprintf(filepath,"fat:/apps/%s/boot.dol",filename);
1>      ~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
1>C:/Users/andre/Downloads/priiloader/priiloader/source/main.cpp:1987:24: warning: '%s' directive writing up to 4095 bytes into a region of size 4086 [-Wformat-overflow=]
1>       sprintf(filepath,"fat:/apps/%s/boot.elf",filename);
1>                        ^~~~~~~~~~~~~~~~~~~~~~~ ~~~~~~~~
1>C:/Users/andre/Downloads/priiloader/priiloader/source/main.cpp:1987:14: note: 'sprintf' output between 20 and 4115 bytes into a destination of size 4096
1>       sprintf(filepath,"fat:/apps/%s/boot.elf",filename);
1>       ~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
1>C:/Users/andre/Downloads/priiloader/priiloader/source/main.cpp:1997:23: warning: '%s' directive writing up to 4095 bytes into a region of size 4086 [-Wformat-overflow=]
1>      sprintf(filepath,"fat:/apps/%s/meta.xml",filename);
1>                       ^~~~~~~~~~~~~~~~~~~~~~~ ~~~~~~~~
1>C:/Users/andre/Downloads/priiloader/priiloader/source/main.cpp:1997:13: note: 'sprintf' output between 20 and 4115 bytes into a destination of size 4096
1>      sprintf(filepath,"fat:/apps/%s/meta.xml",filename);
1>      ~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
1>C:/Users/andre/Downloads/priiloader/priiloader/source/main.cpp:2125:24: warning: '%s' directive writing up to 4095 bytes into a region of size 4091 [-Wformat-overflow=]
1>       sprintf(filepath,"fat:/%s",filename);
1>                        ^~~~~~~~~ ~~~~~~~~
1>C:/Users/andre/Downloads/priiloader/priiloader/source/main.cpp:2125:14: note: 'sprintf' output between 6 and 4101 bytes into a destination of size 4096
1>       sprintf(filepath,"fat:/%s",filename);
1>       ~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
1>HTTP_Parser.cpp
1>C:/Users/andre/Downloads/priiloader/priiloader/source/HTTP_Parser.cpp: In function 's32 GetHTTPFile(const char*, const char*, u8*&, int*)':
1>C:/Users/andre/Downloads/priiloader/priiloader/source/HTTP_Parser.cpp:175:16: warning: 'char* strncpy(char*, const char*, size_t)' output truncated before terminating nul copying as many bytes from a string as its length [-Wstringop-truncation]
1>         strncpy(host_new,buffer,strlen(buffer));
1>         ~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
1>C:/Users/andre/Downloads/priiloader/priiloader/source/HTTP_Parser.cpp:170:16: warning: 'char* strncpy(char*, const char*, size_t)' output truncated before terminating nul copying as many bytes from a string as its length [-Wstringop-truncation]
1>         strncpy(file_new,&buffer[phost-buffer],strlen(&buffer[phost-buffer]));
1>         ~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
1>linking ... priiloader.elf
1>output ... priiloader.dol
1>./OpenDolBoot.exe /home/andre/Downloads/priiloader/priiloader/priiloader.dol ../Installer/data/priiloader.app
1>DacoTaco and _Dax_ 's OpenDolboot : C/C++ Version 0.1
1>opening dol...
1>opened
1>reading data Dol info (header and data)...
1>done reading data
1>starting nand loader injection...
1>checking & editing data before saving...done!
1>Saving to output.app...
1>DONE! check ../Installer/data/priiloader.app to verify data!
1>wiiload /home/andre/Downloads/priiloader/priiloader/priiloader.elf
1>wiiload v0.5.1
1>coded by dhewg
1>set the environment variable WIILOAD to a valid destination.
1>examples:
1> usbgecko mode:
1>  set WIILOAD=COM4
1> tcp mode:
1>  set WIILOAD=tcp:wii
1>  set WIILOAD=tcp:192.168.0.30
1>usage:
1> C:\devkitPro\tools\bin\wiiload.exe <filename> <application arguments>
[... 19 lines cut ...]
2>                        ^
2>C:/Users/andre/Downloads/priiloader/Installer/source/main.cpp:357:27: warning: 'memset' used with length equal to number of elements without multiplication by element size [-Wmemset-elt-size]
2>  memset(FileHash_D2,0xFF,5); //place different data in D2 so that if something goes wrong later on, the comparison will fail
2>                           ^
2>linking ... Installer.elf
2>output ... Installer.dol
2>wiiload /home/andre/Downloads/priiloader/Installer/Installer.dol
2>wiiload v0.5.1
2>coded by dhewg
2>set the environment variable WIILOAD to a valid destination.
2>examples:
2> usbgecko mode:
2>  set WIILOAD=COM4
2> tcp mode:
2>  set WIILOAD=tcp:wii
2>  set WIILOAD=tcp:192.168.0.30
2>usage:
2> C:\devkitPro\tools\bin\wiiload.exe <filename> <application arguments>
2>make: *** [Makefile:107: build] Error 1
2>Project : error PRJ0019: A tool returned an error code from "Performing Makefile project actions"
2>Build log was saved at "file://c:\Users\andre\Downloads\priiloader\Installer\Release\BuildLog.htm"
2>Installer - 1 error(s), 3 warning(s)
========== Build: 0 succeeded, 2 failed, 0 up-to-date, 0 skipped ==========

```

DacoTaco, 2019-03-07:
svn still works?

thats euh, a big suprise xD

anyway, looks like i build correctly.
the error's youre getting is juist the build script trying to send the build dol to the wii/homebrew channel using wiiload. you should have a priiloader.dol & installer.dol now.

few things i need to change or look at though, now that i look at this : 
- don't error when failing to send dol
- generate random git hash when source is downloaded not through git

DacoTaco, 2019-03-07:
both points have now been implemented

ThatWeirdAndrew, 2019-03-07:
Thanks, it works great now!

closed 2019-03-07 by ThatWeirdAndrew

DacoTaco, 2019-03-08:
np, good to hear it works now