DacoTaco/priiloader #218: hacks_hash entry with absolute offset?

Leseratte10, 2019-11-10:
Is there a way to add a patch to the hacks_hash.ini that uses an absolute offset? Meaning, I don't want to search some string A and replace it with another string B, I want to write the string A to some fixed memory offset. 

The place I'd like to write to using the hacks_hash patch is in the middle of a large section of unused RAM, so A) I have no idea what content will be there before and B) even if I knew, it would most likely be a bunch of nullbytes so it would be useless. 

Looking at the code I only found support for the "hash" and "patch" syntax, would it be possible to add a feature that lets me provide an "offset" and "patch" that will be patched?

Leseratte10, 2019-11-10:
Just saw that support for that was included in the past, but has been removed with a6695871346d2bc257d14b8922bc0623119085c6. Was there a particular reason for that? 

I'd like to write a priiloader hack that adds a feature to the system menu. For that, I'd need to write a bunch of custom code to an unused place in memory, and then add a few branches from the system menu code to my custom code. That would be difficult or even impossible with the hacks_hash format. 

Would it be possible to bring the old variant back, maybe without a second file but just with a parameter "offset" instead of "hash" in the hacks_hash.ini?

DacoTaco, 2019-11-10:
the old hacks system used static addresses yes, which would allow what you are doing but made it so a different set of addresses was required for every version of SM and all regions. this is why i made the changes hoping to simplify things a bit (but lets be honest, it created more issues i think...)

however, this indeed won't work for your idea. i like your idea though, and can i ask what youre planning to make, cause im curious :P 

the offset idea isn't bad and i need to look into it to re-add it. sadly my wii isn't unpacked yet but i'd like to add this ye

Leseratte10, 2019-11-10:
Yeah, I get that using the hash method simplifies the hacks because you no longer need to port them to every version & region for new system menu versions. 

I am trying to write a hacks entry that patches the system menu so that when the system menu boots a game, it automatically patches the game for Wiimmfi. That would mean that if you have Priiloader installed and that hack enabled, you could just boot your games through the disc channel and could play online on Wiimmfi. 

I already got the necessary offsets from the System menu (and I already have lots of experience porting offsets from all my Mario Kart modding so it wouldn't be a problem for me to port them to a bunch of other versions), all that's missing would be a way for Priilloader to load my hacks_hash.ini. 

Also, is there a size limit for one hacks entry? How much data can I add (assuming I have enough RAM at the location where I want to add it)?

DacoTaco, 2019-11-10:
oh im loving that idea and its purpose.

i just took a look at the current hacks code and it makes me puke atm.
i think a rework is in order, allow for a hash & offset 
ill try and look into it in the following couple of days

DacoTaco, 2019-11-11:
could you design the hack for me to see what to look out for in the new code?
i'd design the patch format to be the same, but have the hash section be replaced with 'offsets'

right now in particular im looking into how long the offsets&patch section could become

Leseratte10, 2019-11-11:
I don't have the patch ready yet, and it'd be difficult for me to create it without a way to test it during development, so I can't really say how large the patch section would be.

The "offset" parameter would only take one 32-bit value (the address to write to), and the "patch" parameter could become quite long as all the wiimmfi patch code would need to be put into there. Although if there is a size limit for each patch it could also be splitted into multiple offset-patch pairs. 

But the Priiloader should have lots of RAM to spare, given that only the Priiloader and parts of the system menu are in RAM and not a whole game DOL, right? What would be the maximum size of the patch the Priiloader could support?

DacoTaco, 2019-11-11:
technically i have the whole of mem2 to use (i use mem2 when allocating memory) but it's not the memory im worried about.

im thinking of how im going to read the file. in general C/C++ there are functions that do this for you but not in libogc afaik when reading from the NAND.
this means that i need to manually read a string of a certain lenght and then find the newline character(s). 
ill make something and we'll see if its good enough, no worries.
it is indeed an idea to split the patch in multiple parts if it isn't good enough, but ill make something to test before we go into problem fixing :)

Leseratte10, 2019-11-11:
I would just make it read another block and append it when it didn't find a newline in the first block. Then the block size wouldn't matter as it'd just read block after block until it finally finds a newline.

Leseratte10, 2019-11-23:
Any news about that change yet?

DacoTaco, 2019-11-23:
you might have missed it, but im working on it. 
see my branch for the progress
https://github.com/DacoTaco/priiloader/tree/Hacks_rework

as soon as that is done i will poke you with the installer to test.

at this point im still working on loading the hacks from the ini. 
i can load the file line by line but i still need to write the code to process everything.

Leseratte10, 2019-11-23:
Oh, cool, I didn't notice that branch, thanks for the link. Looking forward to testing it when it's done.

DacoTaco, 2019-11-24:
question : are you capable of building priiloader yourself or do you want a dol of the installer to put on your SD card?

cause i think, right now, the hacks rework should be done and all i need to do is add the offset check to the system menu patching.
meaning that testing can begin to see if nothing broke with the old hacks

Leseratte10, 2019-11-24:
I can try compiling it myself. What version of devkitPPC and libogc are you using / do I need to compile it?

DacoTaco, 2019-11-24:
normally latest devkitppc/libogc should work.

i've also just added offset's to the hack applying in theory (i still need to confirm). 

basically they exist off an offset line and a patch line of what it needs to write starting at the offset address
example : 
[Offset Test]
maxversion=1
minversion=1
offset=0x8140a464
patch=0x41424344,0x41424345,0x41424346,0x41424347

DacoTaco, 2019-11-24:
ah ye, linux builds are broken because i have no idea how to get and save the GIT revision using linux. for windows i wrote a tool that runs a script to save it but linux was harder to do...

anyway , 
http://upload.dacotaco.com/boot_hacks_rework.dol
this dol should install priiloader 0.8.3 beta 2 which is made from this branch

Leseratte10, 2019-11-24:
Just installed it. Wii boots normally into the System Menü, and when I hold RESET I get into the Priiloader. However when I try to boot the system menu from Priiloader, I get a DSI exception. (SRR0 81099288, priiloader's trying to load something relative to r9, and r9 is 0x1a948a48 instead of a valid address).

Or does that happen because I don't have a hacks_hash.ini yet?

DacoTaco, 2019-11-24:
ye, im noticing some odd crashes as well. im trying to debug them but atm they aren't making much sense.

you don't have a hacks_hash.ini? not on SD nor on the wii? odd, for me its crashing because its loading the ini...

Leseratte10, 2019-11-24:
Uh, I might have an old one on the Wii. Don't know if the installer removes that. But I don't have one on the SD. 

But if it was crashing while reading the hacks_hash, wouldn't it crash on autoboot (when starting the wii and not entering priiloader) as well?

DacoTaco, 2019-11-24:
it probably will as its the same code running at that point. i hope this isn't a stack/heap issue because of the current code using up more memory

Leseratte10, 2019-11-24:
Well for me it doesn't crash on autoboot. Only when I manually select "System menu". And the DSI exception doesn't look like it ran out of heap / stack space, although I might be wrong, just looking at it without having the corresponding listing.

DacoTaco, 2019-11-24:
well something is up though. as a test i changed the entrypoint of priiloader and at that point it worked fine. 
on my side it crashes as soon as 4 bytes of a hash or patch are added to the list. 
its been to long since ive had to debug a DSI code dump, and i totally forgot how to read them xD

Leseratte10, 2019-11-24:
Basically, "SRR0" is the address the Wii tried to execute (and failed), and the three lines at "CODE DUMP" are the assembly code the console tried to execute. In my case, SRR0 is 81099288, and the code on that offset is 81490004, which translates to `lwz r10, 4(r9)` (load the 32-bit value at `r9+4` into r10). And r9 is not a valid address (but 1a948a48), thus causing the DSI exception. 

If you have the debug listing (assembly disassembler listing with all the function names and variable names present) it should be fairly easy to find the C code line that corresponds to that RAM address.

DacoTaco, 2019-11-24:
ye i know, and with the gcc tools you can basically check where in code it was when the exception happened (using the stack trace and code dump).

still, its hard to figure out what is happening...

DacoTaco, 2019-12-05:
i have come across a fun thing.

in my current build (which has changes since last build i shared here) it seems to crash if the SD is inserted, and boots if its not.
could you verify if im not having a different issue at the moment?

Leseratte10, 2019-12-05:
Can confirm. Autobooting to the System menu works all the time, but starting the system menu from the Priiloader only works when I don't have a USB drive or SD card connected / inserted.

DacoTaco, 2019-12-05:
i had thought i had fixed it untill i started playing around with SD settings, different hacks file etc etc...
but it seems ive come a step closer
im guessing having either of those pushes the stack usage juuuuust far enough to kill itself. 
but i currently have no method of fully checking that out (running and debugging on dolphin)

DacoTaco, 2019-12-05:
i just noticed that while loading in system menu it writes some system menu data to 0x8132FFE0 
priiloader its entrypoint is at 0x81000000 , so its highly likely that this write is just completely destroying priiloader. without that write its working fine too.

somehow, i have a feeling its related to #177 ...

Leseratte10, 2019-12-06:
Yeah, 0x8132ffe0 is the RAM address where the D0 section (data section) of the system menu begins: 

---

    $ wit dump sysmenu-v450-41e-08.dol 
    
    Dump of file sysmenu-v450-41e-08.dol
    
      Real path:         /home/florian/wii/priiloader-wiimmfi/sysmenu-v450-41e-08.dol
      File size:            3b0500/hex =    3867904 = 3777 KiB
      File type:         DOL
    
      Memory map of DOL file:
    
               unused :  off(beg) ..  off(end) :      size : section
         -----------------------------------------------------------------------
                      :         0 ..       100 :       100 : DOL header
                      :       100 ..       500 :       400 : text section T0
                      :       500 ..    3b0500 :    3b0000 : data section D0
                      :    3b0500 ..    3b0500 :         0 : --- end of file ---
    
      Memory map of DOL image:
    
               unused :  off(beg) ..  off(end) :      size : section
         -------------------------------------------------------------------
                      :      3400 ..      3400 :         0 : entry point
             80000000 :  80003400 ..  80003800 :       400 : text section T0
         .!   132c7e0 :  8132ffe0 ..  816dffe0 :    3b0000 : data section D0
         !.           :  81340000 ..  81350000 :     10000 : bss section

---

That means, the system menu has its executable code between 80003400 and 80003800, and has its data section between 8132ffe0 and 816dffe0. If you take the Priiloader DOL (not the installer, the actual DOL) and run `wit dump file.dol` on it, you can check the offsets for the priiloader. It should be compiled in a way that none of the Priiloader code or data is within that memory space (or shortly after it, as other versions of the system menu might be larger, plus, that would be the space I wanted to place my custom code).

DacoTaco, 2019-12-06:
ye, i saw in the dol header of system menu too that it was the data section.

Priiloader has an entrypoint of 0x81000000 which is below the designated address for loaders ( 0x81300000) since it wants to be nowhere near system menu.
its text and data sections aren't anywhere near this address though ( data section ends @ 0x810F5A40 ), but that doesn't mean its memory heap isn't.
the issue was hidden until i started rewriting code and made the heap bigger . thats why i think this is where the issue lies

system menu, like a normal nand application, has 2 stages. 1st stage is the nand code (entrypoint 0x80003400) and the second stage is, im guessing , the data section that contains the actual system menu.

i once lowered it to 0x80E00000 since i was having issues (i don't know the details of it, its was years ago) but killed a lot of forwarders and stuff with it, so i started using mem2 for memory allocations and reverted.

since Priiloader is so high in memory it doesn't have any actual issues with loading anything (besides loaders) as by default apps have entrypoints of 0x80003400 or 0x80003F00. nowhere near Priiloader.
loaders usually work by having an executable at 0x80003400 and loading their actual executable much higher in memory, Priiloader doesn't have this at this point as it would increase the file size significantly and take more space on the nand and i didn't see a point to add it, untill now.
(and i think homebrew channel also didn't load any binaries with those high entrypoints back in the day, but don't quote me on that)

Currently, Priiloader checks if the entrypoint or any section is not in Priiloader's way and refuses to boot those if they are. However, for system menu there are no such checks since booting system menu is essential...

ill have to rethink my strategy in loading executable's ye...
i want to keep it as small as possible as priiloader needs to be as small as possible too.
currently im pitching the idea in my head of loading the executable to mem2 , and writing a small app that loads that app. however, im thinking of maybe trying to write it in asm to keep it small since it shouldn't be rocket science. i haven't written any working asm in my life though...

DacoTaco, 2019-12-08:
heads up : ive started rewriting loading binaries. ive created a loader which priiloader will load into mem2 together with the binary and potential parameters. after that it will launch said code and the code will write the binary to the addresses it needs to.

this will hopefully fix everything. first test (priiloaders installer) was launched successfully

DacoTaco, 2019-12-11:
hey,

if you, or anyone, has time and wants to; could they try and install this priiloader. it includes the new loader. it should support all previous hacks files , and offset is not yet implemented. im going to implement offset now, but this needs to already work.

what has changed : 
- loading system menu (with the new hacks loading)
- loading dol/elf from the menu
- loading installed dol/elf
- loading dol from update server (can't be tested atm but it uses the same code as all the rest)

Leseratte10, 2019-12-11:
Sure, I'll try that version later or tomorrow. So that includes the new loader that should be able to load any DOL independantly of the memory regions?

DacoTaco, 2019-12-11:
correct, it contains the new loader. this loader should be able to load any dol that doesn't want to be fancy and has an entrypoint in mem2 (which afaik is none)

Leseratte10, 2019-12-13:
Just installed it. When I boot the console normally it boots into the system menu, and when I load the system menu from Priiloader it loads as well. I currently don't have any hacks loaded yet. 

However I noticed two things: 

- It takes multiple tries to get into Priiloader, as most of the time, Priiloader interprets me letting go of the RESET button as a signal to boot the system menu - meaning, the Wii is off, I press and hold RESET, I press POWER, I wait for Priiloader to show up, and as soon as I let go of RESET, it boots the system menu
- Once, when booting the system menu from priiloader, it froze. Can't reproduce again.

Leseratte10, 2019-12-13:
I now have both a hacks_hash.ini in /apps/priiloader, and a hackshas.ini in /title/00000001/00000002/data, but it recognizes neither and complains that the hacks_hash.ini is missing.

DacoTaco, 2019-12-13:
do you happen to have an usb gecko?

if not, enable Dump Gecko output in the settings and try opening the menu again. it should write the error msg to fat:/prii.log

Leseratte10, 2019-12-13:
There's a bunch of `----gecko_output_enabled-----`, and this error message: 
´LoadHacks : hacksh_s(states) could not be created. ISFS_CreateFile error -105`
Apparently, -105 means "file exists".

DacoTaco, 2019-12-13:
priiloader should indeed be able to handle that. ill also look into why its trying to recreate that file

DacoTaco, 2019-12-14:
i have come up with a scenario that could possibly have created the situation. did you open the hacks menu , then adjust the hacks on the sd and then reopen the menu?

this could have caused a size indifference with the states and create the error.

i added a ISFS_Delete to delete the file before creating it to kill the issue i just described.

Leseratte10, 2019-12-14:
That is possible, I don't remember exactly what I did.

DacoTaco, 2019-12-21:
ive updated and changed a few things, and updated the link to a newer build ( http://upload.dacotaco.com/boot_hacks_rework.dol )

changes : 
* fixed the issue you were having
* fixed the loader (its stack was still in mem1 , causing the loading of System Menu & some dols to crash the wii )
* made the dol smaller again by dropping the use of some std functions and doing it myself.

i think it should be good now (i hope) so i can finally start working on implementation of applying the offset hacks ...

DacoTaco, 2019-12-21:
correction, the link now installs a priiloader that also applied offset patches.
it loads the patches into memory (mem2) and the loader will apply them after it has loaded system menu into mem1
you can go full out in testing this (including what youre trying to do)

Leseratte10, 2019-12-22:
For some reason I can't save any hacks in this version. I have the hacks_hash.ini (with my new "test" patch) on the SD card, and when I go to the hacks menu the hacks all show up, but no matter which hacks I enable, I always get "saving failed". 

When I look in the "prii.log" file, it is trying to do a bunch of stuff with the "old" hacks (lots of lines starting with "testing" and then the old hack names that I used before with another fork of Priiloader), not the ones currently in the hacks_hash.ini on my SD, and then below that it says `hacksh_s.ini save error -111(5)`

DacoTaco, 2019-12-22:
crap, 
i had this issue as well, but i had figured during dev i created a file that was fucked up so i had deleted the file and it worked fine after that (still does to this day).

can you share the log? im curious what you mean with testing (left over from previous priiloader version?)

thing is, with the ISFS_Open i verify that it exists and i have access to read and write to it, and ISFS_GetFileStats also needs a file descriptor so i need to open the file none the less.

ive also reuploaded (same link) a version that logs the fd value & the return value of that ISFS_Close. maybe it did fail, but it would be a first time i see that fail...

Leseratte10, 2019-12-22:
Oh, I thought ISFS_GetFileStats would take a path as input, not a file descriptor. 
What I meant was that I previously had a fork of Priiloader installed that was translated into german (including the hacks names) and the prii.log of the current version still references these, so it seems to still read the old hacks_hash.ini from NAND. 

These are the entries I was talking about: 

    19:48:44 : testing Keine Hintergrundmusik im Wii-MenÃŒ
    19:48:44 : testing Disk-Channel verschieben
    19:48:44 : testing Recovery MenÃŒ beim Starten
    19:48:44 : testing Blockiere Updates im Recovery MenÃŒ
    19:48:44 : testing Entferne Kopierschutz
    19:48:44 : testing Wii startet mit schwarzem Bildschirm

But given the time stamp they might as well be from an earlier run of Priiloader. I'll reinstall the new version and check again.

Leseratte10, 2019-12-22:
The new version still prints "saving failed". 

    hacksh_s.ini fd : 8
    hacksh_s.ini ISFS_Close : 0
    hacksh_s.ini save error -111(5)

DacoTaco, 2019-12-22:
it will still fail. it can't create it, cause it exists (check your nand, its there) and it'll fail to delete it like it does now... 

i think i found the issue though.
the states file used to have 4 bytes per hack to indicate if it was enabled or not. i changed this to 1 byte, reducing the file in size.
however, when loading the hacks that size is checked and if its not the same, the file is ignored but its file descriptor never closed. 

i changed this to close the descriptor if its still open at the end of loading hacks.
this also explained why once i deleted the file on my wii the problem was solved.

i updated the link/dol

Leseratte10, 2019-12-22:
There is a fork that has been created by "BöhserOnkel" a long time ago (last update 2012): https://forum.wii-homebrew.com/index.php/Thread/40755-Priiloader-German-Mod-Installieren/ 

Quite outdated (and no source for that is available, and I was unable to contact the developer to get the source code). I just had that on my Wii until recently, because for me there was no reason to update. 

I'll test that new version in a couple minutes.

Leseratte10, 2019-12-22:
Just tested the new version, seems to work properly. Hacks like "move disc channel" that I could quickly test work properly, and I just enabled a very short hack with the "offset" function that seemed to work, too. Thanks a lot, now I can start working on the Wiimmfi patch. 

One more thing I noted is that once you have more than 19 hacks in your hacks_hash.ini, the display becomes "weird" because the "save" and "saving" texts overwrite the hack names and you can't read all of them. Maybe (if the number of hacks will increase even more in the future) it might be a good idea to add some scrolling support to that page.

DacoTaco, 2019-12-22:
maybe that scrolling is indeed a good idea, ill work on it when i have time.

ill leave the issue open for now incase you run into problems with the offset patches or other hack/loader related issues.
just wondering, where in memory will you be placing the wiimmfi patches?

(also, i knew about that german mod and have worked with baduncle before. he made the password system in priiloader too iirc. i just thought there was an official fork on github to look at :( )

Leseratte10, 2019-12-22:
I don't (yet) know where in memory I'll put the patches. I'll have to check which blocks of memory are being used by the system menu, both for text/data (I can get these from the DOL) and for stack / heap, and then find some space that is neither used by the system menu nor by DOLs compiled with the official SDK (so, not in use by Nintendo games).

Leseratte10, 2019-12-22:
Just noticed that when I auto-boot into the system menu, the hacks are not applied. When I go into the priiloader and select "System menu", the hacks work.

DacoTaco, 2019-12-22:
thats odd, what does the log\gecko output say exactly?
normally the amount shouldn't matter. it loops them all and tries to apply them (if they are enabled in the states file). it should give an 'applying [hack name' line

Leseratte10, 2019-12-22:
A bunch of invalid stuff, like this: 

    using 0000009B for booting
    loading hacks
    _processLine Exception: Invalid line
    line: [Auto-Press A at Health Screen]ÿÿÿÿÿÿÿÿ
    _processLine Exception: unknown type : ion
    line: ion=518
    Hacks:16

But as I said, it only happens sometimes. And I double-checked the hacks_hash.ini on my SD card, that definitely doesn't have these weird "y" with dots in them.

DacoTaco, 2019-12-22:
shit, you saw my accidental comment xD i pressed enter to quickly xD

thats odd. 
in the hacks loop it loops all loaded hacks and tries to apply all hacks that are enabled in the states file.
it shouldn't try to apply hacks that aren't enabled, and gives an 'applying [hack name]' when its enabled and tries to apply it :/ 

from what i can see of the log though, it failed to load a line correctly, and ditched the whole hack, causing the enabled order to be out of whack. i think that if 1 hack has one of these exceptions , the hack after it will be enabled. is that correct?

so far it hasn't happened for me yet, but ill make a quick patch to disable system menu from booting in dolphin (it crashes my dolphin build) and try to load some hacks. hope it happens here too...

Leseratte10, 2019-12-23:
I would have to re-order the hacks and play around a bit to test that. However I wonder why it doesn't work on an initial boot, but it does work when I press home and click "Wii menu" again. If the file was corrupted, it should be corrupted every time, right?

Leseratte10, 2019-12-23:
Hm, now I made another hacks_hash.ini file for testing, but I can't get it into Priiloader. I have that in /apps/priiloader but it complains that it can't find that file. Interesting, it did work yesterday.

Leseratte10, 2019-12-23:
Finally got it to work. Reinstalling Priiloader or the hacks file didn't work, so I completely reinstalled my System Menu to get rid of all the old stuff, then I reinstalled the Priiloader, and now my patch loads consistently. 

The patch itself seems to work properly, too. It is "just" about 2200 bytes, so in text form it's one line with 6000 chars, and Priiloader seems to handle it just fine. Now I just need to port that hacks patch from 4.3E (which I tested on) to a bunch of other regions, and then find testers in these regions. And find testers with other regions of Mario Kart Wii. 

Thanks a lot for all the changes and fixes in Priiloader.

Leseratte10, 2019-12-23:
There still seems to be a bug in there. I again got an error with the weird ÿ character, with the same patch that worked before (just with a different offset): 

    _processLine Exception: Invalid patch: 0x000ÿÿÿÿÿÿÿÿ

And that ÿÿÿ stuff definitely isn't in my file. I already tried reinstalling my System Menu and the Priiloader.

Leseratte10, 2019-12-23:
I was just looking around the code trying to find what could be causing this problem and noticed that when you read the hacks file ( https://github.com/DacoTaco/priiloader/blob/Hacks_rework/priiloader/source/hacks.cpp#L128 ) you are not using the return value of `fread`. Now I don't know if that is the source of this bug, but you shouldn't assume that fread was able to read the amount of bytes you requested. It will return the number of bytes it read, and that can be smaller than the amount you requested.

[34 more replies]

closed 2020-01-14 by DacoTaco