DacoTaco/priiloader #219: Priiloader won't boot

darksent21, 2019-12-15:
What steps will reproduce the problem?
-Random Crash using USB Loader GX
OR
- Random chance when opening my wii

What is the expected output? What do you see instead?
Expected: loading to System Menu ( I set the Autoboot into System menu)
Output: Black screen. 

*Tried holding reset button on boot
Priiloader doesn't pop out.

*If I try for several times (100+ times)
It randomly pop out and my wii becomes normal again.
I can pop out priiloader normally when holding reset on boot.
But if it crash again using USB Loader GX or turning on my wii randomly it will become black screen again.

What version of priiloader are you using? On what ios/system menu?
I am using 0.8.2

More details:
I have 4.3J Japanese wii.

Similar to these guy from reddit:
I tried loading Priiloader by holding reset and powering on while holding, nothing. Powering on then holding reset, nothing. I tried with and without the USB HDD plugged in. I tried different cables (composite and component, and these cables do work with another Wii). I've tried different TVs as well if that makes a difference. I read that a failed bluetooth module can cause this, but my Wii can be powered on by the remote so that isn't the case for me

DacoTaco, 2019-12-15:
have you enabled the "dump gecko output" option? it should write gecko output to the sd card or usb and tell me where it was when it crashed.
the random part makes this hard to know or figure out where it crashes, but does it also crash when opening priiloader and then selecting system menu?
does this wii have bootmii as boot 2 or only IOS?

darksent21, 2019-12-15:
- It doesn't crash when opening priiloader and selecting system menu. The problem will occur if I open the wii then it suddenly displays black screen.

- Unfortunately the bootmii is installed as an IOS

- I enabled now the dump gecko output from the priiloader menu. (not sure if this will work when the black screen starts to occur).

I manage to pop up the priiloader after doing the "hold reset button" for 43 times (this is the tedious part tbh). My wii seems to work normally again. I'll try to trigger the bug and hopefully it will dump something on the prii.log

darksent21, 2019-12-18:
Okay, now I tirggered the bug.
Last night I played with it. I was in game (loaded from USB loader GX) and press the power button once and it switch to stand by mode. The next morning I open the wii by pressing the power button (turns yellow;standby mode) but it isn't responsive. so I hold the power button to hard reset it and voila, it's now in black screen agan. I checked the prii.log but it seems that it isn't logging anything? Attaching it here:
[prii.log](https://github.com/DacoTaco/priiloader/files/3976455/prii.log)

DacoTaco, 2019-12-18:
from the log i can see it loaded the autoboot dol around 6:48am and it crashed or got stuck there.
i have a suspicion this has to do with the dol's high entrypoint and it overwriting priiloader.
do you still have the dol that you installed for me to test?

also note that if it has to do with the entrypoint of the dol it might get fixed on version 0.9 which is currently in the making

darksent21, 2019-12-18:
Yes. Actually here is where I downloaded it. It's supposedly the forwarder for USB loader GX to autoboot. It's the exact dol that I installed.

https://storage.googleapis.com/google-code-archive-downloads/v2/code.google.com/usbloader-gui/USBLoaderGX_v3.0_IOS249.dol

darksent21, 2019-12-18:
also the weird part is, I'll just keep on doing the "hold reset button" for multiple times in order for the priiloader to work upon boot up. An average of 50-100 attempts. Even unplugging the wii and waiting for 15 minutes. I just need to keep on trying.

But once the priiloader pops out, I can always make it display not unless it will crash again.

DacoTaco, 2019-12-18:
what do you mean with "priiloader pops out"?

also, if you have a black screen, does pressing reset boot the homebrew channel?

darksent21, 2019-12-18:
I mean when the priiloader menu appears again after trying multiple times, it seems that the wii is running normally again. 

-if I encounter black screen, no, it doesnt boot to any homebrew. it stays black.

I just need to restart it again by holding power button. 

example:
wii crashed. Ill turn it off holding power button. Turn it on again then hold reset button. If priiloader menu doesnt show, I’ll just turn it off again and will repeat this method until the priiloader menu shows.

Once its showing again, the wii is normal. If I restart the wii and hold the restart button the priiloader menu will show easily.

But when I encounter a crash, back to my problem. the wii is unresponsive if I press and hold the restart button.

DacoTaco, 2019-12-19:
ye, i have a suspension its the entrypoint. what i don't understand is the randomness of this.

does it also sometimes crash when the usbloader dol is loaded from the priiloader menu (through installed dol or install/load dol)?

i hope it does, cause then i can send you something to test

darksent21, 2019-12-20:
Yes it does. Let me know what I need to use to test it. Give me a week or 2 since I'm having a hard time making the priiloader menu shows after it crash.

DacoTaco, 2019-12-20:
http://upload.dacotaco.com/priiloader_entrypoint_testing.dol

since your wii does not have bootmii @ boot2 , this dol is priiloader itself (not the installer) and therefore has to be loaded through the homebrew channel every time you want to test.

i would like you to disable the autoboot and return to settings (so the menu shows up) and test loading the installed dol (from the priiloader menu) with this version of priiloader and see if it still crashes.
keep in mind it is a work in progress of the version to come and contains a few bugs with the system menu hacks. 
however, it greatly improved on loading binaries, hence the test.

darksent21, 2019-12-20:
Let me check on this once I get home. I just need to make my wii work again.

So this is what it looks like:
Wii -boot-> existing priiloader -select-> HBC -boot-> testing.dol -boot-> Instaled File (USBLoaderForwarder.dol )

BTW, just a quick question,
If it's because of the dol's high entrypoint and it's _overwriting priiloader,_
how come I can still trigger the priiloader randomly and my wii isn't brick yet?

Anyways, I'll give an update after a week of testing. Thank you

DacoTaco, 2019-12-20:
don't forget to disable all autoboot & return to settings in priiloader (the installed one) so that when you boot priiloader from HBC it doesn't try to boot anything.

and thanks for testing.

darksent21, 2019-12-22:
Hello. I would like to say thanks for helping me. Priiloader is very nice and thanks for the effort creating it.

However, I finally gave up.  I’m unable to make my wii work again after attempting to restore it 2 days straight. Maybe it’s really brick now. It’s a shame I didnt have the chance to test the new priiloader.dol. Sorry I can no longer update you but I do believe that its because of the entry point.

Thanks and merry christmas.

closed 2019-12-22 by darksent21