DacoTaco/priiloader #220: Wii Mini and vWii support

Leseratte10, 2019-12-27:
Two more (related) things about Priiloader: Currently, the Priiloader only works properly on the Wii. As far as I know, you can't install it on the vWii (for whatever reason?), and if you install it on the Wii Mini, you might get stuck in the Priiloader as you can't control it without a Wiimote. 

On the original Wii you can just press and hold the RESET button to make Priiloader exit, but on the Wii Mini that doesn't exist. Once you make the Priiloader auto-boot to Priiloader (and then lose / unlink all your Wiimotes) you are screwed and need to take the Wii Mini apart to either solder in a gamecube controller, solder in a reset switch, or solder out the bluetooth module to connect it to a computer to manually link a Wiimote again. 
Could a way be added to 1. be able to get into the Priiloader on the Wii Mini in case of a banner brick (for example, by booting it with the disc tray open instead of closed, or by holding SYNC instead of RESET at boot), and 2. a way to exit Priiloader (holding SYNC while in Priiloader?). 

2nd, as far as I know, the Priiloader cannot be installed on the vWii (virtual Wii of the WiiU). Is there a particular reason for that? Does the WiiU menu validate the Wii menu before booting it? Is there a chance that that will ever be changed so one can use the Priiloader in the virtual Wii as well? Of course that would then need a different way (like holding SYNC) to boot the Priiloader as well.

DacoTaco, 2019-12-27:
for vwii : 
last time i heard(/checked, which was when f0f released info about running code on the system through vWii ) the vWii titles have different boot code that init the system and also is encrypted so it can run on the cpu. with that info (afaik), and not wanting to brick my WiiU i didn't proceed to try and get priiloader running on that. 
i dug up the blog where i read it back in the day : https://fail0verflow.com/blog/2013/espresso/
is this information outdated? have methods been found to edit a channel and keep it bootable?
i guess homebrew can be made bootable (as homebrew channel works on it) but i have no idea what needs to be done for it...

as for the wii mini : 
i don't know anything about the mini sadly, but would the open state be recorded in memory like the gamecube did? if i would want the eject button to make it show the menu, what do i use to move or start something?
the sync button is (afaik) not available because its being captured by libogc for wiiuse (wiimote library). this is why i only used reset & power for moving through the menu with the latest version. 
imo the wii mini has a great lack of interfaces which makes it very hard to navigate with , though having priiloader would help a lot of people..
the wii mini's first step would be to somehow detect its running on the mini imo :/ 

i agree priiloader would be great on those 2 systems, but with the technicalities that i know off now, i don't see it working.

Leseratte10, 2019-12-27:
For the vWii, I don't know how exactly it would work. In a vWii NAND dump, the system menu doesn't seem encrypted though. The question is, is there any authenticity check done before that. Maybe someone more involved with vWii hacking can say more about that. 

Detecting the Wii Mini is pretty simple. Wii software can request the Wii MAC address from IOS, and because the Wii Mini doesn't have that, a Wii Mini has the file `/title/00000001/00000002/data/macaddr.bin` in its NAND that contains the console MAC address that will be returned when software asks for it, so all you'd need to do would be to check if that file exists. 

Navigating in Priiloader on Wii Mini could still be done with a Wiimote. You'd just need a way to get into Priiloader (SYNC or open disc tray?), and a way to get out of Priiloader if you set it to autoboot to Priiloader and then lose all your Wiimotes. 

Libogc is interfering with the sync button? Interesting, didn't know that. Why would it need to do that, given that you can't sync wiimotes in homebrew anyways?

Leseratte10, 2019-12-27:
Just tested it on my Wii Mini: After doing `DI_Init`, `DI_GetCoverRegister` can be used to get the disc drive status. Lowest bit not set means it's closed, lowest bit set means it's open. Doesn't matter if a disc is in there or not. 

So it would be possible to check for a Wii Mini (by checking if that macadr.bin file exists), and if it is a Wii Mini, open the Priiloader menu if the drive tray is open. Then wait for the user to close it again, and when he opens it again while Priiloader is running, boot the System Menu. That way there would be a way to enter the Priiloader without using a rebooter app, and there would be a way to leave the Priiloader without a Wiimote.

DacoTaco, 2019-12-28:
the wii mini stuff is good to know. next time i have time ill add the check to the priiloader installer to prevent installation on a wii-mini so it won't get installed before proper support was added.

are you aware of how far wiiU development has gone? is it possible to unbrick vWii if i brick it from WiiU mode? if it is, i might take a look at vWii and channel boot codes

Leseratte10, 2019-12-28:
So are you planning to make Priiloader 0.9 with that Wii Mini block? Wouldn't it be easy enough to add that disc drive status support in 0.9 so it could be safely used on Wii Mini?

I have never tested it, but here's a tutorial on gbatemp about unbricking the vWii from WiiU mode. You'd just need to have the WiiU mode modded, and make a bunch of different backups beforehand: https://gbatemp.net/threads/guide-vwii-unbrick-guide-by-garyodernichts.528329/

DacoTaco, 2019-12-29:
my idea was to not release 0.9 before wii mini support is added, but have installer block it so nothing bad could happen. i mean, whats the purpose of having priiloader on a wii menu if you can't even access it (without a homebrew settings editor to disable autoboot or use its autoboot override options).

ill add the file & DI checks though so that at least you can open the priiloader menu (hopefully). sadly i don't have any way of testing this properly as i don't have a mini. 

as for WiiU , i figured WiiU mode access would be needed, but i never hacked my WiiU and afaik there is no way to hacking method i like ( that doesn't screw with startup methods much). but i have brain age apparently so ill look into it soon™.

DacoTaco, 2019-12-29:
well now, i think we'll have our WiiMini & Wii input soon seeing that PR :P

Leseratte10, 2019-12-29:
Great, so we'd only need to look into vWii support. I'll also try to get the new hacks stuff finally tested soon so that that branch can be merged as well.

DacoTaco, 2020-01-19:
i did some digging & testing on vWii : 
short answer : Nintendo changed some stuff and prevents System Menu's bootcode to be tempered with. Priiloader can't work on vWii unless it would weirdly enough still boot the nandcode (it shouldn't)
long answer : Every time the powerpc cpu reset's it starts a bootrom that verifies and decrypts the code its going to start running. the code it will do that with are called ancast images. System menu is an ancast image so as soon as you start tempering with its boot binary (which priiloader does) it'll halt and do nothing. it binary still contains nandcode (code that the wii used to run when starting a title) but i don't believe it'll run that, so i can't hack it to jump to other code ( ill test it later to verify that ).

how can other channels like HBC work ?
i don't fully know yet, but from what i think happens is that for compatibility, IOS boots a special piece of code (also an ancast image) on the powerpc when it reset's. this code does the stuff nandcode is suppose to do, and then jumps to the channel's entrypoint, skipping the channel's nandcode. so those don't need to be encrypted.
i haven't verified this, but in theory if i were to install a channel i made on the wii (my own homebrew channel if you will) on vWii it should boot with no modifications required because it already had the binary set up how the wiiU expects it

DacoTaco, 2020-01-24:
closing this, wii mini support has been added and vWii support can't be done by the looks of it.
i tried using the nandloader in the priiloader app to get it to work but failed.
IOS will be needed to change and im not capable of completely disassembling IOS to find the code that boots SM

closed 2020-01-24 by DacoTaco