DacoTaco/priiloader #230: Random freezes with new hacks test version

Leseratte10, 2020-01-06:
I've had two people test the new test version with keyboard and new hacks support, and it looks like that version doesn't work reliably. Both people reported that opening Priiloader and then selecting "System Menu" causes it to freeze after the backmenu / intro animation. Autobooting into the system menu works without freeze, and returning to System Menu from HBC (thus triggering the autoboot sequence) works, too. 

This has been tested both with and without the new Wiimmfi patch enabled, so it's not caused by my hacks but by some of the changes in the new Priiloader.

DacoTaco, 2020-01-06:
did they have any other hacks enabled?
ive had a similar problem last night, but it looked like the auto press A hack was the cause somehow

Leseratte10, 2020-01-06:
He is using 4.3U, and he didn't have any hacks enabled.

DacoTaco, 2020-01-06:
crap, 4.3U has the same dol specs as 4.3E. so the BSS isn't it.
doesn't have a usbgecko either right, to see where it goes wrong?

Leseratte10, 2020-01-06:
No, he doesn't. Do you think the gecko-to-SD output could be helpful? I can try installing priiloader on an NTSC-U wii, too, maybe I get the same crash. I do have a gecko (but I never did any homebrew debugging, just game debugging with the gecko handler).

DacoTaco, 2020-01-06:
you could ask, but the gecko to SD output will just say it booted SM i think. and the gecko is to enable the SM OSreport to gecko hack, to see what SM is doing. 
its just very odd if it only happens if the Priiloader menu was shown...

DacoTaco, 2020-01-07:
if you cause the crash, would you please enable the OSReport to usb gecko hack. that way i can see if its related.

currently im getting random (very rare) crashes that are giving me the following output through the OSReport's from SM

https://pastebin.com/ZWA2x7Zc

Leseratte10, 2020-01-07:
That's an interesting crash. It crashes at 0x813359bc, which is supposed to load a floating-point number into f3 (`lfs f3, -32732(r2)`). r2 looks like it's a valid memory address (0x8169e2a0), so I'd have no idea why that would crash there. Plus, it complains about an invalid instruction, not an invalid data address (ISI instead of DSI exception). Is it possible that Priiloader is corrupting system menu code at or around 0x813359bc?

Looking at the code it seems to be related to fonts / text rendering but that's just a vague guess based on the call flow and the filenames accessed around there.

DacoTaco, 2020-01-07:
Priiloader is loaded in : 
- 0x80003400 - 0x80003900 (boot code)
- 0x81100000 - 0x81199CA0
so no, priiloader is nowhere near those addresses (and that shouldn't matter since the loader does the loading).

it would be great if i could see the state of those registers at the moment of the crash >>

from what i can see in dolphin that is what it does. loads it from somewhere in the toc ( 0x8169e2a0 - 0x7fdc = 0x816962C4) which looks to contain data from the SM binary's data section...

you got the app/dol disassembled? its loading fonts around there somewhere?

Leseratte10, 2020-01-07:
The registers are visible in the output in your link, aren't they? Or were you talking about the floating point registers? The normal registers are between line 61 and 76 in your pastebin. But it complains about an invalid instruction (not an invalid address) so a memory dump from that location would probably be more useful. 

Yeah, I loaded the system menu app file into Binary Ninja, but I just took a quick look. I might be wrong about the fonts.

DacoTaco, 2020-01-07:
ye i was wrong, i meant to say mem dump + registers. basically, what i can do in dolphin but at a notice on the actual wii when its crashing.

i tried booting SM in dolphin a few times but it seemed to work...

Leseratte10, 2020-01-07:
You could just use the gecko code handler to connect to the System Menu with tools like Gecko .NET, then you should be able to see the memory in there, and may even be able to catch the exception. Here's the code handler for 4.3E: https://pastebin.com/53DBKmTH

DacoTaco, 2020-01-07:
ye i know, ive been thinking of adding it to my hacks. 
i don't seem to find gecko.net though, i was sure i had it somewhere in my console hacking folders... 
it'll be for tomorrow to see what its about.

DacoTaco, 2020-01-07:
nvm, just tried it. its a full cpu halt. the gecko code can't access anything as the cpu halted

DacoTaco, 2020-01-09:
just did some looking and stuff using dolphin and compared its osreports to the crash's.

i don't know what that function is, but its called a few times at boot but the first time its called it crashes.
do you happen to have a .map export of the functions (with names) that i could load into dolphin?

Leseratte10, 2020-01-09:
No, I don't have any function names for the System Menu.

DacoTaco, 2020-01-12:
update : 
i discovered that lowering the entrypoint of Priiloader fixes the issue on my side & i don't fully know why yet. 
i also noticed the video buffer getting corrupt when starting system menu or homebrew channel (green lines and pixels just before HBC or SM boots). i checked in dolphin and yes, SM is using the video buffer address that priiloader was using. so chances are it crashes sometimes because its loading from an address priiloader was using somewhere down the line that just escalates into more trouble and a crash.
thats what im guessing though...

Leseratte10, 2020-01-12:
Ah, cool. So just clearing that memory in the Priiloader loader should fix all the inconsistent crash issues?

DacoTaco, 2020-01-12:
yes, i believe it would.

however, a lot more changes would be required (like shutting down the video interface before booting anything and whatever stuff also screws up) so im going to revert the entrypoint change and i'll see afterwards if i need to up it or if that was a stupid move.

at this point the only reason im using 0x81000000 or 0x81100000 is cause otherwise i can't load priiloader in HBC which sucks for development... (if only HBC used a loader like mine xD)

ill see to update the branch and you can try then

DacoTaco, 2020-01-12:
i updated the branch, could you verify?

Leseratte10, 2020-01-12:
Yeah, I'll do that later. Thanks for (hopefully) fixing this.

Leseratte10, 2020-01-14:
It looks like the random crashes are gone with the current state in the Hacks_rework branch. Also, I can confirm that the fix in the ResetFix branch works, and I can now reliably get into Priiloader without having it start the System Menu immediately. 

I currently can't test the keyboard support as I currently don't have a USB keyboard available, but hopefully I can do some tests with that on the weekend. I will prepare and test a new version of the Wiimmfi patch hacks probably tomorrow.

DacoTaco, 2020-01-14:
alright, ill set everything up for doing the PR's

Leseratte10, 2020-01-25:
@DacoTaco I just built from master again ( aca52e54098fa695a49e6de26ad5a2e4a0b6d3f2 ) and it freezes when booting the System Menu, no matter if done by autoboot or by manually selecting System Menu. Anything that changed recently that could cause this? What was the reason for the opendolboot change?

DacoTaco, 2020-01-25:
odd, worked fine for me.
opendolboot has been changed a few times in the past week
first change was to rework it and use the init code used by HBC (open source) instead of the SM code (copyrighted code). you verified that worked iirc.

that commit however did nothing special with opendolboot and just makes it so that for windows machine's it also builds if it doesn't exist yet (it was always trying to build it)

what settings are you using in priiloader ? :/

DacoTaco, 2020-01-25:
it should be solved now

closed 2020-01-25 by DacoTaco