DacoTaco/priiloader #231: Implement HID driver - Get USB Controllers to work

DacoTaco, 2020-01-12:
after the small discussion in #214 , it would be cool to have a HID driver in libogc/priiloader that allows for any USB controller that supports HID to work in priiloader.

DacoTaco, 2020-06-26:
testing dol : https://upload.dacotaco.com/HIDControllers.dol

niuus, 2020-06-27:
'sup @DacoTaco. I will test your dol in a couple of hours, gonna sleep right now.

niuus, 2020-06-27:
Ended up testing it before sleeping... oops!

Okey, so, here's what happened: a lot of lines started showing up, when i plugged the DualShock3, this happened:
https://imgur.com/lw50zyn

After that, i tried with a generic PSX2USB adapter:
https://imgur.com/LHKjRKz

The Xbox 360 Controller did nothing (xinput, i know). I currently have a loaned DS4, that one showed something but it was illegible. and the lines did not stopped.

Excuse the quality from the CRT pics, i'll try with a progressive monitor later. I will check here again in some hours. Off to sleep now!

niuus, 2020-06-27:
Ended up testing it before sleeping... oops!

Okey, so, here's what happened: a lot of lines started showing up, when i plugged the DualShock3, this happened:
https://imgur.com/lw50zyn

After that, i tried with a generic PSX2USB adapter (2 ports):
https://imgur.com/LHKjRKz

The Xbox 360 Controller did nothing (xinput, i know). I currently have a loaned DS4, that one showed something but it was illegible, and the lines did not stopped. Pressing buttons or dpad did not made any difference with any of the controllers.

Excuse the quality from the CRT pics, i'll try with a progressive monitor later. I could suggest you to implement maybe some kind of print to text file, if it is useful to you. I will check here again in some hours. Off to sleep now!

DacoTaco, 2020-06-27:
thanks for that. 
im kinda curious as to how those controllers work and how different it is from my basic HID controllers.
there is not much you can do (sadly) then.

from what i saw online i expected the DS4 controller to work but something must be acting differently from what i interpreted from the HID specs.

are you using linux by any chance?

niuus, 2020-06-28:
TBH the one i was kinda expecting some response was the PSX2USB adapter, which i already got 'working' in Snes9x RX code (and previously worked without issues at all in the Retroarch HID code for Wii or Wii U), but it is giving me some weird frame skipping error for the moment i have not been able to fix. The second controller i was expecting some response was the DS4, but nope, no dice.

Nope, no Linux here, sadly. Windows or Mac OS. I also have a Wii U Pro Controller which i could connect to the USB. If you need some more testing, just lemme know.

DacoTaco, 2020-06-28:
'working and responding to all inputs' ? 
it works without having to install a special driver in windows?
if so, then yes, it should've worked in the program :/
what response did you get?
and what do you mean with a weird frame skipping error?

niuus, 2020-06-29:
> 'working and responding to all inputs' ?
> it works without having to install a special driver in windows?
> if so, then yes, it should've worked in the program :/
Yes, it works without rumble, which is what the full driver install covers.

> what response did you get?
> and what do you mean with a weird frame skipping error?
If you're on GBATemp, we could exchange messages there if you want, or email. I can send you the code.

> also, on a controller that doesn't work, please plug in the controller first and then run this : https://upload.dacotaco.com/http://upload.dacotaco.com/HIDControllersDebugNoThread.dol
Cool, i'll test all the controllers again. I have even found my old Xbox OG to USB adapter, i'll add it to the tests.

niuus, 2020-06-30:
The results are in:

https://imgur.com/a/jsBkA41