DacoTaco/priiloader #247: Add language support

wuniversales, 2020-04-19:
I would like to add the Spanish language to the application, I can translate the app if you pass the strings to me.
Are you going to translate the application?
Thanks greetings

DacoTaco, 2020-05-02:
im not sure what i want to do with this.

we've had translations before ( a german mod ) that just changed all text in code and re-released priiloader as a separate thing. it was managed by BadUncle back then. 
ive removed the code that allowed it to semi work since then as i didn't like the idea of separate releases just for a translation.

a better idea would be to have the translations embed in priiloader itself, but it would increase the size of the priiloader binary significantly...

wuniversales, 2020-05-02:
The same version could be released in different languages, right?
I mean, a priiloader with strings in Spanish, another with strings in English, and so ...
This way we don't increase the size of the priloader, and we have the advantage of translations.
I propose that the priiloader installer of the option to select a language, according to the chosen language, install priiloader only with the selected language.
Cheers

DacoTaco, 2020-05-02:
that would mean building priiloader for every language and putting it in the installer , making that big..

a better option, i agree, but not optimal yet.
now that we have more memory to use ( hallelujah loader internal loader <3 ) i could try and load a file in memory have all translations in there.
however, this would slow boot time and im not sure where to save the translations to either. sd/usb? nand? :/

wuniversales, 2020-05-04:
Yours would be to save the user's language in the nand, if it is not saved in the nand, when extracting the pendrive or the SD it would not load the user's language.

It would be interesting to implement it in the next version, but as you wish ...

Another option would be for the installer to download the file with the user's language from github, and once the language (and priiloader) is installed, the language files will be removed at the end of the installation, since it would be integrated in the wii, thus the Installer would also weigh less.
The only downside to this is that it would require internet to function ...
Although perhaps an "offline" method can be introduced.

The really important thing for me (and I suppose for you too) is the size that is used in the memory of the wii.

DacoTaco, 2020-05-07:
> The really important thing for me (and I suppose for you too) is the size that is used in the memory of the wii.
this is indeed true for me too. i want priiloader to be as small, with as less possible files and as less possible things that can break it as possible..

i like having a translation file next to the installer to be an option, and i'd prefer it to have it maintained within the priiloader git repository so it is open for anyone to ask for changes/fixes or fix it themselves and send it upstream through pull requests.

i would also not let the installer download it since if we are working on a version it could download the wrong one containing stuff it shouldn't yet.
however, if we don't download them then they could break during an (online) update.
including them in the releases could fix that, but online updates are still fucked then

hmm, there has to be a solution to this :)

wuniversales, 2020-05-15:
mmm ...
And keep all the translation files in a "lang" folder (/app/priiloader/lang/) ?
Later the boot.dol would only install the language selected by the user.
Maybe like this?

TheTechRobo, 2021-03-08:
What about two priiloader installers, one "international" one that has all the languages, and one "English" that is the basic smol installer.