DacoTaco/priiloader #248: hacks_hash.ini does not work

ArvoX, 2020-04-19:
The 2 lates version of hacks_hash.ini does not work.

8603307323b168565d675669ce07d1bd1c64940e
error message: missing version before n @ 124

edf41b53b9f10d7d1d5caf5076a246babcec3398 does not work ether.

DacoTaco, 2020-04-19:
are you using this hacks file with a priiloader version before the 0.9 beta? 
cause the hacks file got some reworks and this version only works with 0.9 ...

ArvoX, 2020-04-20:
I use version 0.8.2 downloaded from http://wiibrew.org/wiki/Priiloader
It links to the newest hacks_hash.ini here and says that it's the recommended version to use.

urherenow, 2020-04-20:
“It” says. If you look again, Dacotaco is the author of priiloader. There is no better advice than from him, and he just let you know that the last 2 versions are made to work with the 0.9 beta, not 0.8.2. And it’s a cool beta! Try it out!

DacoTaco, 2020-04-20:
ye ok, i understand. the wiki does link to the one meant for 0.9

this should be an temporarily issue and in future development i'll change the way i work :) 

for now, use this version for 0.8.2 : https://github.com/DacoTaco/priiloader/blob/22159768c82b8c8c2293f7bef3f9ebe3b022a966/priiloader/hacks_hash.ini

closed 2020-04-20 by DacoTaco