DacoTaco/priiloader #250: Wii Mini - Crash enabling "Replace Health Screen with Backmenu"

buggii, 2020-04-29:
In Wii Mini, if you enable "Replace Health Screen with Backmenu" the console will reboot to a black screen. Need to disable the hack.

DacoTaco, 2020-05-02:
i might have found the issue.
can you try replacing the hack on the usb drive with this one : 

``[Replace Health Screen with Backmenu]
maxversion=4610
minversion=4610
amount=1
hash=0x8140A118
patch=0x8140A0B0``

buggii, 2020-05-03:
Unfortunately black screen again.

DacoTaco, 2020-05-07:
*sigh*

i just debugged the whole system menu in dolphin in both 4.3E on the wii and the wii mini system menu only to come to a conclusion my patch was correct except i did -0x68 on an address while i had to do +0x68 ...

try this patch

```
[Replace Health Screen with Backmenu]
maxversion=4610
minversion=4610
amount=1
hash=0x8140A118
patch=0x8140A180
```

DacoTaco, 2020-05-07:
note for myself incase it is wrong : 

3c8081665460103a38848b2c7c84002e

base address : 8132FAE0

8140a118 - 8140a124 -> 280300257c9d23787cbe2b783be00000 -> in file 0xDA638
8140a138 -> instruction that loads the patched address value
patched address : 81658B94 value: 8140A464 -> 8140A4CC

81409DCC - 81409DD8 -> 280300257c9d23787cbe2b783be00000 -> in file 0xDA2eC
81409DEC -> instruction that loads the patched address value
r4 = 81658d0c , r0 = 0x68 -> 81658D74 -> value is 0x8140a118 -> patch = 0x8140A180
file addr -> 329294

81409df881409e3881409e5c81409e80

buggii, 2020-05-08:
Now the hack "Replace Health Screen with Backmenu" is working fine ! No crashes !!! Great man ! Thank you !!!

closed 2020-05-08 by DacoTaco

Leseratte10, 2020-05-22:
Just wanted to note that the current hack is only for the PAL Wii Mini, for the USA version it would need to be ported to version 4609 as well.

DacoTaco, 2020-05-22:
that is correct, but i need the USA app to port this, as the hash will be different.