DacoTaco/priiloader #251: [Bug] Stop disc on startup

wuniversales, 2020-05-07:
The wii does not stop the disk at startup, however the hack to lock the disk on boot is activated.
I have tried it on two wii consoles, and on both the same error.
They both have...
- Version 4.3E
- Open Homebrew Channel v1.1.4

Steps to reproduce the behavior:
- Go to Settings
- Enable "Stop disc on startup"
- Save changes.

Priiloader v0.8.2

Expected behavior:
The disc should not make noise when starting the Wii (It should not start)

Screenshots:
If necessary I will provide them.

DacoTaco, 2020-05-07:
this settings only shuts down the dvd drive when entering the priiloader menu, is this the moment it is still spinning?

wuniversales, 2020-05-07:
Yes, even if you access the priiloader, the disk continues spinning.

DacoTaco, 2020-05-07:
these are regular wii's?
cause on my unit it stops the disc :/

wuniversales, 2020-05-07:
Before (years ago), I don't know if it was with a preloader or priloader ... With that option. the whole disk stopped at startup, now it always plays.
Of course I had another version of Wii before, I think it was 4.2E, I don't remember very well ...

My wii has not been opened, nor modified.
The only thing modified is the ones you see in the report ...

It does not have any inserted disc, it makes the noise of the start of the disc although it does not have cd.

DacoTaco, 2020-05-08:
now im confused a bit,

so with a game inserted it doesn't stop it with that option enabled, right?
or does it also make a constant noise with nothing inserted?
if it is the latter , i would think the dvd drive is acting up and it isn't a software problem.

wuniversales, 2020-05-08:
Hello,
I send you a video with the problem...
Video: http://www.filedropper.com/priiloader
Video (Option 2 in case they remove it): https://www.sendspace.com/file/su78ey
It always loads the cd at startup, it always makes noise when booting.

DacoTaco, 2020-05-09:
'warning unsolicited event' 
what the fuck?

i think that's the real issue there.

DacoTaco, 2020-05-09:
i also hear the disc starting up, which is normal (happens before system menu and priiloader are started) but it takes a while before priiloader shows up :/

wuniversales, 2020-05-09:
But the disk should not start with the active hack, right?
At least in the past I think it did not start if you activated the hack.

The truth, the error 'warning unsolicited event' has never appeared to me, it is the first time it appears ...

The blink thing does it sometimes.

I have the screen set to 60Hz, I changed it to 50Hz to try, but it blinks when you feel like it in both cases ...

DacoTaco, 2020-05-11:
the warning message apparently comes from libogc/wiiuse (the code that handles the wiimotes). 

stopping the disc is not a hack, its a setting of priiloader. there is a difference. 
but youre the second person to mention it like a hack, so either im missing something or there used to be a hack for it? o.O

i also dug through the code a bit and currently priiloader only checks that setting when entering the priiloader menu, not when autobooting anything (system menu, installed file,... ) .
from what i think i hear in the video (was a bit hard to hear sometimes with the video call in the background) , is the disc drive starting up (as normal) and then going silent, cause you entered the priiloader menu.
is this correct?
if this is the case, this is how it is suppose to work now. i also don't understand what the point is of having it stop the disc when booting system menu?
when booting the autoboot (or booting binaries from priiloader) should stop the dvd drive anyway...

wuniversales, 2020-05-11:
> but youre the second person to mention it like a hack, so either im missing something or there used to be a hack for it? o.O

I think it used to work as a trick in the past.
I remember that in the past if this option was activated, the disk did not make noise when starting the wii.

Maybe this was happening with preloader?

> from what i think i hear in the video (was a bit hard to hear sometimes with the video call in the background) , is the disc drive starting up (as normal) and then going silent, cause you entered the priiloader menu.

I do not get it...
What is this option for if the disk makes noise when booting?
The disk starts at all times, it does not matter if you activate the hack or not, it always makes noise.

If you mean that the disk "doesn't rotate" when activating the hack and loading priiloader, I don't know, I suppose it won't rotate and the hack works correctly, my wii luckily doesn't make much noise and I can't appreciate it.

>  i also don't understand what the point is of having it stop the disc when booting system menu?

Avoid noise at startup, at least that use was given before.

Likewise, in the past when activating the hack, the disk channel worked correctly.

If I had a CD inserted, the game would appear on the disco channel, if I had no CD inserted, obviously no game would appear on the disco channel.

regards

closed 2020-09-20 by DacoTaco