DacoTaco/priiloader #252: syntax error: expect 'version' before 'n' @ line 130

Kirbeast, 2020-05-08:
**Describe the bug**
System Menu Hacks don't show up

**To Reproduce**
Steps to reproduce the behavior:
Installed Prilloader with latest hacks_hash.ini from git

**Version**
v8.02

**Expected behavior**
System Menu hacks to appear as normal

**Screenshots**
https://cdn.discordapp.com/attachments/426605554547425281/708421343129108561/JPEG_20200508_145327.jpg

**Additional context**
Wii is on 4.3U

Kirbeast, 2020-05-08:
I swapped lines 130 and 131. 
`maxversion=65535`
`minversion=1`
I don't get the error anymore, now I get a new error at line 133.
syntax error: expect 'hash' or 'patch' before line 133

Kirbeast, 2020-05-08:
Oh wait I'm dumb, I need to use the beta version to use the hacks_hash.ini on the git.

closed 2020-05-08 by Kirbeast

DacoTaco, 2020-05-09:
hehe.
ye, the hacks file on git is meant for 0.9 beta 3 as it contains the new features and hacks :)
sorry for the problem though. do you still have a hacks file for < 0.9 ?