DacoTaco/priiloader #254: New hacks system crashes

Leseratte10, 2020-06-18:
On 4.3U, the following two hacks work just fine: 

```
[Wiimmfi Patch v1 - hidden MASTER]
minversion=1
maxversion=65535
amount=1
master=wiimmfi-v1
offset=0x812ff030
patch=0x9421ff54,0x90010008,0x7c0802a6,0x900100b0,0xbc410014,0x7c6000a6,0x5464045e,0x7c800124,0x54638ffe,0x9061000c,0x3f608000,0x63634000,0x3c808090,0x3cc06874,0x60c67470,0x3ce0733a,0x60e72f2f,0x7c032000,0x4080003c,0x38630001,0x80a30000,0x7c053000,0x4082ffec,0x80a30004,0x7c053800,0x4082ffe0,0x39030004,0x39230003,0x8d480001,0x9d490001,0x2c0a0000,0x4082fff4,0x4bffffc4,0x63634000,0x3c808090,0x3d806e69,0x618c6e74,0x3e60656e,0x6273646f,0x3dc07769,0x61ce6669,0x3de02e6e,0x61ef6574,0x3e007769,0x6210696d,0x3e206d66,0x6231692e,0x3e406465,0x7c032000,0x408003c8,0x38630001,0x80a30000,0x7c056000,0x4082ffec,0x80a30004,0x7c059800,0x40a2ffe0,0x80a30008,0x7c057000,0x40a2ffd4,0x80a3000c,0x7c057800,0x40a2ffc8,0x92030000,0x92230004,0x92430008,0x3903000f,0x39230009,0x8d480001,0x9d490001,0x2c0a0000,0x4082fff4,0x4bffffa0,0x48000368,0x480003a1,0x60000000,0x68747470,0x3a2f2f63,0x612e6e61,0x732e7769,0x696d6d66,0x692e6465,0x2f636100,0x68747470,0x3a2f2f6e,0x61737769,0x692e7769,0x696d6d66,0x692e6465,0x2f616300,0x68747470,0x733a2f2f,0x6d61696e,0x2e6e6173,0x2e776969,0x6d6d6669,0x2e64652f,0x70700000,0x2c030000,0x40810040,0x8061000c,0x800d9738,0x2c000002,0x41820014,0x906da14c,0x38000002,0x900d9738,0x48000020,0x38830003,0x5484003a,0x88a40000,0x7ca42a14,0x7c0028ac,0x7ca803a6,0x4e800021,0x3860ffff,0x2c030000,0x00000000,0x808da14c,0x28040001,0x408100a0,0x9421ff80,0x38610020,0x3c00801d,0x600024f4,0x7c0803a6,0x4e800021,0x38610020,0x808da14c,0x38a00554,0x90bc00c4,0x3c00801d,0x60002544,0x7c0803a6,0x4e800021,0x38610020,0x38810010,0x3c00801d,0x600025f8,0x7c0803a6,0x4e800021,0x48000018,0x0fff1f07,0x00e638c9,0x49fbeffa,0x79022d3a,0x84ab134f,0x7c6802a6,0x3881000c,0x38a00005,0x7ca903a6,0x84a30004,0x84c40004,0x7c062800,0x40820008,0x4200fff0,0x7c062800,0x38210080,0x808da14c,0x41820008,0x38800000,0x00000000,0x7c651b78,0x88c50000,0x2c060068,0x40a20068,0x88c50001,0x2c060074,0x40a2005c,0x88c50002,0x2c060074,0x40820050,0x88c50003,0x2c060070,0x40820044,0x88c50004,0x2c060073,0x40820038,0x88c50005,0x2c060000,0x40820010,0x38c00000,0x98c50004,0x48000020,0x38e50003,0x39050004,0x8d280001,0x9d270001,0x2c090000,0x4082fff4,0x9d270001,0x88c50000,0x2c06006e,0x40a20118,0x88c50001,0x2c060069,0x40a2010c,0x88c50002,0x2c06006e,0x40820100,0x88c50003,0x2c060074,0x408200f4,0x88c50004,0x2c060065,0x408200e8,0x88c50005,0x2c06006e,0x408200dc,0x88c50006,0x2c060064,0x408200d0,0x88c50007,0x2c06006f,0x408200c4,0x88c50008,0x2c060077,0x408200b8,0x88c50009,0x2c060069,0x408200ac,0x88c5000a,0x2c060066,0x408200a0,0x88c5000b,0x2c060069,0x40820094,0x88c5000c,0x2c06002e,0x40820088,0x88c5000d,0x2c06006e,0x4082007c,0x88c5000e,0x2c060065,0x40820070,0x88c5000f,0x2c060074,0x40820064,0x38e00077,0x98e50000,0x38e00069,0x98e50001,0x98e50002,0x98e50006,0x38e0006d,0x98e50003,0x98e50004,0x38e00066,0x98e50005,0x38e0002e,0x98e50007,0x38e00064,0x98e50008,0x38e00065,0x98e50009,0x39050009,0x3925000f,0x8ce90001,0x9ce80001,0x2c070000,0x4082fff4,0x9ce80001,0x38a50001,0x3d208089,0x6129a66c,0x7c092800,0x41820014,0x3d208089,0x6129a738,0x7c092800,0x40a20008,0x38a50040,0x3d2080a0,0x7c092800,0x4181fe40,0x38a00000,0x3d00801a,0x61086ffc,0x7d0903a6,0x4e800420,0x60000000,0x800d9758,0x906da16c,0x900d9758,0x808da16c,0x60002850,0x808da16c,0x600028a0,0x60002954,0x808da16c,0x38800052,0x38a0004d,0x38c00043,0x88fb0000,0x7c072000,0x4082035c,0x88fb0001,0x7c072800,0x40820350,0x88fb0002,0x7c073000,0x40820344,0x8bfb0003,0x637e5f34,0x4bfffc64,0x7fa802a6,0x3c608027,0x2c1f0050,0x40820008,0x60636054,0x2c1f0045,0x40820008,0x60631d14,0x2c1f004a,0x40820008,0x606359f4,0x2c1f004b,0x4082000c,0x3c608026,0x60633e34,0x3c808130,0x3884f000,0x38840004,0x38a0002a,0x7fc803a6,0x4e800021,0x3c608027,0x2c1f0050,0x40820008,0x6063a400,0x2c1f0045,0x40820008,0x606360c0,0x2c1f004a,0x40820008,0x60639da0,0x2c1f004b,0x4082000c,0x3c608026,0x606382b0,0x7c7c1b78,0x389d0004,0x38a0001c,0x7fc803a6,0x4e800021,0x387c0028,0x389d0020,0x38a0001c,0x7fc803a6,0x4e800021,0x387f0020,0x389d0059,0x98640000,0x387c004c,0x389d003c,0x38a0001f,0x7fc803a6,0x4e800021,0x3de0733a,0x61ef2f2f,0x3e003bc0,0x3e20800e,0x2c1f0045,0x4082001c,0x3c608029,0x6063d12c,0x91e30000,0x6231ca0c,0x92110000,0x48000058,0x2c1f004a,0x4082001c,0x3c60802a,0x60630e0c,0x91e30000,0x6231c9cc,0x92110000,0x48000038,0x2c1f004b,0x4082001c,0x3c608028,0x6063f474,0x91e30000,0x6231cb24,0x92110000,0x48000018,0x3c60802a,0x6063146c,0x91e30000,0x6231caac,0x92110000,0x637c1000,0x7f83e378,0x389d005c,0x38a002dc,0x7fc803a6,0x4e800021,0x2c1f0045,0x40820044,0x3c606000,0x38832454,0x38a324a4,0x38c32558,0x909c0068,0x90bc0088,0x90dc00a0,0x3c606129,0x60645c74,0x60655d40,0x3c606108,0x60666f5c,0x909c029c,0x90bc02ac,0x90dc02d0,0x480000d0,0x2c1f004a,0x40820044,0x3c606000,0x38832414,0x38a32464,0x38c32518,0x909c0068,0x90bc0088,0x90dc00a0,0x3c606129,0x606497cc,0x60659898,0x3c606108,0x60666f1c,0x909c029c,0x90bc02ac,0x90dc02d0,0x48000088,0x2c1f004b,0x40820080,0x387d033c,0x80830000,0x909c000c,0x80830004,0x909c0018,0x80830008,0x909c0020,0x8083000c,0x909c0050,0x80830010,0x909c0068,0x80830014,0x909c0078,0x80830018,0x909c0088,0x8083001c,0x909c00a0,0x80830020,0x909c00f0,0x3c803d20,0x60848088,0x909c0298,0x909c02a8,0x3c806129,0x60858aa4,0x6086ab58,0x90bc029c,0x90dc02ac,0x3c806108,0x60847358,0x909c02d0,0x3de0800e,0x3e00801d,0x3e20801a,0x2c1f0045,0x40820014,0x61efe300,0x62104e5c,0x62317240,0x48000040,0x2c1f004a,0x40820014,0x61efe2c0,0x62104e1c,0x62317200,0x48000028,0x2c1f004b,0x40820014,0x61efe418,0x62105258,0x6231763c,0x48000010,0x61efe3a0,0x62104efc,0x623172e0,0x7f83e378,0x3880004c,0x38af0004,0x48000091,0x7f83e378,0x388000fc,0x38b00004,0x48000081,0x7de37b78,0x38800000,0x38bc0000,0x48000071,0x7e038378,0x38800000,0x38bc0050,0x48000061,0x7e238b78,0x38800000,0x38bc0100,0x48000051,0x8061000c,0x2c030000,0x7c8000a6,0x4182000c,0x60858000,0x48000008,0x5485045e,0x7ca00124,0x800100b0,0x7c0803a6,0xb8410014,0x80010008,0x382100ac,0x4c00012c,0x7f45d378,0x3cc08130,0x80c6f000,0x7cc903a6,0x4e800420,0x7d232214,0x7ca92850,0x54aa01be,0x3d4a4800,0x7d43212e,0x7c0048ac,0x7c004fac,0x4e800020

[Wiimmfi Patch v1 for 4.3U]
maxversion=513
minversion=513
amount=2
require=wiimmfi-v1
offset=0x8137bfec
patch=0x4bf83044
offset=0x812ff000
patch=0x8137bff0,0x50726969,0x6c6f6164,0x65722070,0x61746368,0x20726576,0x20312028,0x342e3355,0x29202020,0x20202020,0x20202020,0x20202000
```

and results in the following log during boot: 

```
7:5:59 : Hacks:2
7:5:59 : applying Wiimmfi Patch v1 - hidden MASTER
7:5:59 : added offset to list 0x812FF030
7:5:59 : applying Wiimmfi Patch v1 for 4.3U
7:5:59 : added offset to list 0x8137BFEC
7:5:59 : added offset to list 0x812FF000
7:5:59 : launching sys menu... 0x9302F0A0
```

Adding another patch to the MASTER hack, making the first hack look like this ...

```
[Wiimmfi Patch v1 - hidden MASTER]
minversion=1
maxversion=65535
amount=2
master=wiimmfi-v1
offset=0x812ff030
patch=0x9421ff54,0x90010008,0x7c0802a6,0x900100b0,0xbc410014,0x7c6000a6,0x5464045e,0x7c800124,0x54638ffe,0x9061000c,0x3f608000,0x63634000,0x3c808090,0x3cc06874,0x60c67470,0x3ce0733a,0x60e72f2f,0x7c032000,0x4080003c,0x38630001,0x80a30000,0x7c053000,0x4082ffec,0x80a30004,0x7c053800,0x4082ffe0,0x39030004,0x39230003,0x8d480001,0x9d490001,0x2c0a0000,0x4082fff4,0x4bffffc4,0x63634000,0x3c808090,0x3d806e69,0x618c6e74,0x3e60656e,0x6273646f,0x3dc07769,0x61ce6669,0x3de02e6e,0x61ef6574,0x3e007769,0x6210696d,0x3e206d66,0x6231692e,0x3e406465,0x7c032000,0x408003c8,0x38630001,0x80a30000,0x7c056000,0x4082ffec,0x80a30004,0x7c059800,0x40a2ffe0,0x80a30008,0x7c057000,0x40a2ffd4,0x80a3000c,0x7c057800,0x40a2ffc8,0x92030000,0x92230004,0x92430008,0x3903000f,0x39230009,0x8d480001,0x9d490001,0x2c0a0000,0x4082fff4,0x4bffffa0,0x48000368,0x480003a1,0x60000000,0x68747470,0x3a2f2f63,0x612e6e61,0x732e7769,0x696d6d66,0x692e6465,0x2f636100,0x68747470,0x3a2f2f6e,0x61737769,0x692e7769,0x696d6d66,0x692e6465,0x2f616300,0x68747470,0x733a2f2f,0x6d61696e,0x2e6e6173,0x2e776969,0x6d6d6669,0x2e64652f,0x70700000,0x2c030000,0x40810040,0x8061000c,0x800d9738,0x2c000002,0x41820014,0x906da14c,0x38000002,0x900d9738,0x48000020,0x38830003,0x5484003a,0x88a40000,0x7ca42a14,0x7c0028ac,0x7ca803a6,0x4e800021,0x3860ffff,0x2c030000,0x00000000,0x808da14c,0x28040001,0x408100a0,0x9421ff80,0x38610020,0x3c00801d,0x600024f4,0x7c0803a6,0x4e800021,0x38610020,0x808da14c,0x38a00554,0x90bc00c4,0x3c00801d,0x60002544,0x7c0803a6,0x4e800021,0x38610020,0x38810010,0x3c00801d,0x600025f8,0x7c0803a6,0x4e800021,0x48000018,0x0fff1f07,0x00e638c9,0x49fbe [... cut ...]
7:5:26 : added offset to list 0x812FEE00
7:5:26 : applying Wiimmfi Patch v1 for 4.3U
7:5:26 : added offset to list 0x00000000
7:5:26 : launching sys menu... 0xE182E260
```

so apparently, adding that second patch to the MASTER hack causes the other hack to no longer apply correctly; and also the address for the sys menu is wrong. 

Do you have any idea why that happens?
It can't be the contents of the patch, as that code isn't even running at that point yet, so I'm suspecting another bug in the hack system ...

DacoTaco, 2020-06-19:
well, there goes my idea of it being a stable version...

ill investigate this later

DacoTaco, 2020-06-19:
before i investigate further : is it possible to reduce space on the hack?
you are pushing the limit on memory usage here a bit :/

Leseratte10, 2020-06-19:
You mean it's just running out of memory and crashes? I would have assumed that there's proper handling for running out of memory ...

Leseratte10, 2020-06-19:
If I remember correctly, Priiloader loads all the enabled hacks into an array, and the loader that loads the system menu then applies these hacks, right? And the priiloader gives a ptr to that array to the loader. 

What if Priiloader would store that array to the NAND (would only need to be done when the user changes the hack settings) in the proper, parsed binary format, and the System Menu loader could then read that file in parts? Meaning, load one hack, copy it to its destination, then load the next hack? Then you wouldn't need a large enough RAM buffer for all hacks.

DacoTaco, 2020-06-19:
well, i too would have expected it to give an error, but yes, its allocating memory around 0xE182E260 , something that most certainly is not normal memory ranges afair.

storing the array on nand would be stupid. we already have the hacks file on the nand. its better to load the hacks in the loader then. 
problem with that is that would increase loader size and complexity and make the loader basically a full on application instead of a ~3KB stand alone binary.... that is embed into priiloader.

im going to look into doing the allocation differently/more efficiently but i suggest you look into if you can make the hacks smaller if possible too.
welcome to embed device programming xD

Leseratte10, 2020-06-19:
Are you sure it is allocating memory there? If it allocated memory at that RAM address and tried to load the system menu to there, then how come the first two patches applied correctly, and for the second hack, A) only one of the two patches was applied and B) for the offset 0 instead of the correct one?

Also, maybe that "running out of memory" (if that is actually what causes the bug) is what caused the problems with the Wiimmfi patch for @Debs305 in the other issue (#222).

DacoTaco, 2020-06-19:
because the second hack is the final push it needs to start writing to the other address and all other allocations afterwards are even more bogus than the last.
yes im sure. i checked the addresses after each hack it adds to the list. it does a reallocate to add the new size of the allocated memory, and this is what is causing it to go to far.

imagine having a box of 15 items. you want to add 5 new items but can't, its full.
you need a bigger box, so you get a box of 20. yay! fits!
but what happened to the previous box? well, its disregarded somewhere in a corner, not really taking space but we aren't going to look at it again cause remember? it was to small.
but shit, now we need even more. a NEW bigger box! 
yuy, fits again!

and this keeps going and going with every offset hack. this is how it ends up allocating memory that isn't mem2.
sadly i didn't write the memory manager and this behaviour (disregarding previous boxes and not reusing that space) is not what i expected of it tbh.
i can fix it by first looking at how large the box needs to be, and i will, but eug. another for each loop...

Leseratte10, 2020-06-19:
I mean, even if you have to add another for loop, I imagine that will still be way more performant than having to call realloc all the time to get a larger blob of memory, no?
Maybe you could include that in any of the existing loops? I mean, when you parse the ini file you are going to go through the entire line of code anyways, might as well add a counter there.

DacoTaco, 2020-06-19:
could you give this a try? 
https://upload.dacotaco.com/boot_hacks_fix.dol

should have it fixed.
and no, the counter can't be added as is because you don't know if you have to include the master hacks or not until later when actually applying them. 
you could always assume they are enabled and count them in, but that sounds like a recipe for disaster tbh

Leseratte10, 2020-06-19:
Thanks for fixing, this version works. I will make a new PR with the changes to the Wiimmfi patch (as mentioned in my email in April) as soon as I ported it to the other regions.

closed 2020-06-19 by DacoTaco