DacoTaco/priiloader #257: IOS 80 detected as stub after updating to v0.9

25zcj01, 2020-09-22:
**Describe the bug**
I updated priiloader to v0.9, and, after the update it refused to load the home menu stating that Autoboot failed - IOS 80 detected as stub

**To Reproduce**
Steps to reproduce the behavior:
All I did was update the damn thing

**Version**
v0.9

**Expected behavior**
I expected it to autoboot the home menu, as that’s what I set it to autoload. It didn’t. Homebrew channel works just fine & I can launch titles, but it loads priiloader upon power on, saying that IOS 80 was detected as a stub.

DacoTaco, 2020-09-22:
do you mind running a sys check and posting the result here?
the excel file should show me why it detected it as a stub.

is this on a regular wii or wii mini?
what version of priiloader did you have before 0.9?

25zcj01, 2020-09-22:
It’s on a regular wii
I had v0.8.2 before installing 0.9
Not in a good spot to check settings/run syscheck atm but I’ll edit this post later

loquinator3000, 2020-09-25:
I have sort of the same problem. I was on 0.8.2 on a regular wii and updated to 0.9 and selecting "Installed file" gave me a DSI exception when wiiflow lite was installed. I tried installing Descent wii instead, and autobooting that worked fine, but corrupted my save file. Downgradingn to 0.8.2 after I post this syscheck
[sysCheck.xlsx](https://github.com/DacoTaco/priiloader/files/5280135/sysCheck.xlsx)

DacoTaco, 2020-09-25:
@loquinator3000 : your priiloader also said IOS80 was patched? or a different IOS?
and why is IOS80 patched?
as for the installed file, i believe we should make a different issue for that one

loquinator3000, 2020-09-26:
My bad, priiloader said nothing about my IOS. I don't remember doing anything to IOS 80 except for updating to 4.3U when it came out. Ill put my case in a new issue

25zcj01, 2020-09-29:
Sorry for the super long wait. I checked Priiloader settings, and sure enough ‘Use System Menu IOS’ was disabled. I have no clue why, I probably just fat-fingered the dpad, but thanks for the help @DacoTaco

closed 2020-09-29 by 25zcj01