DacoTaco/priiloader #258: Breakage of "Installed file" in v0.9

loquinator3000, 2020-09-26:
**Describe the bug**
Selecting "Installed file" gives a DSI exception when wiiflow lite was installed. I tried installing Descent wii instead, and autobooting that worked fine, but corrupted my save file.

**To Reproduce**
Installing Wiiflow Lite from the SD card to the autoboot makes a DSI exception when "Installed file" is selected, and returns to the homebrew channel.

**Version**
Priiloader 0.9

**Expected behavior**
Expected autoboot to the installed file without corrupting any files, as in v0.8.2

**Additional context**
[sysCheck.xlsx](https://github.com/DacoTaco/priiloader/files/5287412/sysCheck.xlsx)

DacoTaco, 2020-09-27:
could you upload the dol youre installing?
i want to compare it to what i have and use it for testing :)

loquinator3000, 2020-09-27:
right here! thanks :D
[boot.zip](https://github.com/DacoTaco/priiloader/files/5287898/boot.zip)

DacoTaco, 2020-09-27:
i'll be testing this later, but didn't your boot.dol come with a meta.xml or anything?

DacoTaco, 2020-09-27:
nvm, already fixed it lol.
i was flushing some parts of the dol incorrectly (bug when moving code to the new loader).
once i fixed that, the dol loaded (instead of freezing or crashing)

try installing this : 
[boot.zip](https://github.com/DacoTaco/priiloader/files/5288459/boot.zip)

DacoTaco, 2020-10-03:
poking @loquinator3000

loquinator3000, 2020-10-04:
sorry my bad!! I installed that and it works perfectly now. thanks awesome work!

closed 2020-10-04 by loquinator3000

loquinator3000, 2021-04-01:
are you planning to release this fix as 0.9.1 some time?

DacoTaco, 2021-04-01:
i'll add it to my todo list to release 0.9.1 sometime soon™.

DacoTaco, 2021-04-01:
it still works on your wii, right?

loquinator3000, 2021-04-01:
alright and yes it works great!