DacoTaco/priiloader #259: Runtime IOS patching with PriiLoader?

eku, 2020-10-21:
In order to use the alternatives for WiiConnect24, Nintendo's RSA key and the URLs of the service from IOS 31 and 80 would have to be patched. Can PriiLoader also patch the IOS or only the system menu?

DacoTaco, 2020-10-21:
in theory priiloader could patch system menu's IOS if you know what address it will always be in (or find a spot in memory thats free in which to place your own code that priiloader patches in like it does with wiimmfi patch).

what i don't understand is what system menu's ios would have to do with this?
all it does is start the channels/games does it not?

eku, 2020-10-22:
Then I did not express myself clearly enough. Can PriiLoader also patch IOS 31 and 80 and thus take over the function of [RiiConnect24-Patchers](https://github.com/RiiConnect24/RiiConnect24-Patcher), so that the IOS are not patched permanently but at runtime?

DacoTaco, 2020-10-22:
i think you kinda misunderstood me too :p 

Priiloader -CAN- patch the IOS that is running (it has AHBProt disabled giving it full hardware control), however it can only patch the ios that is selected to have system menu run on.

once system menu has started, priiloader has lost all control over the system.
if system menu would end up loading a different IOS for a channel, than priiloader can do nothing there.

why that tool is patching IOS80 is beyond me (i suppose for some sort of https ping to the servers done by system menu?) , but priiloader can not patch multiple IOS' . it can only (in theory) patch the IOS system menu is running on.

eku, 2020-10-22:
Thanks for the clarification.

closed 2020-10-22 by eku