DacoTaco/priiloader #260: Boot from Disc from Priiloader?

kdjmonaghan, 2020-10-22:
Hi,

I was wondering how difficult of a task this would be for you: 

Currently, I find myself going to the system menu every time when I want to boot a Gamecube game under the rate that it would normally run under Wii. This, over time, has lead me to a position of having to have AA batteries on hand due to the system menu not functioning without a powered Wiimote.

I realise this functionality in theory could be accomplished with homebrew applications such as Nintendont and other such loaders, however, they do multiple different things in their configuration to boot with other parameters set. E.g. Gamecube games in Nintendont are loaded with the Wii's additional hardware in play, and the Wii's clock speed, not downclocked to Gamecube performance. 

For games on the NAND (VC games) I'm finding myself directly booting from Priiloader to avoid the System Menu. This also goes for accessing any Homebrew tool I need via the Homebrew Channel shortcuts, or, installed file if I have an application installed... The only thing missing is booting from disc, in the same mannerism as booting from the Disc Channel.

You already have system menu hacks in place to region free this channel, so for a quality of life change I was wondering?:

How possible is it for you to create a 'forwarder' into the normal Disc Channel boot process from PriiLoader if a disc is present?

Thanks for reading / considering this.

DacoTaco, 2020-11-07:
i love this idea to be honest.
this would also be great to learn how disc booting is done.

however, there is a few questions/problems that i would have in the design of the functionality.
i think region can be ignored in the disc loading code i would write, but things like the wiimmfi patch for wii games i can not apply unless they provide me the code to do so (which i doubt they will?).

is this something youre ok with?

kdjmonaghan, 2020-11-08:
Just being able to boot from disc at all, regardless of wiimmfi patches or not, would be a huge quality of life change and greatly appreciated for me. So yes, I'd be perfectly fine with this.

Glad you also think this is a good idea. It's really one of the only things that I feel is missing from Priiloader

DacoTaco, 2020-11-08:
i feel like it would be a good idea cause at one point i also felt like Priiloader was like a 60% system menu replacement lol.
to make it even more of a system menu replacement, disc booting would need to be included lol.
i can't promise when i'll add this though and to what version ( 0.9.1 or 0.10.0 ), as i will need to look into how disc booting works (i know they have an apploader but i don't know much is needed to load them and if i need to do more than just that)

kdjmonaghan, 2020-11-09:
I believe GeckOS or NeoGamma might be the best place to look at how it can be done as those applications have natural booting into Disc, from what I recall and don't enable the additional memory and higher clock speed the Wii has, booting into Gamecube mode correctly.

Additionally also, GeckOS has the ability to load cheats at title launch - is this something you could do in the 'hacks' menu? I realise this is a separate functionality request, but, figured I'd ask here before making another issue. Some VC games have gecko codes applied to them for controller rebinding or third party adapters such as Raphnet N64 Controller to Wii to enable a custom remapping in a .gct file.

Not sure how this is done in GeckoOS but it's probably pretty simple.

I believe you can find the source code for these online, on github these days for reference. Can link them if you need them.

DacoTaco, 2020-11-09:
maybe in the long run we could implement that, but its better to start with the basics :) 

i do have the geckoOS source somewhere as i helped somebody build it with a modern devKitPPC toolkit (as it should be done).

TheTechRobo, 2020-11-30:
Aren't you already able to do this with the launch title option?

kdjmonaghan, 2020-11-30:
That's launching an installed channel. You cannot load the disc channel from Priiloader currently.

TheTechRobo, 2020-12-01:
Ah, thanks for clarifying.

DraconicNEO, 2021-01-30:
Being able to Autoboot discs would be a good feature to have, especially since it would allow playing GameCube games on a Wii with nothing but a Wii and GameCube Controller.

DacoTaco, 2021-01-30:
little update : i am currently working on booting discs.
GC works completely (thanks wii for having a title that handles booting GC discs! lol )
Wii discs don't work yet and im working on those.

autobooting a game disc sounds like a weird option to me?
it would mean that the wii will start the game thats inserted every time its powered on. no other channels, no HBC options, no nothing.
is this what you want?

kdjmonaghan, 2021-01-30:
No, I'd prefer to just have a menu option inside priiloader under System Menu for boot disc tbh. Autoboot sounds like a setting/hack option... would not want it by default.

Great work with progress though, sounds promising!

DacoTaco, 2021-01-30:
i just made the first commit on the feature branch, you can check it out.
its based on marcan's tinyload, but something isn't right yet. launching the wii game's binary results in a system crash :/
tinyload works, so im doing something wrong haha

DacoTaco, 2021-01-31:
so, most of game launching works from what i can tell.
here is a dol you can load in homebrew channel to try out some games.

GC should work perfectly, Wii games... depend on the game.
i've thrown kirby epic yarn, monster hunter tri, smash bros brawl and other m against it.
kirby's epic yarn and monster hunter booted perfectly, smash and other m did not.
i suspect it to be some issue with the fact those games are dual layer, but im not sure.

i find it weird that geckoOS can launch them, but i can not and my logic should be the same... :/ 

but hey, give it a go and let me know how it went :)

[priiloader.zip](https://github.com/DacoTaco/priiloader/files/5900097/priiloader.zip)

kdjmonaghan, 2021-01-31:
I have a few NTSC-J games and NTSC-U gamecube games laying around - I'll try them out.

I normally have an NTSC Signal so I'll give it a try. Might want to add an option to define video signal if possible, maybe, if it's an issue?

Will try out a few games and write up my findings.

DacoTaco, 2021-01-31:
thanks!

depending on what those ntsc games do, i'll start writing the video (re)initialization code and any issues that come up :)

kdjmonaghan, 2021-01-31:
[image]
I get this when I try to boot my Ocarina of Time: Master Quest disc (Japanese)... My video is set to NTSC-J 480p when I'm trying to boot it, this comes up before the progression scan screen.

This doesn't happen when booting from the disc channel. 

I have region changed my wii to japanese though, which might be why - I'll play around with it. But this error screen is consistent.

DacoTaco, 2021-01-31:
fincs translated the messagebox, its the generic 'error occured, turn off the system' error message. 
could be because of the video mode i guess, but i would need to test it somehow. i had thought that BC (the wii title that launches gamecube discs) would set the video mode correctly hehe

kdjmonaghan, 2021-01-31:
Gamecube Games:
The Legend of Zelda: Collectors Edition (NTSC-J) - boots, no problems, prompts for 480p
The Legend of Zelda: Twilight Princess - boots, seems fine, prompts for 480p
The Legend of Zelda: Ocarina of Time (Master Quest) (NTSC-J) - Generic Error Message...
The Legend of Zelda: Ocarina of Time (Master Quest) (NTSC-U) - Generic Error Message...
Need For Speed Underground 2 (NTSC-U) - boots, no problems, prompts for 480p
Paper Mario: The Thousand Year Door - boots, no problems, prompts for 480p

None of my PAL games boot when on Progressive Scan initially, but they boot from the Wii System Menu. This might be something to do with my Wii tbh as it has a WiiDual installed to it's motherboard. 

Wii Games: (All do not prompt for progressive scan, preferably would like this)
Super Smash Bros Brawl. (NTSC-U) - boots, 480i even when in 480p
Animal Crossing City Folk (NTSC-U) - boots, 480i even when in 480p
Super Mario Galaxy (NTSC-U) - boots, 480i even when in 480p

So something seems unique about Master Quest so far... not sure what's up there. If you don't own it can you burn it to a DVD and see if you can figure out what's wrong? Not got another game that's doing it atm. (Also does this when set to Interlaced, 480i)

DacoTaco, 2021-01-31:
alright, thanks!

looks like overal most things boot up at least, which is good!
most other notes mentioned seem to be video mode not getting set before launching the titles, except Master quest which is acting special.

i tried booting some stuff in dolphin, but half the stuff ive tried crash in dolphin ... >_<
then again, my PAL Ocarina of time copy flat out crashes on my wii too...
less fun is that it crashes in GeckoOS too, meaning its a bug i can't figure out by looking at other code.

i'll investigate further some other time :)
we are getting there... slowly xD

first up will be the video mode, which might get some things going.
after that ill look into the booting issue i got

DacoTaco, 2021-02-06:
update : i got my PAL copies to boot. i had to add a few things to get them to work : 
* set drive (audio) streaming mode
* read BI2 from disk so the internal drive state is correct
* Set video mode (+ set video mode in sram)

however, a libogc patch is required and im talking to wintermute to have it integrated. 
plus side : we can now boot more gc games then geckoOS and probably a lot of applications that can boot GC discs hehe

kdjmonaghan, 2021-02-06:
Nice, if you want me to test the copies of Master Quest (USA/NTSC-J) that failed before, let me know.

DacoTaco, 2021-02-06:
OOT/Master quest boot.... in dolphin.
on my wii they still give the general error, which suck

kdjmonaghan, 2021-02-06:
Strange... I wonder why... anti piracy measures? Not sure... really strange.

DacoTaco, 2021-02-07:
what insomnia isn't good for.
figured it out. dolphin wasn't picky about the audio config, but an actual wii was.

### **_Bootmii @ Boot2 recommended_**
### **_PLEASE ENABLE DUMPING GECKO OUTPUT IN THE SETTINGS FIRST._**
this dol is experimental and changes system video configuration before booting gamecube games. i have no idea how a wii will react if you start mixing regions (and therefor having set ntsc video signal on a pal wii or vice versa). 
this dol dumps the video signal mode before changing it, so having the log means that worst case we know the initial value and can set it back using a tool.

with that out of the way, i was able to boot zelda OOT & WW (PAL copies using a PAL wii) now so it should work perfectly.
sadly i don't have any non-pal games :/ 
[priiloader.zip](https://github.com/DacoTaco/priiloader/files/5938433/priiloader.zip)

kdjmonaghan, 2021-02-07:
I tried it there and master quest now boots properly. I'll try a few games later / tomorrow and let you know how those go.

Seems like it works though! Well done.

Just have to ensure that 480p is respected when booting Wii games now.

DacoTaco, 2021-02-07:
what do you mean, 480p has to be respected? ( didn't set video mode for wii games yet in that dol btw)

TheTechRobo, 2021-02-07:
Would help test - I have JustDance 4 and 2017, Mario Kart Wii, NSMBW, Mario Power Tennis, and a few others - but not allowed to mod the Wii. Once i get a new computer, if testers are still required, I'll try everything in dolphin. But right now I have no machine good enough, heh.

kdjmonaghan, 2021-02-07:
> what do you mean, 480p has to be respected? ( didn't set video mode for wii games yet in that dol btw)

Ah, unlike Gamecube games, you do not get a prompt for progressive scan. With component/hdmi you can force it with NTSC Gamecube games by holding B on boot for the prompt (like 60hz prompts in PAL).

Just so long as the Wii remains 480p when booting a wii game, is what I meant by respecting the setting defined, though, sounds like you are handling this also. :) 👍

DacoTaco, 2021-02-07:
> Would help test - I have JustDance 4 and 2017, Mario Kart Wii, NSMBW, Mario Power Tennis, and a few others - but not allowed to mod the Wii. Once i get a new computer, if testers are still required, I'll try everything in dolphin. But right now I have no machine good enough, heh. Pentium Dual-Core 2.1Ghz (computer mid 2000s) for the win
hehe. you can test it in dolphin if you want. i suspect it will run at least the same as on hardware, if not better (because some things are ignored by the emulator, but most are identical)

> Ah, unlike Gamecube games, you do not get a prompt for progressive scan. With component/hdmi you can force it with NTSC Gamecube games by holding B on boot for the prompt (like 60hz prompts in [...]

a ye, that might be a different flag that controls that. i don't touch that flag.
i was setting the PAL60 flag (that GC uses) in a test earlier this weekend, but it kept setting my system (even system menu) in PAL60 which my capture device doesn't like hehe and removed it again.

it does change the sram video flag (a different flag from PAL60), which i have no idea if that controls anything on the wii outside of GC games. thats what the warning is for hehe. 
i suspect it doesn't do anything to wii code as the sram was set to NTSC on my PAL wii when i first looked at it ( 0 = NTSC )

im curious how games (wii and gc games) from a different region will act tbh :')

kdjmonaghan, 2021-02-07:
Update:

OK so I can't get it to boot MQ again and it either changes my video setting to 60hz or 50hz, depending on what region I'm on?

I'll do some more thorough tests tomorrow, but seems like it has an issue. 

I'll enable gecko os dump tomorrow also when testing.

DacoTaco, 2021-02-07:
alright, thanks!

TheTechRobo, 2021-02-08:
Sounds good, I’ll try it in dolphin once I get a new pc. 

Thanks for all this work on priiloader :D

DraconicNEO, 2021-02-08:
[Here](https://github.com/iGlitch/MinimaLauncher) is the source code for Minimalauncher in case that might be helpful for implementing Disc booting in Priiloader (I'm not saying people should Just Use this instead, that doesn't feel like a proper autoboot solution)

DacoTaco, 2021-02-08:
> 
> 
> Sounds good, I’ll try it in dolphin once I get a new pc.
> 
> Thanks for all this work on priiloader :D

no problem. i learned a lot from this, and with it [libogc also got a little bit better hehe](https://github.com/devkitPro/libogc/commit/bc4b778d558915aa40676e33514c4c9ba2af66b8)

> 
> 
> [Here](https://github.com/iGlitch/MinimaLauncher) is the source code for Minimalauncher in case that might be helpful for implementing Disc booting in Priiloader (I'm not saying people should Just Use this instead, that doesn't feel like a proper autoboot solution)

has this code been tested with Ocarina of time (GC) ? 
cause as is, im not seeing any code that configures the DVD player's audio streaming buffer, which is required for (at least) that game ([see the 3 things i had to implement to get it to work](https://github.com/DacoTaco/priiloader/issues/260#issuecomment-774471358))

> 
> 
> Update:
> 
> OK so I can't get it to boot MQ again and it either changes my video setting to 60hz or 50hz, depending on what region I'm on?
> 
> I'll do some more thorough tests tomorrow, but seems like it has an issue.
> 
> I'll enable gecko os dump tomorrow also when testing.

im curious to all the results of your testing though :)

kdjmonaghan, 2021-02-09:
> no problem. i learned a lot from this, and with it [libogc also got a little bit better hehe](https://github.com/devkitPro/libogc/commit/bc4b778d558915aa40676e33514c4c9ba2af66b8)
>
>
> has this code [...]

ok so the first time I booted Master Quest NTSC-J (without gecko on) it booted... now it doesn't boot and tries to set the video to 576i and doesn't display a video. How strange. My Wii's native video feed is PAL but it is region changed to NTSC-J and video setting set to 480p (before boot) to get MQ to work normally. As said before, this works on the disc channel.

Otherwise, same as before.

DacoTaco, 2021-02-09:
could you try changing video mode in the wii system menu settings (to something else) and back again?
there should also be some gecko output in prii.log on the sd once you enable that setting about changing video mode, mind sharing that too?

kdjmonaghan, 2021-02-09:
```
--------gecko_output_enabled------
--------gecko_output_enabled------
--------gecko_output_enabled------
--------gecko_output_enabled------

```

Only contents.

Tried on 480i, 480p, 576i... same result.

DacoTaco, 2021-02-09:
grmz, thats odd and very annoying.
[the code has gprintf's about video](https://github.com/DacoTaco/priiloader/blob/LaunchDisc/priiloader/source/main.cpp#L1240-L1302) so no idea why nothing is in there. not even the 'reading dvd...' line is there...

i assume the tv says 'Loading DVD' and then goes black? 
i'll look into the whole 480p flag when i can (iirc its a different flag than the PAL60 flag)

kdjmonaghan, 2021-02-09:
First reading dvd is pal 480p
Second is NTSC-J 480p

```
--------gecko_output_enabled------
17:51:40 : reading dvd...
17:51:45 : GC disc detected
17:51:46 : video mode : 0x00000000
17:51:46 : PAL50
17:51:47 : new video mode : 0x00000001
17:51:47 : booting BC...
Y4Ÿ8"†¶Ñ ûž}_ þJ TŠ7 ô¨<>éLË
--------gecko_output_enabled------
17:53:49 : BootState:255
17:53:50 : Bootstate 255 detected. DiscState 2 ,ReturnTo 0 & Flags 129 & checksum 2180973056
17:53:50 : Input_Init():1
17:53:51 : Bootstate 255 detected. DiscState 2 ,ReturnTo 0 & Flags 129
17:53:52 : FAT_Init():1
--------gecko_output_enabled------
17:54:03 : reading dvd...
17:54:09 : GC disc detected
17:54:09 : video mode : 0x00000000
17:54:10 : PAL50
17:54:10 : new video mode : 0x00000001
17:54:11 : booting BC...
--------gecko_output_enabled------
17:54:23 : BootState:255
17:54:23 : Bootstate 255 detected. DiscState 0 ,ReturnTo 0 & Flags 130 & checksum 2197749760
17:54:24 : Input_Init():1
17:54:25 : Bootstate 255 detected. DiscState 0 ,ReturnTo 0 & Flags 130
17:54:25 : 255:System Menu
17:54:26 : using 0000009B for booting
17:54:27 : loading hacks
17:54:27 : input shutdown
```

DacoTaco, 2021-02-09:
NTSC-J -> PAL50 ? that doesn't sound right lol.

kdjmonaghan, 2021-02-09:
I wipeds the log and did a few tests.

Here's my log
[prii.log](https://github.com/DacoTaco/priiloader/files/5954497/prii.log)
:

DacoTaco, 2021-02-09:
ye, it's hella slow because of all the gecko debugging output its dumping to SD, sorry for the scare. i should've warned you !
it should have set it to NTSC, not PAL50. that might be where the issues lies though.
you can disable the setting btw, i think ive seen enough to do some code changes hehe :)

DacoTaco, 2021-02-09:
i think i got it figured out.
configuring the audio buffer for audio streaming clears the Disc ID from memory, but not memory's cache.
this might be why it worked once (value's were still cached) but not after that.
try this one, it saves the ID in memory and uses it to configure the video signal.

[priiloader.zip](https://github.com/DacoTaco/priiloader/files/5954772/priiloader.zip)

kdjmonaghan, 2021-02-10:
Looks like it's working. Booted twice fine now and successfully changed to 480p both times at the prompt.

[prii.log](https://github.com/DacoTaco/priiloader/files/5960591/prii.log)
[image]

DacoTaco, 2021-02-10:
finally!

thats very satisfying to see!
the log also confirms it detected it as NTSC-J , so thats perfect! :)

also damn, that menu looks better on the japanese version haha
all i have to do now is wait for libogc to get a new version before i can merge my branch and have it set for the next version.

everything works now? wii games, gamecube games, other regions,... ?

kdjmonaghan, 2021-02-11:
Everything seems to work from my testing. 👍

DacoTaco, 2021-02-13:
feature has been merged into master and will be released in the next major release

closed 2021-02-13 by DacoTaco

TheTechRobo, 2021-02-13:
can't wait!

DacoTaco, 2021-02-13:
you can use it in dolphin!
oh, wait. useless feature there xD

kdjmonaghan, 2021-05-25:
@DacoTaco do you have a release schedule listed anywhere? Just curious when you plan to do the next major release for this feature. Thanks.

DacoTaco, 2021-05-27:
sorry, i don't have a schedule listed anywhere. i usually do thinks ad-hoc. currently my life is a bit chaotic as im in the middle of switching jobs.

there is also a different project taking more of my time (god i want to talk about this project so bad... xD ).

I had wanted to rework the sd/usb access before releasing a new major version so that i'd have the 2 features together, but that rework is currently on hold...