DacoTaco/priiloader #263: Launch Title DSI after installing Zelda OOT Randomizer

TyMiles2012, 2020-11-07:
**Describe the bug**
After installing an OOT Randomizer WAD, selecting Launch Title on the Priiloader menu results in a DSI exception.  Otherwise, the WAD works perfectly fine on WiiMenu and USBLoaderGX

**To Reproduce**

1.  Install OOT Randomizer WAD
2.  Reboot Wii into Priiloader
3.  Select Launch Title

**Version**
0.9 Stable

**Expected behavior**
Install OOT Randomizer, go back to Priiloader and select Launch Title, launch OOT Randomizer WAD.

**Screenshots**
[image]

**Additional context**
So far, this is the only WAD I've installed that consistently done this.  I have other patched OOT WADs like the practice ROM that work just fine.  Randomizer is the only WAD that causes this.

DacoTaco, 2020-11-07:
thats odd, but i'll investigate.
could you share the wad?
that way i can install it myself and check.

TyMiles2012, 2020-11-07:
I ended up resolving the issue by restoring a fresh NAND and starting a new softmod from scratch.  I don't know if this was caused by user error in the past, but considering this is a modified WAD, do you still want it just in case?

DacoTaco, 2020-11-07:
ye, i'd still like to check it out.
it might have been something odd when updating the play recordings (the log that keeps track of what you played on what day), and i want to see if it does something weird on my wii. just in case :)

TyMiles2012, 2020-11-07:
Ah okay.  Here you go then.  Before I reflashed my NAND, any random seed I would generate would cause this issue.  The OOT Randomizer devs tested a WAD I sent them I installed that caused the issue and they said it was fine when they looked at it.  WAD was too big to upload here so I hosted it on google drive.

https://drive.google.com/file/d/1s8ZD0PEVIrNSbV2R_hrUnCH6i2DbIch0/view?usp=sharing

DacoTaco, 2020-11-08:
ye, seeing the address of the crash im pretty sure it was priiloader crashing , but the problem is where did it crash and why :)

chances are i'll close this later, but not after having tested it on my wii :p 
thanks for the wad

DacoTaco, 2020-11-08:
just installed it, and it ran fine on my wii sadly.
if you were to have the issue again please report it so we can look at it.
for now im closing this.

also, shit that emulator does not run well on my wii haha.

closed 2020-11-08 by DacoTaco