DacoTaco/priiloader #264: DOLs that require AHBPROT no longer autoboot correctly

DongVGH, 2020-11-28:
I recently updated from 0.8.2 to 0.9 (also tried beta 4), and everything works very well (love the USB keyboard and power/reset button support, thanks for those!), but Nintendont doesn't autoboot as it should anymore. It shows a screen that says it's not granted AHBPROT support, and reboots after 10 seconds, despite the same DOL booting just fine back in 0.8.2.

Tried re-installing and using a Nintendont forwarder instead, both the original FIX94 one and jmlee337's update, neither of which change anything besides a longer black screen time before the same message displays. Rolling back to 0.8.2 works fine again, as expected, but giving up the ability to boot into DOSBox or SRB2Wii is a bit too much!

DacoTaco, 2020-11-28:
huh, odd.
the meta.xml argument to disable ahbprot should work.

i'll expand my argument tester app a bit and do some testing.

DacoTaco, 2020-11-28:
i just did a test on my wii and the tester shows everything working with the current (under construction) 0.9.1
[image]

ive also reverted to 0.9 and it was working there too.

i have been able to reproduce the issue however by replacing the meta.xml with theirs.
something in there is probably upsetting the priiloader xml parser

DacoTaco, 2020-11-28:
fixed it.

please use the provided installer to install a fixed priiloader version and verify if it works 
[boot.zip](https://github.com/DacoTaco/priiloader/files/5610340/boot.zip)

DongVGH, 2020-11-28:
Fix works perfectly, thank you. This also fixed another bug I noticed after going offline yesterday, where in the Load/Install File menu the last entry would be garbled and Priiloader would code dump a couple seconds after having it on-screen.

DacoTaco, 2020-11-28:
ye, that bug has also been reported and fixed so far.
i think i need to release 0.9.1 at one point xD

closed 2020-11-28 by DacoTaco