DacoTaco/priiloader #265: Code Cleanup

urherenow, 2020-11-29:
There seems to be a few unused variables floating around. I'm not a real programmer, but doesn't cleaning these out save space?

/home/urherenow/projects/priiloader/Installer/source/main.cpp: In function 's8 PatchIos(s8)':
/home/urherenow/projects/priiloader/Installer/source/main.cpp:1392:5: warning: unused variable 'downgrade_fix' [-Wunused-variable]
 1392 |  u8 downgrade_fix[] = { 0x20, 0xEE, 0x00, 0x40, 0x18, 0x2B, 0x1C, 0x3A, 0x32, 0x58, 0x88, 0x19, 0x88, 0x13, 0x42, 0x99, 0xD2 };
      |     ^~~~~~~~~~~~~
/home/urherenow/projects/priiloader/Installer/source/main.cpp:1395:5: warning: unused variable 'trucha_hash1' [-Wunused-variable]
 1395 |  u8 trucha_hash1[] = {0x20, 0x07, 0x23, 0xA2};
      |     ^~~~~~~~~~~~
/home/urherenow/projects/priiloader/Installer/source/main.cpp:1396:5: warning: unused variable 'trucha_hash2' [-Wunused-variable]
 1396 |  u8 trucha_hash2[] = {0x20, 0x07, 0x4B, 0x0B};
      |     ^~~~~~~~~~~~
/home/urherenow/projects/priiloader/Installer/source/main.cpp:1399:5: warning: unused variable 'identifyCheck' [-Wunused-variable]
 1399 |  u8 identifyCheck[] = {0x68, 0x68, 0x28, 0x03, 0xD1, 0x23};

DacoTaco, 2020-11-29:
hehe, i know about these ye.
hehe, i know about them.

they are unused when patching the running IOS , and therefor don't influence the installer's dol size (because gcc doesn't include them in the binary) but it is cleaner for them to not be there.

however, they are actual IOS patches and i don't know what to do with them. i should delete them, but then i don't have them on my computer if i were to ever need them. :/ 
they are a bit of a storage place at this point xD

urherenow, 2020-11-29:
Put them in a comment(s) in the code. Problem solved.

DacoTaco, 2020-11-29:
i solved it differently :P 

i removed them from the installer code and moved it to my own channel code (which ive had on my pc for a long time and is based on priiloader ).
the channel code is now also on git for shits and giggles

closed 2020-11-29 by DacoTaco