DacoTaco/priiloader #266: Is it safe to install Priiloader on my Wii?

acc-testerino, 2020-12-08:
The Wiki states that it might brick the console if I have custom ISOs, which I'm pretty sure I do, and it also says that if I have any questions I could ask on github, so here I am.

This is my syslog
```
SysCheck HDE v2.4.0 HacksDen Edition by JoostinOnline, Double_A, R2-D2199, and Nano
...runs on IOS58 (rev 6176).

Region: NTSC-U
System Menu 4.3U (v513)

Drive date: 12.18.2008
Homebrew Channel 1.1.2 running on IOS58

Hollywood v0x21
Console ID: :)
Console Type: Wii
Shop Channel Country: Brazil (16)
Boot2 v4
Found 90 titles.
Found 49 IOS on this console. 10 of them are stubs.

IOS3 (rev 65280): Stub
IOS4 (rev 65280): Stub
IOS9 (rev 1034): No Patches
IOS10 (rev 768): Stub
IOS11 (rev 256): Stub
IOS12 (rev 526): No Patches
IOS13 (rev 1032): No Patches
IOS14 (rev 1032): No Patches
IOS15 (rev 1032): No Patches
IOS16 (rev 512): Stub
IOS17 (rev 1032): No Patches
IOS20 (rev 256): Stub
IOS21 (rev 1039): No Patches
IOS22 (rev 1294): No Patches
IOS28 (rev 1807): No Patches
IOS30 (rev 2816): Stub
IOS31 (rev 3608): No Patches
IOS33 (rev 3608): No Patches
IOS34 (rev 3608): No Patches
IOS35 (rev 3608): No Patches
IOS36 (rev 3608): Trucha Bug, ES Identify, NAND Access
IOS37 (rev 5663): No Patches
IOS38 (rev 4124): No Patches
IOS40 (rev 3072): Stub
IOS41 (rev 3607): No Patches
IOS43 (rev 3607): No Patches
IOS45 (rev 3607): No Patches
IOS46 (rev 3607): No Patches
IOS48 (rev 4124): No Patches
IOS50 (rev 5120, Info: rev 4889): No Patches
IOS51 (rev 4864): Stub
IOS52 (rev 5888): Stub
IOS53 (rev 5663): No Patches
IOS55 (rev 5663): No Patches
IOS56 (rev 5662): No Patches
IOS57 (rev 5919): No Patches
IOS58 (rev 6176): USB 2.0
IOS60 (rev 6400, Info: rev 6174): No Patches
IOS61 (rev 5662): No Patches
IOS62 (rev 6430): No Patches
IOS70 (rev 6912, Info: rev 6687): No Patches
IOS80 (rev 6944): Trucha Bug
IOS202[60] (rev 65535, Info: Hermes-v55.1): Trucha Bug, NAND Access, USB 2.0
IOS222[38] (rev 65535, Info: Hermes-v55.1): Trucha Bug, ES Identify, NAND Access, USB 2.0
IOS223[37] (rev 65535, Info: Hermes-v55.1): Trucha Bug, NAND Access, USB 2.0
IOS224[57] (rev 65535, Info: Hermes-v55.1): Trucha Bug, NAND Access, USB 2.0
IOS249[56] (rev 21007, Info: d2x-v7): Trucha Bug, NAND Access, USB 2.0
IOS250[57] (rev 65535, Info: d2x-v7): Trucha Bug, NAND Access, USB 2.0
IOS254 (rev 65281): BootMii
BC v6
MIOS v10
Report generated on 12/04/2020.
```

Is it safe to install Priiloader without bricking it?

DacoTaco, 2020-12-08:
hehe, thanks for the sysinfo and question.
i don't see these often :p 

i see some patches but it should be fine.
over the years ive added bugs and fallbacks that should catch any potential problems (like the installer running on an IOS it can't rely on ).

feel free to install and enjoy priiloader!

if you don't trust it, you can always take a nand backup with bootmii beforehand, which can be used to restore your wii :)

closed 2020-12-08 by DacoTaco

acc-testerino, 2020-12-08:
Cool, thanks for the quick answer!

I do have a nand backup, but unfortunatelly I can't install miiboot as boot2 on my Wii, that's why I want to install Priiloader. But yeah, if it things go south I probably won't be able to afford a nand programmer

DacoTaco, 2020-12-08:
if things oddly do go south (like if you were to somehow find a copy of priiloader 0.5), you could try and find somebody that can restore the nand. probably cheaper :P

but ye, chances of it going south is very very very low :)