DacoTaco/priiloader #267: Define Network Settings from Priiloader

kdjmonaghan, 2021-01-18:
Had an instance recently where I realised no homebrew application exists to set the Wii Network Settings without a wiimote - to the point where the person in question without the sensor bar, decided to use burning paper towels (due to a lack of candles) to generate IR lights to navigate to the settings. (successfully, I might add).

Is this something that could possibly be added in a future revision? Being able to set your network access point within Priiloader?

Thanks.

TheTechRobo, 2021-01-18:
> decided to use burning paper towels (due to a lack of candles) to generate IR lights to navigate to the settings. (successfully, I might add).

I won't even comment on the safety of that. :P 

But yeah, good feature request, especially considering that Daco said that he eventually wants it to be a sort of replacement to the system menu...kind of.

DacoTaco, 2021-02-10:
information to myself : 
found the network settings on wiibrew : https://wiibrew.org/wiki//shared2/sys/net/02/config.dat
more research and testing is needed though