DacoTaco/priiloader #268: Load Priiloader as a Channel (for vWii)

DraconicNEO, 2021-01-30:
Hello I was wondering if there was a way to load Priiloader as a Channel on vWii due to the inability to install to the System menu. I saw that you did it in #220 so it looks possible. Most people typically use Priiloader for its ability to load patches and vWii can be unbricked from Wii U mode.

Obviously things like locking the system menu wouldn't work since you have to load Priiloader Manually and BootMii isn't functional on vWii period.

TheTechRobo, 2021-01-30:
"vWii can be unbricked from Wii U mode."

Really? I don't have a Wii U but I thought that when it's completely bricked you're screwed.

DraconicNEO, 2021-01-30:
> 
> 
> "vWii can be unbricked from Wii U mode."
> 
> Really? I don't have a Wii U but I thought that when it's completely bricked you're screwed.

yes there's actually a guide to unbrick the vWii from in Wii U mode (requires haxchi or similar exploit) [Here](https://www.youtube.com/watch?v=Ay8KH3kK4_A) is a link to the guide (yes it's a video unfortunately, But I have little doubts this will ever change due to the stagnation of the homebrew community on Wii U). You can also use your own nand backup if you have one.

It's important to note that while this can fix vWii bricks, it cannot fix Wii U bricks. If you somehow bricked on the Wii U side it's not fixable.

TheTechRobo, 2021-01-30:
I did know the difference between a vwii brick and a wii u brick, but i had no idea you could fix vwii from wii u menu

DacoTaco, 2021-01-30:
@TheTechRobo , he's right. ive done some vWii modding using ftpii in wii u mode. technically, if you know whats wrong you can fix it from wiiu mode.

@DraconicNEO : what advantage would you have from having a priiloader channel? it can't patch system menu since it can't load it (its encrypted). is the public decryption key that the cpu uses to decrypt it public? cause without it nothing can touch system menu. that screenshot was generated by replacing homebrew channel's binaries with priiloader , but in theory it can be made its own channel.

DraconicNEO, 2021-01-30:
> @DraconicNEO : what advantage would you have from having a priiloader channel? it can't patch system menu since it can't load it (its encrypted). is the public decryption key that the cpu uses to [...]

Oh Originally I thought it could be possible to apply temporary patches after boot, I don't know why but I thought that it could somehow work.

Well as for the keys The vWii keys were made public, quite a while ago actually. Actually not Just the vWii keys, both the vWii and Wii U keys.

Taken from [Page on WiiUBru](https://wiiubrew.org/wiki/Encryption_Keys)

vWii Ancast Key:
`ce3641b2660253f5a7e789db297be2c1585b3054`
Used to decrypt vWii System Menu and new NANDloader binaries at load time. 

Wii U Ancast Key:
`2ba6f692ddbf0b3cd267e9374fa7dd849e80f8ab`
Used to decrypt the Cafe OS kernel at load time.

I don't know if there's some other issue with doing it but we do have the keys. So it doesn't seem like encryption is one of them.

DacoTaco, 2021-01-30:
oh, having the keys might change some things actually.
thats... interresting :)

indianaliam1, 2021-06-26:
Great, this got buried. As a "I left two Wiis at my Grandmas for Wii Sports and I would feel bad if I took them back to run homebrew on it" club member, I think that looking into this would be interesting. Now, I'm a vita guy primarily, so I wouldn't know the half of it, but hey, it'd be cool to see.

QuickBellHacker, 2021-07-10:
hello @DacoTaco how are you : ) can you please add vWii ancast key into that dol file please thank you if you can

[priiloader.zip](https://github.com/DacoTaco/priiloader/files/6795412/priiloader.zip)

DraconicNEO, 2021-07-10:
> 
> 
> hello @DacoTaco how are you : ) can you please add vWii ancast key into that dol file please thank you if you can
> 
> [priiloader.zip](https://github.com/DacoTaco/priiloader/files/6795412/priiloader.zip)

That's the version you modified to run on vWii right?

TheTechRobo, 2021-07-10:
@QuickBellHacker  It's a bit more than adding the keys, extra code needs to be added to Priiloader if I'm correct.

indianaliam1, 2021-07-10:
well that's obvious, idiot

TheTechRobo, 2021-07-11:
Gee, thanks. I wasn't talking to you, either.

DraconicNEO, 2021-07-11:
> 
> 
> well that's obvious, idiot

If you're not going to contribute to this in a meaningful way then don't bother commenting.

DraconicNEO, 2021-09-09:
> 
> 
> Bump?

Currently QuickBellHacker is taking a break from priiloader on vWii. Don't know when he'll get back to it.

TheTechRobo, 2021-09-09:
Isn't Daco the developer?

DacoTaco, 2021-09-09:
i am. last time i checked i am xD

i want to look at this, but i have a hard time finding time and motivation to do so..

TheLegendOfXela, 2021-09-09:
Sending all my love and support. I hope you can find the time to finish this! It would be beneficial to do many. Thanks for all you’ve done!

DraconicNEO, 2021-09-11:
> Isn't Daco the developer?

He is, but last time I checked. Quick Bell Hacker was the one who got Priiloader Booting on vWii, although he still hasn't said how he did it yet.

DacoTaco, 2021-09-11:
he did? any link to where he/she has shown this? 
i mean, in theory i have gotten priiloader running on vWii myself (by replacing HBC's dol with priiloader) but i didn't know anyone else did this.

TheTechRobo, 2021-09-11:
All I've seen is them asking for Daco to patch their binary for this to work, but neo might be in dms or smth

TheLegendOfXela, 2021-09-11:
If/when PriiLoader is on the Wii U, please add the ability to disable the vFilter, and enable the WC24 flags. Would it also be possible to boot into a different resolution than what the Wii U is set to?

DraconicNEO, 2021-09-12:
> All I've seen is them asking for Daco to patch their binary for this to work, but neo might be in dms or smth

He showed a teaser on YouTube and talked about it on his server, but he didn't actually say how he did it.

DraconicNEO, 2021-11-01:
> 
> 
> @DraconicNEO this?
> 
> https://www.youtube.com/watch?v=HGDDAq5c98c

Yeah I'm pretty sure that's it.

Ploggy, 2021-11-01:
> i think thats the video, ye

So, the main reason I'd want Priiloader on vWii is there's a new wip HB called Fakemote. 
Fakemote is made by Xerpi and allows hid controllers to work On Wii (Bluetooth support is planned) Currently it works with Usbloader and WiiFlow because you can choose the IOS before loading a game, From what I understand with Priiloader you can choose an IOS to boot the system from? I Understand modifying a System IOS (especially on vWii is a big no no?) :P 

But if Priiloader can be added to vWii we could have systemwide hid support that will work from booting vWii ?(in theory)?

DacoTaco, 2021-11-01:
hehe, fakemote is getting traction quickly huh. thats nice to see ! (he contacted me some time ago to ask about ios/hid stuff)

that said, in theory it could work, yes. if priiloader could load system menu.

vWii has a few issues. the biggest one is that system menu's application is an ancast and encrypted. we'd need the decrypt it first before booting it. but we know the keys, so ok.

second problem is installing it. to install it, it would need to be installed as a separate title that isn't system menu because installing in system menu would brick vWii (since priiloader is not an ancast image).

third problem is that if it is installed and running, it would need a cios or patched ios to even access the system menu ancast to decrypt and run it. 

this is why we want the technical information behind the video, we want to know how it was done. if its done through a different way then how i ran priiloader on vwii ( by creating a separate channel and installing it ) , i'd like to know to see if there is a way to run it flawlessly.

Ploggy, 2021-11-01:
Fakemote is awesome, been testing it on Discord, works really well when it works, but I'd really like the ability to set it up and not worry about it not working after loading another app, lol

I saw earlier in the this thread that the ancast keys were out there now.. that's handy :P

If its possible to add priiloader to a channel, we could load that channel directly from the WiiU menu thanks to fix94's hbc2hbl.rpx (I think that's what it's called) wouldn't that serve the same purpose? or once you exit the Priiloader channel you lose all patches?

That guy seems to be gone, I seems like he did something similar to you, but since his Vid starts in the Priiloader menu I guess he could have done something differently?

DraconicNEO, 2021-11-01:
> vWii has a few issues. the biggest one is that system menu's application is an ancast and encrypted. we'd need the decrypt it first before booting it. but we know the keys, so ok.
>
> second problem is [...]

Since we have the keys could we possibly build Priiloader as an ancast image so it would be accepted, or are there other problems with building Priiloader as an ancast image?

Ploggy, 2021-11-01:
> hbc2hbl? had to look up what that is. is there any source of that? in theory you could make it boot a different title (it being priiloader or w/e) and work from there i suppose. would require some work, since the title would need to have AHB enabled, patch the running IOS to have nand access and ES_DiVerify accessible, and change IOS (or force authentication) when booting system menu.
> 
> as soon as priiloader closes (aka boot something else) it looses all power over the system, the patches would still be applied, but are lost once system menu is restarted.
> 
> i also suspect he either hijacked a title or made a new one, but i wanted to be sure.

Sure here's the Github https://github.com/FIX94/hbl2hbc 

I can live with the losing of patches if the System restarts since we would be booting the channel from WiiU, the only time you would need to restart vWii would be to boot back to WiiU (well no the only time but mostly)

DraconicNEO, 2021-11-01:
> Sure here's the Github https://github.com/FIX94/hbl2hbc
>
> I can live with the losing of patches if the System restarts since we would be booting the channel from WiiU, the only time you would need to [...]

I think he meant every time you return to the Wii Menu i.e. from a Game or HBC.

DacoTaco, 2021-11-01:
> Since we have the keys could we possibly build Priiloader as an ancast image so it would be accepted, or are there other problems with building Priiloader as an ancast image?

im not going to say its impossible, because its technically possible, but it would require to brute force the private key that is used to encrypt the ancast images. thats how asymmetric encryption works , private key can encrypt, public key decrypts.

Ploggy, 2021-11-01:
> I think he meant every time you return to the Wii Menu i.e. from a Game or HBC.

Ahh, I see. I misread. yeah, that's not great :/

DraconicNEO, 2021-11-01:
> 
> 
> > > vWii has a few issues. the biggest one is that system menu's application is an ancast and encrypted. we'd need the decrypt it first before booting it. but we know the keys, so ok.
> > > second problem is installing it. to install it, it would need to be installed as a separate title that isn't system menu because installing in system menu would brick vWii (since priiloader is not an ancast image).
> > 
> > 
> > Since we have the keys could we possibly build Priiloader as an ancast image so it would be accepted, or are there other problems with building Priiloader as an ancast image?
> 
> im not going to say its impossible, because its technically possible, but it would require to brute force the private key that is used to encrypt the ancast images which will (currently) take a lifetime to get. thats how asymmetric encryption works , private key can encrypt, public key decrypts.

Oh, right... Well I guess that's out (unless anyone has a way to brute force keys or change the Wii U's keys).

Ploggy, 2021-11-01:
> im not going to say its impossible, because its technically possible, but it would require to brute force the private key that is used to encrypt the ancast images which will (currently) take a [...]

That's pretty definitive then :P We're not going to be able to brute force it that's for sure.. lol
Thank You for your answers, I suppose to have Fakemote work in other apps, its possible to just modify the app (snes9x for example) to support Fakemote IOS? It's a long way around the problem, but it is a way :P

DacoTaco, 2021-12-22:
@DraconicNEO : are those 'keys' confirmed? according to the failoverflow writeup, those seem to be hashes. and taking a SHA1 hash from my wiiu's otp, which contains the common key, gets me the hash of the common key

Ploggy, 2021-12-23:
@DacoTaco What key are you looking for?
I know the Espresso Wii U Ancast Key, Starbuck Wii U Common Key and the Espresso vWii Ancast Key are out there?

QuickBellHacker, 2021-12-23:
I am just going to change my vWii ios and i will give you news once i am done 
: )

DacoTaco, 2021-12-23:
meanwhile i extracted the ancast images from my wiiu myself. 
Can't decrypt just yet as i miss the IV though. 

@QuickBellHacker : vwii ios changes? you altered vwii ios to boot priiloader?

Nolij, 2022-03-16:
> I think he meant every time you return to the Wii Menu i.e. from a Game or HBC.

Believe that could be worked around in a similar way to how USB Loader GX allows return to wii menu to go back to USB Loader GX instead (works on vWii). Not sure if this is at all what the issue is though.

DacoTaco, 2022-03-17:
i have no time to investigate, but i assume this is different.
i can only assume usbloader gx patches the loaded game to boot a different title when returning to wii menu in vWii?
can somebody validate?

Nolij, 2022-03-17:
> i have no time to investigate, but i assume this is different. i can only assume usbloader gx patches the loaded game to boot a different title (like a forwarder or something) when returning to wii menu in vWii? can somebody validate?
> 
> if that is the case, this patch they do does not help priiloader working in vWii by patching SM

I'm not super experienced with Wii homebrew but I can take a look after work and see if I can discern anything from it. I doubt it's of any value for this I just wanted to address a potential "did you try this?" when I saw the comment I replied to. Will still look at it though.

Awesome-neos145, 2022-04-17:
I'd think you guys may wanna check this out: https://www.youtube.com/watch?v=fzTmV99yAU8&t=2s&ab_channel=QuickBellHacker

DacoTaco, 2022-04-17:
ye, sure. but it would be great to know what is in that wad. a wad install sounds like a terrible idea and i'd rather have it he told us what he did so we could implement it ourselfs and have an official release. this is just.... bad

TheLegendOfXela, 2022-04-17:
QuickBellHacker’s lack of communication is incredibly frustrating.

Ploggy, 2022-04-17:
@QuickBellHacker please mate, what changes did you make to get this to work?

QuickBellHacker, 2022-04-18:
I am sorry if it's a bad idea but I am going to do it anyway since many users are asking and I don't want to let them down,
I use every original Wii ios in vWii most of them doesn't work but some of them do work, 
I am going to look at the priiloader source code and see if I can design it for vWii 
and i am sorry for not responding because of my problems

TheLegendOfXela, 2022-04-18:
I think the issue the devs are having is they want to know what you are doing so they can work together with you

DacoTaco, 2022-04-18:
> I am sorry if it's a bad idea but I am going to do it anyway since many users are asking and I don't want to let them down, I use every original Wii ios in vWii most of them doesn't work but some of [...]

it is no problem that you did not respond, i can understand as im also somebody that sometimes does not reply due to problems.
i would just like to know how it all works, and how it can bootstrap the ppc, since it requires an ancast to even boot up.
in the video you seem to be installing a single wad, which i assume is a modded System menu version. 
what was in that wad @QuickBellHacker ?

it is like @TheLegendOfXela said, i would like to work together and see if we can make an official version

AndrewPiroli, 2022-04-18:
> I'd think you guys may wanna check this out: https://www.youtube.com/watch?v=fzTmV99yAU8&t=2s&ab_channel=QuickBellHacker

This WAD is just a channel that loads the priiloader .dol
It does not have any of the interesting features of priiloader such as system menu hacks or autoboot, and it certainly does not load before the vWii system menu. A cute thing to look at, but not useful in the slightest.

On the WiiU, both the PPC and ARM processors have secure bootroms that perform signature checks on the binaries they are running. Without the signing private keys or a PPC bootrom exploit, no one is modifying the vWii SM or the boot sequence.

Nolij, 2022-04-18:
> This WAD is just a channel that loads the priiloader .dol
> It does not have any of the interesting features of priiloader such as system menu hacks or autoboot, and it certainly does not load before [...]

To the best of my knowledge, we have a theoretical workaround for this: Load vWii with a WAD forwarder on Wii U menu to special Priiloader channel instead of launching regularly (we know this can be done, otherwise WAD forwarders wouldn't exist). Or, this could also be skipped and the user could just run the WAD manually on the vWii menu same way shown in that video. That modified Priiloader would then loadsa patched system menu similarly to how USB Loader GX loads games so return to main menu redirects to USB Loader GX (patching the app that is returned to is also known to be working, otherwise USB Loader GX would've bricked my vWii 100x over), except in this case we'd have it return to Priiloader.

The only issues with the above is whether or not the Wii menu can be loaded like a game (not sure about that, and don't have enough experience to say anything with confidence on that), and whether or not Priiloader patches work on the vWii menu.

As stated, I know very little about the internal workings of the Wii, Wii U, or Priiloader. Just going based off what I see and my experience as a developer elsewhere and my experience as a user of Wii & vWii homebrew. I could be completely wrong, but so far the above is the only potential workaround that I haven't seen any observed issues with.

QuickBellHacker, 2022-04-18:
I am going to give up on priiloader for vWii project it's unnecessary, right?

TheLegendOfXela, 2022-04-18:
> I am going to give up on priiloader for vWii project it's unnecessary, right?

No one wants you to give up! Having PriiLoader on the vWii would be amazing. They just want you to communicate with the developers so you can work together.

AndrewPiroli, 2022-04-18:
This sparked some conversation on Discord. It *might* be possible to create a second copy of the System Menu with a different title id, install Priiloader (would need some engineering work to patch some things as IOS treats the SM title differently) to that.
It gets around the un-modifiable boot process problem, but you would have to launch it through a forwarder on the Wii U side, not the normal vWii entry point.

None of that engineering work has been done, so there's no way to know what's needed and how feasible it is.

RE:
> it's unnecessary

There's not much need for the brick protection (see vWii Decaffeinator) or autoboot (forwarders from Wii U mode as mentioned). The only interesting thing would be the System Menu hacks, which may or may not work in their current state.

TheLegendOfXela, 2022-04-18:
I have an hbl2hbc forwarder that looks identical to the vWii forwarder ready to go whenever.

DacoTaco, 2022-04-18:
> The only interesting thing would be the System Menu hacks, which may or may not work in their current state.

in their current state they wouldn't. but if we have the PPC running priiloader we can load, decrypt, patch and run System Menu ourselves, instead of the powerpc doing it

LegendaryGamerLand36, 2022-04-18:
> I am going to give up on priiloader for vWii project it's unnecessary, right?

None of us wanted you to leave or quit Priiloader on vWii is very interesting for those who dont have a Wii only a Wii U but the most interesting part is how hard you worked on it loading Priiloader on your vWii and the system hacks worked on it
so please dont give up we all need you

DraconicNEO, 2022-04-26:
> I am going to give up on priiloader for vWii project it's unnecessary, right?

Please don't give up on Priiloader for vWii, it's a very cool project that many people have been looking forward to.

QuickBellHacker, 2022-04-30:
I won't, I will continue priiloader for vWii project later!

DacoTaco, 2022-06-26:
Gary and I have been talking about this recently. we have 2 ideas on how to do this : 
* Alter System Menu TMD to use an IOS like Mini or StarStruck to gain control of the starbuck and make expresso jump after verifying SM and before booting it (timing attack, known and used attack)
* Patch SM IOS to use a nandloader to boot SM like it would boot any other title on the system.

in both cases priiloader would be running natively, without any redirection from wiiu mode, but both require some patching on the starbuck side, which is possible

Ploggy, 2022-06-26:
Ahh, awesome, it's great knowing Gary has looked into this. He always brings quality stuff to the WiiU! 
Hopefully you guys can figure it out :)

DacoTaco, 2023-06-29:
the branch is merged! vwii support is official! \o/