DacoTaco/priiloader #270: DSI Exception

eku, 2021-02-18:
**Describe the bug**
When I start the menu item System Menu Hacks in PriiLoader, I get a DSI Excpeption.

**To Reproduce**
Steps to reproduce the behavior:
Just enter menu "System Menu Hack"

**Version**
0.9 from release tab

**Expected behavior**
Shows System Menu Hacks

**Screenshots**
[image]

**Additional context**

PAL Console on v4.1 System Menu

hacks_hash.ini from v0.9 release put in apps/priiloader/

DacoTaco, 2021-02-18:
i suspect this is a known issue that has been fixed (but not released yet).
could you open the included dol in homebrew channel and verify the menu works there?

[priiloader.zip](https://github.com/DacoTaco/priiloader/files/6004851/priiloader.zip)

eku, 2021-02-18:
Yes, this version works as far as the system hacks are concerned, but it has other issues.

The start screen says in the footer
```
Error autobooting systemmenu!
ES_DiVerify failed! Is the IOS patched?
IOS v58
systemmenu v450....
```

and saving systemmenu hacks gives the error `saving failed`.

TheTechRobo, 2021-02-18:
Well, it says "Is the IOS patched?". Check if you have HBC 1.0.7 or later, or if you have the correctly patched IOS 36.

eku, 2021-02-18:
Well, I'm sure I have. But please have a look at my syscheck: [sysCheck.zip](https://github.com/DacoTaco/priiloader/files/6005214/sysCheck.zip)

DacoTaco, 2021-02-18:
ye, thats because its trying to do stuff that require elevated nand rights (something it doesn't get from HBC).

i don't recommend patching any IOS HBC uses though, this was just a test to confirm that this is indeed the bug i already patched.

please install the latest Priiloader 0.9 with all included fixes from the zip below :  
[boot.zip](https://github.com/DacoTaco/priiloader/files/6005225/boot.zip)

TheTechRobo, 2021-02-18:
@DacoTaco I said that since that's what the README.md says, so when it asked if the ios was patched I assumed that was the issue.

DacoTaco, 2021-02-18:
> 
> 
> @DacoTaco I said that since that's what the README.md says, so when it asked if the ios was patched I assumed that was the issue.

under normal circumstances it would be right.
but right now he was running priiloader under HBC's IOS instead of under system menu. hence the lack of privileges hehe.

closed 2021-02-18 by DacoTaco

eku, 2021-02-18:
@DacoTaco Great, the version works as expected. Many thanks for the quick support.

DacoTaco, 2021-02-18:
np