DacoTaco/priiloader #271: [FR] File browser to install file

eku, 2021-02-20:
Hi @DacoTaco

There is a [German mod](https://forum.wii-homebrew.com/index.php/Thread/8918-Priiloader-German-Mod-v0-8-Beta1/) of the PriiLoader, which brings a file browser to select the installtion of the startup file and does not limit the file to `bootl.dol` but offers all DOLs.

Since I recently switched from this mod to your original, I miss the function.

As a workaround, you have to create a directory under `apps` each time and put the DOL in it as `boot.dol` before it is offered for selection in PriiLoader.

Are you enthusiastic about the function and would you perhaps implement it in PriiLoader? Unfortunately, I have not been able to find the source code of the mod, which would have simplified the adoption of the function start.

DacoTaco, 2021-02-20:
i was always under the impression the german mod kinda died long ago.
BadUncle also had commit rights to priiloader back when it was on googlecode. i wonder why he never ported that feature to the main codebase ...

can you provide screenshots of some kinda of this file browser?
currently priiloader can install any homebrew in the apps folder, but im curious how it looked in the mod.

eku, 2021-02-21:
@DacoTaco please follow the link I gave in my first post. From there you can download the package and try it out in Dolphin or on a real Wii to get an idea of how it works.

If I remember correctly, the mod also includes a wad installer.

eku, 2021-03-16:
Hello @DacoTaco 
I would like to ask if you still expect information or whatever from me? Is there anything else I can do to speed up the implementation?

DacoTaco, 2021-12-19:
[image]

finally got the menu to show up. i understand how it was implemented. ill have to look to reproduce it with the current code

DacoTaco, 2021-12-21:
@eku : i was wondering what the use case of this change?
priiloader currently lists everything in the apps folder (just like HBC) + all files in the root of the fat device (sd or usb). 

there are 2 options currently in priiloader to list homebrew applications : 
* place the dol/elf on the root of the SD/USB , any name is possible and priiloader will pick it up. HBC however will not show it
* place the dol/elf as boot.dol/boot.elf in a folder in apps and priiloader will pick it up, just like HBC would.

is there a reason why this menu is preferred over the current system, in which we are more in line with HBC?

eku, 2022-04-05:
@DacoTaco There are programmes that are especially adapted to the start function of the Prilloader, such as `uneoboot.dol` for ULGX. You do not want to put these in the apps directory or in the root directory of the SD card.

Currently, you are forced to put them in one of the two places you mentioned. I'm not sure what pattern the name has to follow.

DacoTaco, 2022-04-05:
why would having the forwarders in the apps directory be bad? they do work there, if named correctly. 
that way you could even use them in HBC too.
see guides like <https://gh.andrewtech.net/wiiflow_forwarder.html> for more info. the idea in this guide is the same as for UsbLoaderGX. instead of a folder names wiiflow_forwarder you could name it something like UsbLoaderGX_Forwarder