DacoTaco/priiloader #277: Pair Wii Remote in Priiloader Menu

DraconicNEO, 2021-03-05:
I think the ability to pair Wii remotes should be added to Priiloader so people can Pair a Wii remote without having to boot to the system menu. This would be useful in sticky situations where you are only able to boot to Priiloader but don't have a way to pair the remotes or any other way of controlling Priiloader without a Wii remote.

TheTechRobo, 2021-03-05:
Yes, good idea. I think was suggested a while ago.

DraconicNEO, 2021-03-06:
> 
> 
> Yes, good idea. I think was suggested a while ago.

Really? I looked around but couldn't find an Issue for it.

TheTechRobo, 2021-03-06:
Yeah, maybe no Issue. But probably on gbatemp. 

Was just saying, very good idea, lots of poep,e would benefit. :)

DacoTaco, 2021-03-06:
i'd love to add that possibility, but this is sadly a libogc/wiiuse issue.
afaik, no homebrew can sync because of this

TheTechRobo, 2021-03-06:
Out of curiosity, what's the issue?

DacoTaco, 2021-03-06:
i have no idea to be honest,i never looked into it very deeply as wiiuse code a thing on its own xD
i think its not responding or doing something right with the sync messages/events.

i got a very old homebrew proof of concept on my pc that can sync wiimotes, but i never did find out where the issue lies.
needless to say, if homebrew could sync wii's it would be huge hehe

TheTechRobo, 2021-03-06:
@DacoTaco Yeah, would definitely be awesome if it could sync wiimotes. Hopefully the issue gets fixed sometime

Leseratte10, 2021-05-12:
> i got a very old homebrew proof of concept on my pc that can sync wiimotes, but i never did find out where the issue lies.

@DacoTaco Do you still have that homebrew somewhere, maybe even with source code? Can you post that somewhere so others can take a look at it? I'd be interested in adding wiimote sync support to homebrew, too.

TheTechRobo, 2021-05-25:
I second Leseratte's request. Having that homebrew might really help

DacoTaco, 2021-09-11:
did some digging. the app has everything build in except for the bluetooth stack, which was build back in 2008ish against libogc.
however, by the time libogc go wiiuse , [it reworked its bluetooth stack to support wiiuse better](https://github.com/devkitPro/libogc/commit/0fa1eb17c34676161ae2c6327e5a2e08f8eb668f), but it seems the integration wasn't completed. there are missing functions & wiiuse integrations that still needed to be done. i got the app to build, but the bluetooth/wiiuse integration of said app is still completely broken.

MasterLuma, 2022-10-03:
I agree that the ability to Sync Wiimotes in Priiloader would be a nice and possibly life-saving feature for a lot of people. However, GameCube controllers work perfectly in Priiloader, and if someone is in a dire situation where they need to control Priiloader's options, they could always just use that instead. But I understand that not everyone has or wants to bother with finding a GC controller.

TheTechRobo, 2022-10-03:
Also the latest Priiloader releases support navigating with the front panel buttons like BootMii does

DacoTaco, 2022-10-03:
and keyboards also work :p 

i still need to look into this though.. but its all very new to me and i lack the time to do it

DacoTaco, 2025-09-06:
i can close this, as any homebrew to be build with latest libogc can now sync wiimotes!

closed 2025-09-06 by DacoTaco