DacoTaco/priiloader #278: Region Free Patch didn't work on MattKC's $1 wii?

TheTechRobo, 2021-03-06:
Maybe something to look into. Haven't finished the video so it may be fixed at the end. Will update soon.

https://www.youtube.com/watch?v=sdWmm33VzBM

DacoTaco, 2021-03-06:
ye, the japanese patch has issues. its a known thing.
i think ive investigated this before, but i can't remember what conclusion i came to then haha.
i -think- the issue has been there even before i switched hacks format many years ago.

i also sadly don't have a Japanese wii nor a different region game to try it with..

TheTechRobo, 2021-03-07:
Couldn't you try dolphin?

DraconicNEO, 2021-07-03:
> 
> 
> Couldn't you try dolphin?

Yes he could, but he probably won't since the Japanese Region is the most heavily neglected region in the scene. These patches apparently even work on Korean Systems (somehow). yet they don't work on Japanese systems or Japanese games in a different region.

DraconicNEO, 2021-07-03:
> i also sadly don't have a Japanese wii nor a different region game to try it with..

You Could Always try to find a cheap imported Wii game and Try out on that, it doesn't actually have to be a good game, just NTSC-J region.

As for the system you could always region change your system or find a cheap Wii, buy it and Region change it (assuming it isn't already Japan region).

DacoTaco, 2021-07-06:
I did a small amount of research , though im only mentioning it for future records. 
#148 seems to end with this issue, so im closing that one.
i also took a look at starpatch, but apparently there is no 4.3J version for it. i assume this is why the original hack didn't exist and why my hash extension also fails.

i think the only way to fix this is to disassemble system menu and see what the hacks actually change (doh :( )

scor57, 2021-09-08:
iirc currently selecting all of the region free options seperately does fix it. its what ive heard from a friend so might not be accurate, but no harm in trying right? its annoying to do but it at least sorts it

DraconicNEO, 2021-09-09:
> 
> 
> iirc currently selecting all of the region free options seperately does fix it. its what ive heard from a friend so might not be accurate, but no harm in trying right? its annoying to do but it at least sorts it

It's not accurate. The region Free options just straight up don't work on Japanese systems. 
people with any doubt should just region change to Japan to see for themselves.

scor57, 2021-09-11:
oh alright

DacoTaco, 2021-09-11:
could somebody try the following on a japanese system menu?

i threw the 4.3E & 4.3J binaries against ghidra and compared the instructions/patch. i think an address offset was off.

(information for me incase im wrong : 
4.3E -> cmpne patched to stw r0, -0x5a28(r13), 0x3F0 from where the rest writes (0x5638).
4.3J -> now patched to stw r0, -0x59e8(r13), 0x3F0 from where the rest writes (0x59E8). )

> [Region Free EVERYTHING JP]
maxversion=512
minversion=512
amount=2
require=RegionFreeAllv1
hash=0x38000000,0x2c000000,0x40820010,0x38000036,0x900daa08,0x480017
patch=0x38000001,0x2c000000,0x900DA618,0x38000032
hash=0x38000000,0x2c000000,0x40820010,0x38000036,0x900daa08,0x480014
patch=0x38000001

itsmattkc, 2021-09-18:
Sorry for the late reply, I was planning to test this a few days ago but have been fighting off a cold.

I can confirm, with the new changes, enabling `Region Free EVERYTHING JP` works perfectly. Tested on a Japanese 4.3J Wii and a PAL copy of Mario Kart Wii.

DacoTaco, 2021-09-18:
thanks for testing mattkc ! 
this verifies my own testing , so i can finally close this!

closed 2021-09-18 by DacoTaco