DacoTaco/priiloader #279: Pick which button to use while booting

TheTechRobo, 2021-03-07:
So, like, we could choose to have SPACE BAR on the keyboard for booting priiloader instead of reset...Just an idea. i wouldn't use it but some epople might.

DacoTaco, 2021-03-07:
and would the button work as a B button throughout priiloader like the esc does now?
what improvement would this bring? is escape incorrectly mapped to B ?

TheTechRobo, 2021-03-07:
No issue that I know of but some may want fine control over it. :thinking:

Especially since maybe parts of their keyboard are broken.

urmum-69, 2021-03-07:
if this is something you seriously need, you should simply compile priiloader with the necessary change
I guess the only argument for it is that the spacebar is an essential key, and maybe someone would still use a keyboard with a broken esc key, but wouldn't with a broken spacebar
Still does not seem remotely worth the effort for the close to zero benefits it would have

DacoTaco, 2021-03-07:
i share that opinion tbh. i felt like the escape key is something that has the same meaning then a B button on a controller when reviewing the pull request. 
i also feel like the user should know what he is doing. if he ends up using a keyboard with a broken escape, its on her/him.

on top of that, [a cheap USB keyboard replacement can be bought dirt cheap on ebay](https://www.ebay.com/itm/MINI-TASTIERA-KEYBOARD-LINQ-K2-USB-MULTIMEDIALE-CON-FILO-USB-NERA/162824265668?hash=item25e9151fc4:g:DS8AAOSwP85cMzRz)
i still have a headache from making these work haha

TheTechRobo, 2021-03-07:
@urmum-69 good point. Wasn't a good idea now that I look at it again. I'll close the PR.

closed 2021-03-07 by TheTechRobo