DacoTaco/priiloader #282: System Menu crashes when the Region Free Wii patch is enabled, the Region Free GameCube patch is disabled, and a GameCube disc is inserted.

Pokechu22, 2021-05-01:
**Describe the bug**
The System Menu crashes when the Region Free Wii patch is enabled, the Region Free GameCube patch is disabled, and a GameCube disc is inserted.  This is a crash where it freezes, and the audio gets stuck emitting the same pitch (whatever was already playing).

**To Reproduce**
Steps to reproduce the behavior:
1. Install priiloader.
2. Enable the Region Free Wii patch, but leave the Region Free GameCube  (No VM Patch) patch (and everything region free patch) disabled.
3. Insert a GameCube disc (I used Paper Mario: The Thousand Year Door, US release).
4. The System Menu will crash when it finishes reading the disc.

**Version**
Priiloader 0.9.0 or 0.9.1; System Menu 4.3U.  I think it is the March 1 2021 hacks_hash.ini.

**Expected behavior**
The system menu not to crash, and allow me to play US-region GameCube games.

**Additional context**
I only own US gamecube games [and datel ones, but those are blocked in MIOS], so I didn't enable the patch initially.  When that patch is enabled (or Region Free Wii is disabled), everything works fine.

DacoTaco, 2021-05-02:
eug, i really hate the hacks issues since i don't know what they are actually doing hehe (i never disassembled SM , i only ported the hacks to the new format).

i verified the issue on my pal wii, it indeed crashes.

i verified the priiloader hack against the preloader hack and noticed that i had basically enabled the hack for SM 4.3 when porting the old offset hack to the new value-based hacks. the hack was never officially ported to 4.3
i guess thats why 4.3 had the 'region free everything' hack, because of the issue.

can you live with the fact that SM 4.3 gets these hacks removed & replaced with the region free everything?

Pokechu22, 2021-05-02:
Yeah, I think just removing those hacks would be fine; as long as it's not easy to accidentally get it into a state where it crashes it should be good.

DacoTaco, 2021-05-02:
the hack has been disabled in the release/fixes branch's hacks file (and merged into master). 
thanks for reporting it :)

closed 2021-05-02 by DacoTaco