DacoTaco/priiloader #283: Wiimmfi Recieved Updates Earlier This Year, Priiloader needs them

Riekoche, 2021-05-26:
**Priiloader Doesn't Take Discs Online Anymore**
Wiimmfi team did something, I couldn't tell you what but they released a bunch of updates earlier this year from January to February. Breaking the Priiloader Wiimmfi Auto Disc Patcher functionality(I use this very often). 

**To Reproduce**
Steps to reproduce the behavior:

1. Insert Disc That Is Not Mario Kart Wii
2. Load Disc From Disc Channel
3. Attempt To Go Online
4. Failure

**Version**
v0.9

**Expected behavior**
To be able to connect to Wiimmfii servers through the Disc Channel.

Riekoche, 2021-05-26:
Discord message link to the latest announcement regarding online functionality: https://canary.discord.com/channels/658414568246214677/698264696976834660/815257370225672192

DacoTaco, 2021-05-27:
isn't this the patch that was created earlier this year?
see https://github.com/DacoTaco/priiloader/pull/276

i'd suggest you update hacks_hash.ini with this one : https://github.com/DacoTaco/priiloader/blob/release/priiloader/hacks_hash.ini
then go back into the priiloader hacks menu, & save the hacks.
it should save the new updated hacks

Riekoche, 2021-05-27:
> isn't this the patch that was created earlier this year?
> see #276
>
> i'd suggest you update hacks_hash.ini with this one : https://github.com/DacoTaco/priiloader/blob/release/priiloader/hacks_hash.ini [...]

 — 02/27/2021 was the last known announcement about Wiimmfii updates on their Discord server. I will grab the hacks_hash.ini that you provided in this thread an see if that fixes anything to see if we can close this thread.

Riekoche, 2021-05-27:
> isn't this the patch that was created earlier this year?
> see #276
>
> i'd suggest you update hacks_hash.ini with this one : https://github.com/DacoTaco/priiloader/blob/release/priiloader/hacks_hash.ini [...]

I updated the priiloader using the self updater and replaced the hash_hacks file. 

Now I'm getting Couldn't find any hacks for System Menu version: v513.

Atleast I'm headed in the right direction, this seems to be operator error on my end. I'm sorry to open this thread.

DacoTaco, 2021-05-27:
no hacks found? where did you place the file on your sd card? :/ 
the hacks file most certainly contains hacks for 4.3

DacoTaco, 2021-07-04:
@Riekoche any update on this?

DacoTaco, 2021-07-06:
closing due to inactivity

closed 2021-07-06 by DacoTaco