DacoTaco/priiloader #284: [Enhancement] Priiloader setting to Change Video Mode

DraconicNEO, 2021-07-04:
hello I think there should be a feature in Priiloader that allows you to change the video mode of the system. Not Just between the video modes of your region but also between NTSC and PAL modes. I feel like this is very much needed because most of the time I see people recommending Priiloader to someone asking about Region changing, the person who wants to region change wishes to do so because of something related to Video modes. 
I don't really think this should be released as a Priiloader hack because then it might not work on all system menu versions (Just like Region Patches in #278 ).

DacoTaco, 2021-07-05:
why would the user need to change video mode from priiloader, let alone into a video mode from a different region?
isn't it the job of the loader to do so? (like in priiloader when booting a game).
though i have no idea how system menu responds when booting a game from a different region?

to me the video mode switch/force sounds like a way to make a wii look bricked if the tv does not support the selected video mode...

now that i think about it, with priiloader being able to boot physical games now, i don't think this feature is necessary ?

DraconicNEO, 2021-07-06:
> why would the user need to change video mode from priiloader, let alone into a video mode from a different region?
> isn't it the job of the loader to do so? (like in priiloader when booting a game). [...]

Hmm I guess, I mean I thought it was a good idea, mainly because someone makes a post "Display European Wii in NTSC" and everyone responds redirecting them to Priiloader. I have had mixed results when Booting PAL Games on an NTSC-U system menu and Vice versa. I felt like the ability to change Video mode could help with that, although I'm not entirely sure. 

It probably would cause a lot of problems from people setting incorrect Video modes though, and it's probably very much redundant because of the Disc booting feature.
(Closing this thread now)

I do feel like there should be something stated in the readme that Region Free patches don't help with video Mode related issues.

closed 2021-07-06 by DraconicNEO

DacoTaco, 2021-07-06:
i didn't mean to demotivate you on the rfc btw, i was just wondering about the use case and its uses.
i understand that the user expects that region free also means no video issues, while from a technical standpoint it is a whole different case since system menu does not set the video depending on the game's region. this indeed means something else (priiloader or a different loader) would need to do so.
personally, i just find it dangerous to make a permanent change like that, knowing there are more non-vulnerable boot2 wii's than fully hackable ones

TheTechRobo, 2021-09-13:
if this was added, there would probably need to be a popup saying "If you can see this screen, press X. Settings will be reverted in 10 seconds." and also a way to revert it after, such as TAB on a USB keyboard...

DraconicNEO, 2021-09-14:
> if this was added, there would probably need to be a popup saying "If you can see this screen, press X. Settings will be reverted in 10 seconds." and also a way to revert it after, such as TAB on a USB keyboard...

Probably, although it's probably still very risky.

FishamanP, 2023-12-12:
This is actually a very useful feature when mixing Wiis with cables from other regions. If you attempt to use an RGB SCART cable on an NTSC region Wii, the picture is entirely red unless the video mode is patched. A similar thing would happen trying to use an S-Video cable on a PAL region Wii. USB loaders can force a video mode for games, but Priiloader should ideally have a solution for the System Menu.