DacoTaco/priiloader #286: Can't update old version of priiloader

wojtekpolska, 2021-07-07:
**Describe the bug**
Can't update Priiloader to modern version

**To Reproduce**
Steps to reproduce the behavior:
Im not sure

**Version**
Priiloader v0.7 (r138)

**Expected behavior**
Priiloader updates to most recent version

**Screenshots**
[If applicable, add screenshots to help explain your problem.]

**Additional context**
I tried both trying to install trough homebrew channel from SD, and using built-in Update function trough wifi

in both cases i see the 1st black screen, but after half a second or so it immiedetely goes to "Press A to exit to loader..."
when trying trough HBC, it goes back to HBC
when trying trough Priiloader Wi-Fi update, it says "Exiting..." but nothing happens

DacoTaco, 2021-07-07:
can you upload a system check excel to verify something?
i suspect IOS80 and/or IOS58 are still in a state that priiloader does not expect.

wojtekpolska, 2021-07-07:
> can you upload a system check excel to verify something?
> i suspect IOS80 and/or IOS58 are still in a state that priiloader does not expect.

how can i do that @DacoTaco

wojtekpolska, 2021-07-07:
> yup, thats the one. i would need the csv output to verify the system menu version and all ios ver

i dont know whats sensitive, so i will just put the files here, and trust you. 
I guess i can delete them from here after u are done

wojtekpolska, 2021-07-07:
[IOS80.zip](https://github.com/DacoTaco/priiloader/files/6778406/IOS80.zip)

[IOS58.zip](https://github.com/DacoTaco/priiloader/files/6778407/IOS58.zip)

wojtekpolska, 2021-07-07:
also interesting thing - Homebrew Browser thinks Priiloader is at its most recent version

TheTechRobo, 2021-07-07:
I presume that's the Open Shop Channel? I don't think the HBB has that newest version.

Worst case scenario, you could try uninstalling, and reinstalling? But I think you should see if Daco can fix the bug first.

wojtekpolska, 2021-07-07:
> I presume that's the Open Shop Channel? I don't think the HBB has that newest version.
> 
> Worst case scenario, you could try uninstalling, and reinstalling? But I think you should see if Daco can fix the bug first.

its open shop channel but its still called homebrew browser in homebrew channel

DacoTaco, 2021-07-07:
in the IOS 58 csv it says what system menu version you have installed and it is an (unofficial) version priiloader does not like, nor supports. ( System Menu 4.1E (v54450) )
4.1E should not be v54450. also IOS80 has an unofficial modded version installed too.
i suspect a DarkCorp installation or at least the fucked up System menu 4.1 version 'to block updates' which was a terrible idea, i have no idea why anyone made that shit.

In any case, this is why priiloader refuses to install. i suggest you make sure your homebrew channel is up to date and find a way to force an installation of IOS80 & System menu 4.3 ( **! in that order !** )

and please take a bootmii backup before doing anything

wojtekpolska, 2021-07-07:
@DacoTaco 

i am not on 4.1 
i was on 4.1 and the console had some ancient homebrew  so i formated it and updated trough sertings to 4.3E

wojtekpolska, 2021-07-07:
what should i do?

wojtekpolska, 2021-07-07:
how can i fix these 2 files to be normal again?

TheTechRobo, 2021-07-08:
Formatting does NOT remove/reinstall any IOSes or homebrew as far as I know.

IIRC, it only touches save files and channels.

But I could be wrong so don't trust me on that. I'm not a Wii expert.

DacoTaco, 2021-07-08:
formatting is indeed not the way.
i contacted somebody on reddiot's /r/wiihacks since they are more capable in telling you what to do then i am (i would give correct, but maybe dangerous advise haha).

he told me to follow this guide : https://www.rwiihacks.com/tutorials/32004fix/
this should fix your issues. keep in mind that the update removes priiloader , so keep the installer on the SD.
just like it says in the guide, you'll need to reinstall any programs it might have deleted afterwards, but just follow the wiiguide untill the system is back like how you had it (hbc, bootmii, priiloader etc etc)

DacoTaco, 2021-07-10:
closing due to fix mentioned and no activity. if problems would come up, don't hesitate to comment in this issue or make a new one ! :)

closed 2021-07-10 by DacoTaco

wojtekpolska, 2021-07-12:
sorry i kinda forgot about this, but im nott sure what to do exactly. 

You said to get IOS80 but idk which version
[image]

the guide says to get system meny v 4.1E (in my case)
but it doesnt output a wad like the guide says it should 
[image]

wojtekpolska, 2021-07-12:
I think i will try with the system menu 4.1E option. Thanks for linking the guide and all :D

wojtekpolska, 2021-07-12:
OK it worked! im on Ver. 4.3E and Priiloader installed!
Thanks for helping me :D

DacoTaco, 2021-07-12:
no problem. happy to see its all resolved now ! 

(the latest IOS80 was a good option, but there is no need for it anymore atm)