DacoTaco/priiloader #287: `No-Delete ...,RZDx` not working

kipcode66, 2021-07-12:
**Describe the bug**
The `No-Delete HAXX,JODI,DVDX,DISC,DISK,RZDx` option in the Settings menu does not allow for a modified Twilight Princess save file to be copied over to the console.

**To Reproduce**
Steps to reproduce the behavior:
- Install Priiloader 0.9.1
- Enable the `No-Delete HAXX,JODI,DVDX,DISC,DISK,RZDx` setting
- Decode the save from [data.txt](https://github.com/DacoTaco/priiloader/files/6804311/data.txt) using (on linux): `base64 --decode data.txt > data.bin`
- Copy the `data.bin` save file under `/private/wii/title/RZDE/` folder of your SD Card
- Put your SD Card into your wii and copy over the save from your SD Card to your console
- See an error message after the copy finishes. (My console is in french, so the message is `La copie des données a peut-être échouée`)

**Version**
0.9.1

**Expected behavior**
The save file should copy over to the console without errors.

**Additional context**
I have been working with 2 other people on a program that would allow to load custom code into Twilight Princess (code RZDE). I have been told that Priiloader could (un)patch the fix from Nintendo that would prevent modified saves from getting copied over, but Using the latest release (0.9.1), I have not been able to load a save modified using the program, but I have been able to load a save from an other game, also modified in a similar way using the same script.

TheTechRobo, 2021-07-13:
I speak French, and (for those that don't) that error message means "The copying of the data might have failed" (or something similar).

The file is probably corrupt. You might not have saved it correctly. The only deletion that Priiloader is fixing is the deletion on boot; if normal Twilight Hack copies correctly , I don't think this is a Priiloader issue.

DacoTaco, 2021-07-13:
what happens if you put the save on an SD card , boot system menu and then insert the sd into the wii? 
the hack indeed prevents the whole deletion thing, but im not sure what system menu does besides on startup.
would it delete it, refuse to copy it, or throws the data corrupt error message?

its been too long since the creation of the twilight hack and i have no idea how system menu interacts with it.
thanks for that translation btw. despite that i should know french, i don't haha.

Masamune3210, 2021-07-13:
There is a check in later System Menu's to disallow the copy if the TP hack is detected, but it only applies to the first save in the pack, and was worked around by adding the hacked save into the third slot iirc, maybe make sure you are using the latest version of the hack?

Masamune3210, 2021-07-14:
also, make sure you have a TP save already on your consoles' nand before trying to copy the exploit

kipcode66, 2021-07-14:
> also, make sure you have a TP save already on your consoles' nand before trying to copy the exploit

When I try to copy any file over an existing one, the menu won't allow me to copy the file over.

> The file is probably corrupt. You might not have saved it correctly. The only deletion that Priiloader is fixing is the deletion on boot; if normal Twilight Hack copies correctly , I don't think this is a Priiloader issue.

It shouldn't be corrupt, since I'm able to load the file through Dolphin and SaveGame Manager GX. I also verified the signature and the certificate and it's valid according to Dolphin's algorithm. Also, using the same algorithm, and using an other game's save, the modified save copies over to the console without a problem.

> There is a check in later System Menu's to disallow the copy if the TP hack is detected, but it only applies to the first save in the pack, and was worked around by adding the hacked save into the third slot iirc, maybe make sure you are using the latest version of the hack?

I made the script for the hack myself, and the current save I tried to use has the hack on the 3rd file already.

> There is a check in later System Menu's to disallow the copy if the TP hack is detected, but it only applies to the first save in the pack, and was worked around by adding the hacked save into the third slot iirc, maybe make sure you are using the latest version of the hack?

Even when I put the SD Card in after the boot menu, the problem lies when I try to copy the file over from the SD Card to the console. It basically throws the message above that the copy "might have failed".

DacoTaco, 2021-09-11:
i don't know what to do with this one to be honest.

the patch patches out a function call ( FUN @ 000379a8 ) which probably does the check & deletion. 
i checked in ghidra, and the one call that patched out is the only one.
so it must have some checks somewhere else too, but i have no idea where and without somebody who knows where to look, fixing this would require a system menu disassembly to go through the save menu :/