DacoTaco/priiloader #288: Updating from Priiloader 0.9 seems to fail

InternalLoss, 2021-09-09:
**Describe the bug**
When trying to update Priiloader 0.9, `error getting versions from server` appears.

**To Reproduce**
Steps to reproduce the behavior:
1. Select Check for Updates

**Version**
Priiloader 0.9 (r0x03225439 - this might be a Wiimmfi test build)

**Expected behavior**
Priiloader 0.9.1 update appears.

**Additional context**
I can see the request succeeds via packet capture: the data is
`0000005bc1995e11aae6b57df571917396966a7b4d8fc2a10000005a000000040c4365bbb377d9a468f7cd58d285299ad08c75b1`

[image]

DacoTaco, 2021-09-09:
hm. thats odd. the content does seem to be correct (and its downloading the bin instead of the dat if wireshark is correct).
you are sure this is what the wii received?
did it also try to get a version.dat ?

the error is shown if the downloaded data size is 0, there is no data or the size of the received data is not correct (it expects 52 in the current 0.9 builds )

also, commit 0x3225439 is the 0.9 tag, so this either the release build or a custom build.

DacoTaco, 2021-09-09:
eug, problem found. its a combination of my webhost suddenly sending http headers in lower case (and not camel case) and my shitty http parser (of when i didn't know fully what i was doing).

this sadly means that all online updating is down again and only a bugfix can fix it...

guess ill beter start planning to rewrite the http parser... -_-

(and i don't even find time to do all the other stuff needed in priiloader *looks at issue tracker* )

InternalLoss, 2021-09-09:
oops! sorry for introducing more bugs >_<

DacoTaco, 2021-09-09:
no its fine. i'd rather have it known and fixed. i just don't know when ill get to ever release 0.10 xD
( which has the dvd stuff and usb stuff if i ever find time to look at why that is an issue to begin with)

DacoTaco, 2021-09-09:
i shouldn't have done this (due to my cold/health) but i kinda fixed the issue.
you can test it with the attached installer. 
there won't be any updates (obviously) , but it will download the version file correctly.
thanks for the investigation. it helped to find the issues :)

[boot.zip](https://github.com/DacoTaco/priiloader/files/7139946/boot.zip)

closed 2021-09-09 by DacoTaco

DacoTaco, 2021-09-10:
seems i was a bit to quick to commit & close this issue. had a few bugs in the new parser ( it was 1AM after all... )

this should have it all fixed @InternalLoss 
[boot.zip](https://github.com/DacoTaco/priiloader/files/7145623/boot.zip)

InternalLoss, 2021-09-11:
Can confirm, I now get `no major updates` :)