DacoTaco/priiloader #290: "Can't find hacks_hash.ini on NAND" error on Wii Mini

tanchek, 2021-10-16:
**Describe the bug**
Hi. I used this tutorial to install Priiloader on my Wii Mini: https://wii.guide/priiloader. Before doing that, I successfully hacked the console through the BlueBomb exploit and installed Homebrew Chanell. After that, I have successfully installed Priiloader v0.9.1. After that, I turned off the console, plugged a USB keyboard into the USB port of my Wii Mini, turned on the console, and pressed Esc on the keyboard. The Priiloader menu is launched. I clicked on "System Menu Hacks" but got the error "Failed to mount FAT device. Can't find hacks_hash.ini on NAND". I've tried reinstalling Priiloader; tried to uninstall Priiloader and reinstall it; tried to uninstall Priiloader; Reset Wii to factory settings and reinstall Homebrew Chanell + Priiloader; tried to format the USB-stick and re-download the installation file to it; even tried another flash drive - it didn't help. I have not tried installing cIOS, as I understand that Priiloader is recommended to be installed on the console with the original iOS. I don't know what else I can do to get Priiloader to function on my console. Please help me.

P.S. This is not the first Wii Mini I have installed the Priiloader on. On my other Wii Mini, this problem did not appear, everything worked out the first time.

**To Reproduce**
Steps to reproduce the behavior:
1. Hack Wii Mini via BlueBomb exploit and install Homebrew Chanell
2. Launch Priiloader v0.9.1 installer via Homebrew Chanell
3. Install Priiloader on Wii Mini by pressing the A / + button
4. Turn off the console, plug USB keyboard into the USB port of Wii Mini, turn on the console, and press Esc on the keyboard
5. Move to System Menu Hacks
6. Enjoy error "Can't find hacks_hash.ini on NAND".

**Version**
Wii Mini 4.3E + Priiloader v0.9.1 (downloaded here: https://wii.guide/priiloader)

**Expected behavior**
System Menu Hacks page should open (but the above error appears)

**Screenshots**
[image]
[image]

**Additional context**
[image]
[image]

DacoTaco, 2021-10-16:
from what it looks like, i'd say the priiloader installation has no problems. 
however, it has issues accessing your usb stick (thats what the mount error says). can you try another usbstick?

i can't read your language, but does the usb stick have an custom name ?

tanchek, 2021-10-16:
Thanks for the answer.

The USB flash drive does not have a custom name. I have tried another USB stick as well (the one I previously used to install the Priiloader for my second Wii Mini). I can try to get one more another USB stick and try to install from it.

Absolutely right. Wii Mini has one USB port. And the keyboard is plugged in the moment I try to open `System Menu Hacks`. But shouldn't Priiloader open a Menu Hacks page without a flash drive plugged in? Otherwise, how can I open `System Menu Hacks` if the keyboard is not connected?

DacoTaco, 2021-10-16:
no, the first time you open (and save) the hacks, priiloader needs the sd/usb to get the hacks from. once saved, it can use the saved copy on the nand.

DacoTaco, 2021-10-16:
no, loadPriiloader.dol will not have resulted in the hacks being written to nand. the only thing that dol does it force priiloader to show the menu instead of launching the system menu (or whatever the autoboot settings are)

as for the flash drive in the usb port, it would need to have been inserted before entering the hacks menu, and you would need to use a wiimote to enter the hacks menu. once the hacks are listed and you saved the hacks (aka by selecting the save menu), the usb stick is no longer needed unless you want to update the hacks.

DacoTaco, 2021-10-16:
that is indeed to order ye. its a bit annoying with the wii mini only having 1 usb port. but apparently the usb hub of the keyboard also works :)

closed 2021-10-16 by DacoTaco