DacoTaco/priiloader #291: [Enhancement] Ability Force Video mode on games

DraconicNEO, 2021-11-14:
I think it would be a good feature for people importing games to be able to force the Video mode of Disc based games (maybe also possibly channels for Wiiware?). This would be very useful for people since they could just pop in their import discs without having to worry if their TV is Multiscan or that their Scart cables won't work.

DacoTaco, 2021-11-14:
isn't this the same as your other RFC in #284 , or am i missing something?

DraconicNEO, 2021-11-14:
> 
> 
> isn't this the same as your other RFC in #284 , or am i missing something?

It wouldn't change the video mode on the Wii menu and would Probably only Apply to Disc Based Games. On that last one if someone were to change the video mode to something unsupported by their display it would affect their whole Wii, this would probably only do Disc based games without affecting the System menu or Priiloader Menu.

DacoTaco, 2021-11-14:
im still not sure i follow completely.
are we talking about disc based games run by priiloader, or by system menu? 
if its launched by priiloader, i thought it already sets the video mode before launching the game 
if its launched by system menu, that would require a priiloader hack to patch system menu i think.

DraconicNEO, 2021-11-14:
> im still not sure i follow completely. are we talking about disc based games run by priiloader, or by system menu? if its launched by priiloader, i thought it already sets the video mode before [...]

Yeah I was thinking of games launched by the system menu which when I think about it, definitely would require a hack.