DacoTaco/priiloader #292: Region Free Everything on 4.1E PAL

Astrov003, 2021-11-14:
My apologies if this was answered before, but I was searching in many places and could not find a direct information.

Priiloader v0.9.1 is installed on my white Wii PAL console, which runs 4.1E system menu (v450).
Everything works fine (autoboot and the rest), except that I am missing Region Free Everything option in the System Menu Hacks.
I only see three separate options for Wii, GC and Channels.

I have tried with several different hacks_hash.ini files (I can confirm that they successfully load when I boot up Priiloader , as the listing of options change if I use different hacks_hash.ini). I currently have the latest one I found here in your repository.

Hacks do work (for example, remove background music), but I do not see this particular option in question.

Is it even available on my version of console and system menu, or is it another issue I should check?

Thank you.

DacoTaco, 2021-11-14:
ok so, this actually has something to do with a history lesson hehe.

i didn't make the hacks, i only ported them to the (current) hacks format used by priiloader.
they initially were created/updated to each system menu version separately.
before 4.3 the region hacks were always separated into 3 hacks and i can only assume this is because they were made/updated by different people/at different times.

with 4.3 however, they were merged into 1 hack.
i don't know why.

i'd need to verify with the system menu code, but i think the region free everything hack can be ported to the older system menu versions so its not longer separated hacks.
this is what you would prefer, no?

Astrov003, 2021-11-14:
Thanks for the reply DacoTaco.

If I understood correctly, Region Free Everything includes these three options (Wii, GC, Channels) and nothing more?
The issue I am trying to solve, is to be able to play NTSC Virtual Console games on my PAL console. I went with Region Free Channels enabled, but I just get a black screen, WiiMote loses sync, and I have to restart the console (home button does not work, as it is not the issue with video mode).

I was under the impression that Region Free Everything gives me the additional functionality for VC, but if this is not the case (and Region Free Channels should solve this), than combining them into one hack would not solve the issue, if I am correct?

From the UI point of view, I don't mind having either one combined or three separate options. Anyhow, thanks you for your intention to look into it.

What boggles me now, is, should Region Free Channels hack enable NTSC VC on a PAL system, and if it does, what could be the problem on my system?

Thanks again.

DacoTaco, 2021-11-14:
hmm, im not sure if VC falls under the channels hack or not.
it could be, not sure.

ill see if i can port the hack to 4.1E and we'll know for sure.

something that could be wrong, if the hack is suppose to work, is maybe corrupted data or a missing ios.
do you have a syscheck i could look at?

Astrov003, 2021-11-14:
Yup, here it is.
I just did a remodding a few days ago with ModMii.

>  SysCheck HDE v2.4.0 HacksDen Edition by JoostinOnline, Double_A, R2-D2199, and Nano
> ...runs on IOS58 (rev 6176).
> 
> Region: PAL
> System Menu 4.1E (v450)
> Priiloader installed
> Drive date: 12.18.2008
> Homebrew Channel 1.1.2 running on IOS58
> 
> Hollywood v0x21
> Console ID: 150844960
> Console Type: Wii
> Shop Channel Country: Serbia (101)
> Boot2 v4
> Found 113 titles.
> Found 56 IOS on this console. 6 of them are stubs.
> 
> IOS3 (rev 65280): Stub
> IOS4 (rev 65280): Stub
> IOS9 (rev 1034): No Patches
> IOS10 (rev 768): Stub
> IOS11[60] (rev 65535, Info: ModMii-IOS60-v6174): Trucha Bug, NAND Access
> IOS12 (rev 526): No Patches
> IOS13 (rev 1032): No Patches
> IOS14 (rev 1032): No Patches
> IOS15 (rev 1032): No Patches
> IOS16 (rev 54321): Trucha Bug, ES Identify, Flash Access, NAND Access
> IOS17 (rev 1032): No Patches
> IOS20[60] (rev 65535, Info: ModMii-IOS60-v6174): Trucha Bug, NAND Access
> IOS21 (rev 1039): No Patches
> IOS22 (rev 1294): No Patches
> IOS28 (rev 1807): No Patches
> IOS30[60] (rev 65535, Info: ModMii-IOS60-v6174): Trucha Bug, NAND Access
> IOS31 (rev 3608): No Patches
> IOS33 (rev 3608): No Patches
> IOS34 (rev 3608): No Patches
> IOS35 (rev 3608): No Patches
> IOS36 (rev 3608): No Patches
> IOS37 (rev 5663): No Patches
> IOS38 (rev 4124): No Patches
> IOS40[60] (rev 65535, Info: ModMii-IOS60-v6174): Trucha Bug, NAND Access
> IOS41 (rev 3607): No Patches
> IOS43 (rev 3607): No Patches
> IOS45 (rev 3607): No Patches
> IOS46 (rev 3607): No Patches
> IOS48 (rev 4124): No Patches
> IOS50[60] (rev 65535, Info: ModMii-IOS60-v6174): Trucha Bug, NAND Access
> IOS51 (rev 4864): Stub
> IOS52[60] (rev 65535, Info: ModMii-IOS60-v6174): Trucha Bug, NAND Access
> IOS53 (rev 5663): No Patches
> IOS55 (rev 5663): No Patches
> IOS56 (rev 5662): No Patches
> IOS57 (rev 5919): No Patches
> IOS58 (rev 6176): USB 2.0
> IOS60 (rev 6174): Trucha Bug
> IOS61 (rev 5662): No Patches
> IOS62 (rev 6430): No Patches
> IOS70[60] (rev 65535, Info: ModMii-IOS60-v6174): Trucha Bug, NAND Access
> IOS80[60] (rev 65535, Info: ModMii-IOS60-v6174): Trucha Bug, NAND Access
> IOS90 (rev 54321): Trucha Bug
> IOS202[60] (rev 65535, Info: hermesrodries-v6): Trucha Bug, NAND Access, USB 2.0
> IOS222 (rev 65280): Stub
> IOS223 (rev 65280): Stub
> IOS224[57] (rev 65535, Info: hermesrodries-v6): Trucha Bug, NAND Access, USB 2.0
> IOS236[36] (rev 65535, Info: rev 3351): Trucha Bug, ES Identify, NAND Access
> IOS245[37] (rev 21007, Info: d2x-v7final): Trucha Bug, NAND Access, USB 2.0
> IOS246[38] (rev 21007, Info: d2x-v7final): Trucha Bug, ES Identify, NAND Access
> IOS247[53] (rev 21007, Info: d2x-v7final): Trucha Bug, NAND Access, USB 2.0
> IOS248[55] (rev 21007, Info: d2x-v7final): Trucha Bug, NAND Access, USB 2.0
> IOS249[57] (rev 65535, Info: d2x-v10beta52): Trucha Bug, NAND Access, USB 2.0
> IOS250[56] (rev 65535, Info: d2x-v10beta52): Trucha Bug, NAND Access, USB 2.0
> IOS251[38] (rev 65535, Info: d2x-v10beta52): Trucha Bug, ES Identify, NAND Access
> IOS254 (rev 65281): BootMii
> BC v6
> MIOS v10
> Report generated on 11/13/2021.

DacoTaco, 2021-11-14:
hmm, looks ok. a lot of ios' so a missing ios might not be the issue. 
ill keep you updated

Astrov003, 2021-11-14:
Yes, some of the ios' are remnants of the ancient hack done by someone, and also MiiMod ads a lot of them (belts and suspenders I guess).

Thanks, Daco.

DacoTaco, 2021-11-14:
according to a user on discord (thanks AndyRogi) , the hack worked on his system using a fresh system menu/ios install.

so either the channel is broken, or its ios.

can you try and boot the channel from within priiloader? does that work?
if not, i'd suggest to try and reinstall the channel or verify the wad with dolphin to see if it works.

Astrov003, 2021-11-14:
Ok, I have verified the wad (Super Mario Bros. NTSC), and it runs on Dolphin.
Then I reinstalled the channel on Wii with YAWMM (I usually do uninstall first to remove all traces).
I tried booting from Priiloader, but got black screen again (for comparison, I managed to boot PAL Super Mario Bros.).

Since you mentioned channel's IOS, I have checked the wads with ShowMiiWads: SMB NTSC uses IOS9, shile SMB PAL uses IOS17.
In my syscheck I see them both unpatched. Could this be probably related to the issue? I am just guessing now, but perhaps NTSC wads ask for pathed IOS on PAL consoles? Or it shouldn't matter, because of Region Free Channels enabled?

Astrov003, 2021-11-14:
Just to add:
I tried with some SNES and N64 wads, both NTSC.
Super Mario 64 NTSC works, while Super Mario World NTSC gives the same behavior with black screen.

So, it seems that PAL console has troubles with running NES and SNES NTSC titles.

DacoTaco, 2021-11-14:
ah, so some VC channels work in system menu, and some don't? 
its funny you mention SM64, that was a game AndyRogi also tested (his US wii was changed to 4.1E, testing the US version of the channel)

Astrov003, 2021-11-14:
Apparently so, some do work. Different behavior between NES/SNES and N64 games. I did not yet do an extensive testing to confirm if all titles of a particular system behave like this.

DacoTaco, 2021-11-14:
thats interresting, seems some VC channels just dont like being run when started from a different region. i added a different issue to test that stuff and see if i can fix the channels being run from priiloader. system menu will stay broken though, since this might be because of how channels act, not system menu

Astrov003, 2021-11-14:
Probably so, since running the same channels from USB Loader GX gives the same behavior.

One question though: do Priiloader hacks (such as Region Free) translate to USB Loader GX, or they solely affect the System Menu?

DacoTaco, 2021-11-14:
they only affect System menu.
Priiloader loads system menu into memory , then applies the hacks to the loaded system menu.
usb loader gx would need to apply its own hacks.

Astrov003, 2021-11-14:
Ok, got it.
I did some additional testing - I wanted to see if there is a difference between early and later released NES VC channels. The original SMB was released as one of the first in 2006, while Mega Man 5 as one of the last in 2011.

But no, same output.

DacoTaco, 2021-11-14:
AndyRogi suspects N64 VC work fine, while others might cause issues. I suspect its a video mode issue

Astrov003, 2021-11-14:
Most definitely. It seems that the system menu gets confused about video mode when you try to run NTSC game on a PAL system, and freezes. If I run the same channel over USB Loader GX, and select 'force NTSC' option, the game loads.
However, this freeze only applies to NES and SNES games, while N64 games run via the system menu, which is peculiar.

I am also fairly confident to confirm that "Region Free Channels" hack does apply to Virtual Console channels, as, when I disable it, I get the message "This channel cannot be used' by the system menu.

DacoTaco, 2021-12-19:
the hacks have been merged in the main branch, so to use the merged (and fixed hacks) , use the following hacks file : 
https://raw.githubusercontent.com/DacoTaco/priiloader/master/priiloader/hacks_hash.ini

closed 2021-12-19 by DacoTaco