DacoTaco/priiloader #295: Game Disk not Starting

ConWai, 2021-12-23:
**Describe the bug**
With installed Priiloader the genuine game disk of Wii Play PAL starts up with a black screen and won't go any further.
The DVD-drive keeps spinning. The Wii-Remote connection gets cut off.
After uninstalling Priiloader the game works again.
All my other disc games work fine with Priiloader installed including Wii Sports.

I can run a backup of the game from USB-Loader GX.
It also works starting the Game Disk directly from inside  USB-Loader GX using cIOS 249.

If someone wants any additional information I try to do my best.

**To Reproduce**
Steps to reproduce the behavior:
- install Priiloader
- open disk channel with Wii Play inserted
- click "Start"
- the screen switches to black until force rebooting the console

**Version**
Priiloader v0.9.1 or v0.9. (_With Priiloader v8.2 Wii Play is working fine!_)
Softmodded Wii 4.3E (PAL)

DacoTaco, 2021-12-23:
what hacks are enabled in priiloader? any settings i can try to reproduce this with?

ConWai, 2021-12-23:
I testet it with "Region Free Everything" and "Block Disk Updates" enabled and disabled. Other than that all hacks are disabled.

DacoTaco, 2021-12-23:
any priiloader settings by accident like using a different ios for system menu? :/ 

i just tried it on my wii with Mario Strikers Charged Football and it booted fine ( Priiloader 0.10 4.3E , Pal disc, no hacks enabled )

ConWai, 2021-12-23:
Thank you for testing.
I didn't change the IOS setting. It uses IOS80.
Maybe there is something messed up on my Wii. For some (other) reason the region free hack also does not work in any of the versions. The game banners do load in the disk channel though after applying the patch, but it behaves exactly like my wii play.

Could reinstalling the original IOS80 fix anything? I did not try that because I fear to brick the console. (I can only use the IOS based BootMii).

DacoTaco, 2021-12-23:
it might be best to first post a syscheck result to see what is installed on the wii (ios, versions, patches, etc etc )

ConWai, 2021-12-23:
I hope this helps:
SysCheck HDE v2.4.0 HacksDen Edition von JoostinOnline, Double_A, R2-D2199, und Nano
...laeuft auf dem IOS58 (rev 6176).

Region: PAL
Systemmenue 4.3E (v514)

Laufwerksdatum: 13.02.2007
Homebrewkanal 1.1.2 benutzt IOS58

Hollywood v0x11
Konsolen-ID: 10******* (removed for privacy ^^'?)
Konsolentyp: Wii
Shop-Kanal-Land: Germany (78)
Boot2 v4
Es wurden 138 Titel gefunden.
Es wurden 54 IOS gefunden, von denen 9 funktionslos (Stub) sind.

IOS4 (rev 65280): Funktionslos (Stub)
IOS9 (rev 1034): Keine Patches
IOS10 (rev 768): Funktionslos (Stub)
IOS11 (rev 256): Funktionslos (Stub)
IOS12 (rev 526): Keine Patches
IOS13 (rev 1032): Keine Patches
IOS14 (rev 1032): Keine Patches
IOS15 (rev 1032): Keine Patches
IOS16 (rev 512): Funktionslos (Stub)
IOS17 (rev 1032): Keine Patches
IOS20 (rev 256): Funktionslos (Stub)
IOS21 (rev 1039): Keine Patches
IOS22 (rev 1294): Keine Patches
IOS28 (rev 1807): Keine Patches
IOS30 (rev 2816): Funktionslos (Stub)
IOS31 (rev 3608): Keine Patches
IOS33 (rev 3608): Keine Patches
IOS34 (rev 3608): Keine Patches
IOS35 (rev 3608): Keine Patches
IOS36 (rev 3608): Keine Patches
IOS37 (rev 5663): Keine Patches
IOS38 (rev 4124): Keine Patches
IOS40[60] (rev 65535, Info: ModMii-IOS60-v6174): Trucha Bug, NAND Zugriff
IOS41 (rev 3607): Keine Patches
IOS43 (rev 3607): Keine Patches
IOS45 (rev 3607): Keine Patches
IOS46 (rev 3607): Keine Patches
IOS48 (rev 4124): Keine Patches
IOS50 (rev 5120): Funktionslos (Stub)
IOS51 (rev 4864): Funktionslos (Stub)
IOS52[60] (rev 65535, Info: ModMii-IOS60-v6174): Trucha Bug, NAND Zugriff
IOS53 (rev 5663): Keine Patches
IOS55 (rev 5663): Keine Patches
IOS56 (rev 5662): Keine Patches
IOS57 (rev 5919): Keine Patches
IOS58 (rev 6176): USB 2.0
IOS60 (rev 6400): Funktionslos (Stub)
IOS61 (rev 5662): Keine Patches
IOS62 (rev 6430): Keine Patches
IOS70[60] (rev 65535, Info: ModMii-IOS60-v6174): Trucha Bug, NAND Zugriff
IOS80[60] (rev 65535, Info: ModMii-IOS60-v6174): Trucha Bug, NAND Zugriff
IOS202[60] (rev 65535, Info: hermes-v5.1): Trucha Bug, NAND Zugriff, USB 2.0
IOS222[38] (rev 65535, Info: hermes-v5.1): Trucha Bug, ES Identify, NAND Zugriff, USB 2.0
IOS223[37] (rev 65535, Info: hermes-v5.1): Trucha Bug, NAND Zugriff, USB 2.0
IOS224[57] (rev 65535, Info: hermes-v5.1): Trucha Bug, NAND Zugriff, USB 2.0
IOS236[36] (rev 65535, Info: rev 3351): Trucha Bug, ES Identify, NAND Zugriff
IOS245[37] (rev 21008, Info: d2x-v8final): Trucha Bug, NAND Zugriff, USB 2.0
IOS246[38] (rev 21008, Info: d2x-v8final): Trucha Bug, ES Identify, NAND Zugriff
IOS247[53] (rev 21008, Info: d2x-v8final): Trucha Bug, NAND Zugriff, USB 2.0
IOS248[55] (rev 21008, Info: d2x-v8final): Trucha Bug, NAND Zugriff, USB 2.0
IOS249[56] (rev 65535, Info: d2x-v10beta52): Trucha Bug, NAND Zugriff, USB 2.0
IOS250[57] (rev 65535, Info: d2x-v10beta52): Trucha Bug, NAND Zugriff, USB 2.0
IOS251[38] (rev 65535, Info: d2x-v10beta52): Trucha Bug, ES Identify, NAND Zugriff
IOS254 (rev 65281): BootMii
BC v6
MIOS v10
Bericht wurde am 23.12.2021 generiert.

closed 2021-12-23 by ConWai

DacoTaco, 2021-12-23:
you deleted the previous message, but it did look like you had a hacked/altered IOS80 ye.
glad to see that was the issue :)

as for region hack, it should work. maybe try updating hacks ini could help? i don't know

TheTechRobo, 2021-12-24:
BTW: They didn't delete the message, they just edited it. You can just press the "edited" text to see older revisions

DacoTaco, 2021-12-24:
weird, i got 2 emails though. one for the syscheck and one with the message seen above *shrug*

ConWai, 2021-12-24:
Yes I editited it. Sorry for the confusion. I didn't want to bother you with so much information after I fixed it, hoping you didn't read it already.