DacoTaco/priiloader #299: RFC - Allow hacks to work with multiple system menu version ranges

DacoTaco, 2022-04-16:
**Describe the RFC**
Devnul is making a SM mod for wii mini to allow the SD card functionalities to be restored to the wii mini.
however, for safety the version number has been altered so an SM uninstall is not needed.
This breaks the hacks though.
It would be great if hacks work on multiple 'ranges' of hacks or use wildcards

**Expected behavior**
v512 (4.3) hacks to be loaded on the new version

ZweiBR, 2022-05-01:
Do you have a date when the stable version will be released?

Leseratte10, 2022-08-10:
Hm, making the hacks so that you can add multiple version ranges may be interesting, but then whenever someone makes a custom Wii menu and decides on a newer version number scheme the hacks would need to be updated? 

As this is a "request for comments" issue, here's my suggestion - no idea if that would be simple to implement or if it's a good idea, it's just what I thought of: 

Leave the hacks as they are, indicating the "actual" version of the Wii menu they support, but have some way for the system menu to override that version. 

For example, the TMD seems to have a couple padding / unused fields like at 0x104 (60 bytes) or 0x1e2 (2 bytes). What if Priiloader was made to read that value from the TMD, and if it's not 0, assume that *that*'s the actual System Menu version? 

That way for "normal" System Menus straight from Nintendo that value would be 0 and Priiloader would read and use the normal version from the TMD. But then if someone takes the original v512 System menu, updates it to make it compatible with Wii Mini and SD card and whatnot and gives the result a new version number v10000 or whatever, they could set that 2-byte padding value to 512 and Priiloader would "know" that this v10000 system menu is - internally - just a modification / extension for v512 and all v512 hacks can be loaded, too.

urherenow, 2022-08-10:
@Leseratte, you have no clue what you are talking about. You don't even have a clue what RFC stands for.

Leseratte10, 2022-08-10:
I don't? I was assuming it stands for [Request for Comments](https://en.wikipedia.org/wiki/Request_for_Comments). Does it not? Then what does it stand for? 

Also, my point still stands. Priiloader's hacks file shouldn't need to hard-code potential modded version numbers of System Menu mods, in my opinion.

DacoTaco, 2022-08-10:
RFC can mean 2 things. either [Request for Comments](https://en.wikipedia.org/wiki/Request_for_Comments) or [Request for Changes](https://wiki.en.it-processmaps.com/index.php/Checklist_Request_for_Change_RFC). i can understand the confusion, but in my book RFC is Request for Changes.
doesn't mean anyone can't pitch in ideas, i actually encourage people to do so!

however, i don't fully like the TMD idea. it means fucking around with official data structures and their content. a version change is the least you have to do, but using 'unused' data? no thanks.

i still prefer the wildcard idea, since we could use to to basically make hacks work , for example SM 4.3, by saying version 51*.  
is a modded SM created with 4.3 as base? cool, just make it version 516, 517 & 519 and bingo, hacks still work.

urherenow, 2022-08-11:
I have limited internet access, so DacoTaco beat me to it. Even in the most basic of IT studies, RFC is a "Request for Change". I could start listing tech companies that use the term, and in this manner, but it's a moot point since the tag on the first post literally says "RFC - Request For Change".

Leseratte10, 2022-08-11:
Okay, if I google "RFC" then out of the first 25 results, 15 plus the info box are about "Request for Comments", and only one is about "Request for Changes" (and 9 about unrelated stuff) Sorry for assuming the most common use ...

Also, the first post does **NOT** say "Request for Change". When I posted it had the Tag "RFC" which DacoTaco has renamed from "RFC" to "Request for Change" *after* I posted.
 
As for DacoTaco's comment, the idea about the TMD was just a first suggestion. Maybe it should be done differently, like it's done in the cIOS. 

Maybe having wildcards or multiple supported versions for hacks is a good idea, for example when you have a hack that only works on all PAL system menus but not on others. I just think it's a bad idea to use that same system for a bunch of different modified system menu versions. If you start adding custom version numbers to the official Priiloader hacks_hash.ini then everyone who makes custom system menus needs to make sure not to use duplicate version numbers to not cause wrong hacks to be loaded. 

If you somehow manage to include in the system menu a flag that the menu with version 10000 is actually, internally, version 514, then Priiloader just needs to be updated once and then everyone who makes a custom system menu can just flag it with the actual, original version. Whether that should be done in the TMD or elsewhere can be discussed, that was just a first suggestion. 

Why the negativity here? Just because I misunderstood "RFC"? It's never been mentioned here before, it wasn't in the tag description, and even searching for RFC in the repo didn't show an explaination yesterday (just today after the tag's been renamed).

urherenow, 2022-08-11:
Because none of your comments suggest you know what you're talking about. Most "hacks" rely on fixed addresses. Depending on the underlying patching system, locating a specific string of bytes reasonably close to a location, may be sufficient to get the required result. But... When people modify a (or create a custom) file, addresses will likely change, rendering the hacks useless at best, or crash or even damage the system at worst. 

If you're using a modified system menu, then why would hacks be needed anyway? What was modified? Why weren't the desired hacks built in as a part of the modification? 

In the end, all of your issues/concerns are problems of your own making. If you want to experiment with other files that aren't a part of the standard guides/installers, you need to know what you're doing, including which hacks need to go with whatever you're using.

Leseratte10, 2022-08-11:
Did you check the hacks_hash.ini? 

Most hacks have been rewritten to no longer rely on fixed addresses. Instead, they do pattern matching and search for A, replace with B. This makes them work with lots of different versions, and given that a modded Wii Menu is not a recompile of source code but a binary patch instead, most if not all patches that apply to the original would still apply to the patched version.

Why would hacks be needed with a modded system menu? That's your question? If you ask that question, I can hand that right back to you, or the creator of this issue report - if they're making a modded version, why add Priiloader support for their hacks?

Maybe so that the user can still choose in Priiloader which of the patches they want, and which they don't want?

Yes, if I were to make my own System Menu I would certainly be capable of just adding the hacks I know are compatible. But I thought the point was releasing it and having others use it who might not know that?

DacoTaco, 2022-08-11:
both of you chill the fuck down, seriously. if your comments didn't have some valid points i would've deleted it all.
i altered the RFC label so it was obvious and there is no confusion now, and in the future.

as for the issue...

@urherenow : as mentioned, hacks do *not* use fixed addresses. they used to, back in the day, but with priiloader i moved away from such practices so they could support multiple system menu versions, as they already do. 
if they change SM that much that the hacks no longer work, then they would need to talk to me. however, i see your argument on wildcards being very dangerous and it is valid...
grmz

as for the mod, the mod is created to re-enable SD on the wii mini. aka, a semi clean SM. the hacks are obviously the user to choose. like im sure i have hacks enabled that you don't, and vice versa. im glad he didn't bake them in the SM mod.
usually i'd agree to follow standard guidelines and reject the change.
however, i think re-adding SD to the wii mini is a good deal, and upgrading SM is indeed safer than uninstall + reinstall.

I sadly can't find the specs of what devnul wanted and he seems to have left the discord i was in. but maybe its better for him to alter IOS in memory to allow downgrade/overwriting on installation and then install?
in such case he wouldn't need to upgrade the version and i wouldn't need to create some hacky system for the hacks

urherenow, 2022-08-12:
For the record, the second sentence of my argument (and no, I did not just edit it), covers how the hacks currently work. Obviously, it's not good enough when you start altering things, or this entire thread would not exist, nor the need to check versions.

DacoTaco, 2022-08-12:
i am closing this. Devnul seemed to agree with me that his mod basically is installing a SM versions of the same version that has SD card stuff in it. meaning the version increase isn't needed when he patches IOS when installing.

therefor, this change is no longer needed

closed 2022-08-12 by DacoTaco