DacoTaco/priiloader #301: Guide to Unistall Priiloader

Miguel-Czar, 2022-04-22:
Hello, DacoTaco, nowdays many people have installed priiloader in their wii, there are many guides to unistall priiloader, but I haven't found any guide to Unistall It correctly without bricking your wii, I think you should say us some tips if we want to unistall It without bricking, please.

TheTechRobo, 2022-04-22:
If you rerun the priiloader installer, doesn't it show an uninstall button?

DacoTaco, 2022-04-22:
in the installer it indeed should show what button to press to uninstall. 
im not so sure what a guide can do when
1) nearly nobody (that i know off) actively whats to uninstall priiloader
2) it is a 1min job ( run installer in HBC, press button to uninstall)

closed 2022-04-23 by DacoTaco

SuperMarioYAHOOYAHOO, 2023-05-22:
> in the installer it indeed should show what button to press to uninstall. im not so sure what a guide can do when
> 
> 1. nearly nobody (that i know off) actively whats to uninstall priiloader
> 2. it is a 1min job ( run installer in HBC, press button to uninstall)

wuite a late reply, but if i install priiloader first, is it safe to install cios without uninstalling? i need to use USB Loader GX