DacoTaco/priiloader #302: Wii Swiss Booter fails to autoboot

NO-ob, 2022-05-02:
**Describe the bug**
Wii swiss booter boots from the priiloader Installed File option but not when installed file is set as autoboot. I also tried directly installing the swiss.dol in priiloader and booting that. That didn't work either. Both result in a black screen and no display output the wii power button keeps the green light and pushing it makes it go through the boot sequence again

**To Reproduce**
I followed the steps here https://gbatemp.net/threads/how-to-set-up-and-use-swiss-on-the-wii.291505/ for getting swiss working which included installing cmioswiigator and then using wii swiss booter 0.3 to boot into swiss which is stored on the sdgecko in the gamecube memory card slot. installing wii swiss booter in priiloader and launching it via the 'Installed file" option boots into swiss fine but setting the installed file as autoboot and rebooting the system fails to boot into swiss

**Version**
0.9.1

**Expected behavior**
Swiss to boot as it does when launching wii swiss booter from the homebrew menu

**Screenshots**
[If applicable, add screenshots to help explain your problem.]

**Additional context**
Add any other context about the problem here.

DacoTaco, 2022-08-16:
Hi, sorry for the late response.
ive been thinking about this one, and i have no idea at the moment what could cause this issue besides something in swiss. 

to verify if it is a problem in swiss or priiloader, could you provide a log file?
you can enable it from the settings menu by enabling 'dump gecko output' and sending me the Prii.log on your sd card (or usb drive).

i suspect, and hope, the last log line before the crash to be 'BootDolFromMem : shutting down...' or something about the dvd drive shutting down.

DacoTaco, 2022-08-18:
@NO-ob

NO-ob, 2022-08-18:
Hi here is a log from trying to auto boot wii swiss booter doesn't say much, trying to auto boot the swiss dol doesn't create a log file. I'm going to get my other wii set up and will test it out on there to see if theres any difference
[wii swiis booter prii.log](https://github.com/DacoTaco/priiloader/files/9377190/wii.swiis.booter.prii.log)

DacoTaco, 2022-08-18:
thats an oddly... empty log. very odd, i'd have expected more to be honest...

NO-ob, 2022-08-19:
Yeah i cant get it to log at all now i'm going to go get a couple sd cards incase thats a problem

NO-ob, 2022-08-20:
Hi I have tried again with a new sd this is all that gets printed when trying to auto boot then leaving it for a few minutes then force powering off because nothing happens
[prii.log](https://github.com/DacoTaco/priiloader/files/9387314/prii.log)

Heres a log from booting into the priiloader menu with the reset button and manually booting the installed file option which boots into swiss fine with wii swiss booter.
[prii.log](https://github.com/DacoTaco/priiloader/files/9387318/prii.log)

One thing to note is that the swiss.dol is being loaded from the gamecube memory card sd adapter not the wii sd. 
I tried with it loading from the wii sd card also and that didn't seem to make a difference

Manual boot: [prii.log](https://github.com/DacoTaco/priiloader/files/9387338/prii.log)
Auto boot: [prii.log](https://github.com/DacoTaco/priiloader/files/9387342/prii.log)

DacoTaco, 2022-08-20:
could you try to boot it without an usb inserted?
also, priiloader can not access an SD Gecko afaik, so the file it loads is from the front slot SD.

i also assume it works when booted from HBC?

NO-ob, 2022-08-20:
Manual booting the wii swiss booter homebrew works without a swiss.dol on the main sdcard so i assume that deals with the sd gecko? yeah it works from the homebrew channel, autobooting the homebrew channel works also

heres a log with my usbs unplugged after trying to autoboot and then forcing off after blank screen
[prii.log](https://github.com/DacoTaco/priiloader/files/9387680/prii.log)

DacoTaco, 2022-08-30:
i just tried launching this on my wii.
i didn't do the full setup, as i only needed to know if priiloader launched the booter correctly.
the swiss booter is the only bit Priiloader has control over.
i checked the output of my usbgecko and it seems the booter is loading fine, therefore the problem lies in swiss booter/cmios/the dol loader/swiss, not priiloader unless they can tell me what im doing wrong.
[boot.zip](https://github.com/DacoTaco/priiloader/files/9451589/boot.zip)

however, as all other homebrew loads fine, im sure the bug is in their code.

the only difference between autoboot and launching manually is that the sd/usb interfaces & video interface is init (aka VI, which controls video and its clock is used for the gamecube controller port)

use the following dol to install a version that inits video and fat on startup 
[boot.zip](https://github.com/DacoTaco/priiloader/files/9451625/boot.zip)

closed 2022-08-31 by DacoTaco