DacoTaco/priiloader #308: Priiloader 0.10.0

Leonardo5681, 2022-07-31:
Hi, Priiloader works fine and it's great for my wii. But when is version 0.10.0 out? Why can't I wait to try it I just can't install the beta because I only have a wii and I don't want to brick it. I say hurry up but do you at least know the release date? And I would like to know where you are with 0.10.0. Apart from this and a very useful homebrew for the wii.

Masamune3210, 2022-07-31:
This is kind of in poor taste IMO, things are better not rushed, especially when they have the chance of possibly causing a brick. It will likely be done whenever its done. Why are you so rushy to install a new updated version anyway?

DacoTaco, 2022-07-31:
I'd like to release 0.10 soon ™️  but i'd need to tackle the 2 issues marked as version blocking first. sadly don't have time to do so recently.
i hope to get it done in my holiday between now and 2 months

Leonardo5681, 2022-08-03:
Oh sorry I hope you will be able to pass these blocks. I will always be updated on Priiloader for 0.10.0. If you want to email me when the version comes out you can.

closed 2022-08-14 by DacoTaco

Leonardo5681, 2022-08-30:
@DacoTaco have you passed the two blocks? It is not to rush you and just out of curiosity

TheTechRobo, 2022-08-30:
@Leonardo5681 https://github.com/DacoTaco/priiloader/issues?q=is%3Aissue+is%3Aopen+label%3AVersionBlocking