DacoTaco/priiloader #309: Support for 1.0E/U?

wowokaybye, 2022-08-07:
Priiloader installs without issues on a 1.0E system menu, but once it is installed the system menu can't boot anymore.
System Menu 1.0 would benefit so much from Priiloader support, since you could block the disc updates this way.

DacoTaco, 2022-08-08:
and what error does it give? what happens when you select to boot SM?
i need more info here...

wowokaybye, 2022-08-08:
Thank you for your response. Once I install Priiloader, it automatically starts when turning on the Wii and simply says something along the lines of "unable to autoboot system menu" or "unable to boot system menu". It can start the HBC just fine. But sadly I couldn't try if the hacks work, since I'm unable to access the menu.  Once I'm home I will be able to provide screenshots of the issue. If you want to get in contact on Discord let me know!!

DacoTaco, 2022-08-08:
try to enable the dump debug setting in the settings and booting system menu.
after that, disable the setting again and send me the Prii.log from your SD card

DacoTaco, 2022-08-08:
please verify by building the latest commit. it should work, just like i told hallowizer it should be done.

wowokaybye, 2022-08-08:
Sadly, it still doesn't work for me. I provided the .log file here:

wowokaybye, 2022-08-08:
23:55:23 : SD: Mounted
23:55:23 : AutoBoot:System Menu
23:55:23 : BootMainSysMenu Exception -> ES_GetStoredTMDSize error -1017
23:55:23 : FAT_Init():1
23:55:27 : priiloader v0.10.0 BETA 1 (Sys:2)(IOS:9)(Aug  8 2022 17:11:01)
--------gecko_output_enabled------
--------gecko_output_enabled------
--------gecko_output_enabled------
23:55:45 : BootMainSysMenu Exception -> ES_GetStoredTMDSize error -1017

wowokaybye, 2022-08-08:
Installer.log 
HackMii v1.2 installer starting up
PVR = 00087200
running under IOS 61 rev 0x151d
50 titles are installed
Found IOS 61: revision: 0x151d.
Found IOS 60: revision: 0x1900.
Found IOS 57: revision: 0x161d.
Found IOS 56: revision: 0x151d.
Found IOS 55: revision: 0x151e.
Found IOS 53: revision: 0x151e.
Found IOS 51: revision: 0x1300.
Found IOS 50: revision: 0x1400.
Found IOS 38: revision: 0xf1b.
Found IOS 37: revision: 0xf1d.
Found IOS 36: revision: 0xd17.
Found IOS 250: revision: 0xff00.
Found IOS 249: revision: 0xff00.
Found IOS 223: revision: 0xff00.
Found IOS 222: revision: 0xff00.
Found IOS 16: revision: 0x200.
Found IOS 10: revision: 0x300.
Found IOS 70: revision: 0x1a1f.
Found IOS 254: revision: 0xff01*
Found IOS 35: revision: 0xd15.
Found IOS 34: revision: 0xd14.
Found IOS 33: revision: 0xc13.
Found IOS 31: revision: 0xd15.
Found IOS 30: revision: 0xb00.
Found IOS 28: revision: 0x60e.
Found IOS 22: revision: 0x40d.
Found IOS 20: revision: 0x100.
Found IOS 17: revision: 0x307.
Found IOS 15: revision: 0x20b.
Found IOS 14: revision: 0x208.
Found IOS 13: revision: 0x111.
Found IOS 12: revision: 0x10d.
Found IOS 11: revision: 0x100.
Found IOS 21: revision: 0x30e.
Found IOS 2: revision: 0x1e2.
Found IOS 9: revision: 0x30a.
Found IOS 4: revision: 0xff00.
launching IOS 61 for the installer...

IOS launched...

IOS versions: Installer: 61, HBC: 0
starting preparations

device id: NG0484f859
Waiting for mini to load ...
Mini version: 1.3
NAND minpage: 0x200
set NAND minpage to 0x40
Calling boot2_run(1, 61)
Waiting for boot2 to load ...
Waiting for IOS ...
IOS loaded: IOS61 v21.29
IPC started (531)
boot2_parse_blocks: examining existing boot2 layout
boot2 block 0:
00000000  00 00 00 20 00 00 0f 00 00 00 0a 00 00 00 02 a4  ... ............
00000010  00 00 02 08 00 00 00 00 00 00 00 00 00 00 00 00  ................
00000020  00 01 00 00 b3 ad b3 22 6b 3c 3d ff 1b 4b 40 77  ......."k<=..K@w
00000030  16 ff 4f 7a d7 64 86 c8 95 ac 56 2d 21 f1 06 01  ..Oz.d....V-!...
00000040  d4 f6 64 28 19 1c 07 76 8f df 1a e2 ce 7b 27 c9  ..d(...v.....{'.
00000050  0f bc 0a d0 31 25 78 ec 07 79 b6 57 d4 37 24 13  ....1%x..y.W.7$.
00000060  a7 f8 6f 0c 14 c0 ef 6e 09 41 ed 2b 05 ec 39 57  ..o....n.A.+..9W
00000070  36 07 89 00 4a 87 8d 2e 9d f8 c7 a5 a9 f8 ca b3  6...J...........
offset 0xE00:
[... 166 lines cut ...]
00000060  01 01 01 01 01 01 01 01 01 01 01 01 01 01 01 01  ................
00000070  01 01 01 01 01 01 01 01 01 01 01 01 01 01 01 01  ................
parse_blocks: found 3 valid blockmaps
found 3 valid blockmaps
using generation 6
boot2 blocks: 02 03 04 05 06
boot2bin @5,0 (blocks=2,3 - bootmii=0)
boot2 blocks: 06 05 04 03 02
boot2bin @5,1 (blocks=6,5 - bootmii=0)
boot2 block layout: 1 1 2 2 1 0 0
Assembling boot2 in RAM
copyright size = 34
copyright text = 'Copyright 2008-2012 Team Twiizers.'
tmd ends at 0x80372a6c, data starts at 0x80372aa0
bootmii version  = '1.5'
Ok for boot2 install
preparations done!

[33;1mThe test results are in:

[37;1m  Using IOS versions:      Installer: 61, HBC: 61

  The Homebrew Channel:    [32;1mCan be installed[37;1m
  BootMii:                 [32;1mCan be installed[37;1m
......................................................

DacoTaco, 2022-08-08:
IOS9 is returning Invalid arguments

wowokaybye, 2022-08-08:
That's fine. Thank you for your hard work

DacoTaco, 2022-08-13:
ok, so i did some invenstigations : 
* Priiloader used to assume the last SM .app is the booting dol, this is incorrect and makes it fail to boot
* Priiloader was loading the TitleTMD using the TMDViews but those don't have the index of the boot content.
* the code as it loads the TMD using ES calls (not views, actual TMD) , but IOS9 don't have these calls implemented.

to this day, if you have a SM > 1.0, and you select IOS9 to boot SM with, it will fail because of the missing etTmd + ES_Identify calls.

IOS < 28 -> no ES_GetStoredTmd* calls
IOS < 20 -> no ES_Identify call

so im thinking that if the running IOS < 28, to just try and load the files directly?
im not sure that would work with modern IOS9 versions, who need to have nand access patched. 
i mean, i can patch that into ios from priiloader, but the question is if i should have an ios patch in priiloader, just for some ancient ios versions and not break functionality/ rare cases :/

wowokaybye, 2022-08-13:
I thank you for further looking into this. At the end of the day it is fully your decision if it is worth the trouble. :) Obviously it would be terrific to have Priiloader support for all 4 Wii versions but I understand that it is not requested by many.

DacoTaco, 2022-08-14:
btw, since you have System Menu 1.0 somewhere, could you check if launching titles works correctly?

im currently implementing that IOS < 28 fix and would like to know this so i can design the code correctly

wowokaybye, 2022-08-14:
I tried launching the following titles on 1.0 using Priiloader:
Homebrew channel ✓
Photo Channel ✓
Forecast Channel ✓
Random VC Game ✓
All work :)

wowokaybye, 2022-08-14:
Launching Disc works aswell

DacoTaco, 2022-08-14:
and it showed the titles and everything? that sounds good! 
means i will only have to alter the System menu booting!

wowokaybye, 2022-08-14:
Yes it correctly showed all the titles installed :)

DacoTaco, 2022-08-14:
could you try the latest commit on main?

i made priiloader check IOS version. when its < 28 it will load the tmd from nand and if it is loading to an ios < 28 it patches ios to have nand access

that should work, i think :)

wowokaybye, 2022-08-14:
Now booting system menu doesn't spit out an error, but it boots into a black screen instead, the disc drive starts as if the Wii menu starts, but nothing happens. Log doesn't seem to output anything?

wowokaybye, 2022-08-14:
If I have the white theme on, the screen stays white. So I think it doesn't boot anything at all

DacoTaco, 2022-08-14:
re installation has cleared the settings, so maybe logging was turned back off?

either case, what does the log say if you enter the priiloader menu and hit launch System Menu ?
kinda sucks i don't have access to System menu 1.0 😅

wowokaybye, 2022-08-14:
Loggin is on, I mean that it doesn't write anything into the log when I press System Menu. It seems to stop there (or crash?)  13:53:55 : GetTitleTMD : Load TMD from nand
13:53:55 : using 3702D2CF for booting
13:53:55 : loading hacks
13:53:55 : LoadHacks : hacks_hash.ini not on FAT or Nand. ISFS_Open error -106
13:53:55 : input shutdown
13:53:55 : Hacks:0
13:55:02 : SD: Mounted
13:55:02 : Reset Button is held down
13:55:02 : FAT_Init():1
13:55:02 : priiloader v0.10.0 BETA 1 (Sys:2)(IOS:9)(Aug 14 2022 11:41:58)

wowokaybye, 2022-08-14:
Do you have Discord? I feel like we could communicate there better :0

DacoTaco, 2022-08-14:
thats odd, that means priiloader is indeed crashing before it even got to decide what file to load, as it should have given something like "using 00000002 for booting" or something a like.

it must be stuck trying to get the tmd of nand. 

ye i have discord. i can usually be found on the r/wiihacks discord. if you want me to DM you, just give me your handle (i don't like posting my handle in public)

wowokaybye, 2022-08-14:
Great! I edited my message above with my handle!

DacoTaco, 2022-08-14:
SM 1.0 is fully bootable. but we need to verify if all hacks can still apply or not

DacoTaco, 2022-08-14:
@Leseratte10 : the wiimmfi patch seems to be broken in 1.0E according to @wowokaybye. this is probably something you can fix

Leseratte10, 2022-08-15:
Wiimmfi patch is broken in 1.0E? What does that mean? The Wiimmfi patch isn't even implemented in versions <= 4.0 (it only supports 4.1, 4.2, 4.3), and if I were to add some more versions then I don't think 1.0 would have a high priority.

DacoTaco, 2022-08-15:
@wowokaybye : how were you able to test the wiimmfi patch? did it show up in your list somehow? :/