DacoTaco/priiloader #310: Disc drive doesn't spin anymore

BaamAlex, 2022-08-11:
My disc drive doesn't spin anymore. I tested a bit with rvloader a few days ago. After uninstalling all that stuff, the disc drive doesn't spin anymore. The thing is, ctgp or usb loader can read and play the disc normally. The disc drive just doesn't work in the regular wii menu.

Now I don't know if this is a priiloader or rvloader problem. I posted already an issue on aurelios github repo, but didn't get an answer from him until now.

And I don't want to restore my nand backup when the problem could be solved otherwise. I hope you can help me.

DacoTaco, 2022-08-11:
thats odd. 
this is also the first time i hear about this.

could you check the priiloader settings? is it set to shut down the dvd drive on startup?

BaamAlex, 2022-08-11:
It is disabled. I restored my nand backup today, but that didn't change anything. Now i swapped out the disc drive with another one (obviously). Let's hope the old drive is just broken somehow. I report later if this fixed my problem.

BaamAlex, 2022-08-12:
Update... changing the disc drive doesn't change anything. The wii menu is still unable to read the disc.

DacoTaco, 2022-08-12:
could you disable all hacks in priiloader and check again?

BaamAlex, 2022-08-14:
I could try that yes.

DacoTaco, 2022-08-14:
thing is, i don't do anything special with the drive unless a setting or hack makes me, which is why i find this issue weird.
i would assume some hack or setting (like the system menu ios setting?) might be interfering :/

DacoTaco, 2022-09-05:
@BaamAlex : do you have bootmii @ boot2? could you check if everything is ok with priiloader 0.10 beta 2? 
i found , and fixed, some minor dvd drive issue

BaamAlex, 2022-09-05:
I have found the solution. It's even a little embarrassing to admit it. There was no problem with the Priiloader. My console is NTSC. And my discs are PAL. Of course, I have to enable "Region Free Everywhere" in the system hacks for that. This was so obvious to me that I must have somehow forgotten that this option was enabled all the time. How stupid do you think I felt when I found that out...?

The issue can be closed if you want. Cheers.

DacoTaco, 2022-09-05:
oh good, thank god its nothing special thats broken !
good that you figured it out!

closed 2022-09-05 by DacoTaco